mavlink/mavlink #893: I got an attributeError in mavgenerate.py

tnpd, 2018-04-20:
AttiributeError in mavgenerate.py.
~~~
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1540, in __call__
    return self.func(*args)
  File "/home/tn/mavlink/mavgenerate.py", line 141, in browseXMLFile
    xml_file = tkinter.filedialog.askopenfilename(parent=self, title='Choose a definition file')
AttributeError: 'module' object has no attribute 'askopenfilename'
~~~

hamishwillee, 2018-04-20:
Please describe environment - OS, version of Python. What you were doing to get the error - ie just starting the file or something else?

I just tried this in Ubuntu 16.04, Python 2.7 (default version) following installation instructions [here](https://mavlink.io/en/getting_started/#install) and it started and built common.xml for MAVLink 1.0 fine.

tnpd, 2018-04-20:
Sorry.
I posted incomplete text accidentally.
My enviroment is Ubuntu 16.04 and python 2.7.
I was trying to "Getting Started" written in mavlink.io.

tnpd, 2018-04-20:
I tried "python -m mavgenerate". 
I could see GUI interface after it, but it threw the error I posted above when I pushed the "Browse" button.
Sorry, I don't have build log.  How can I get it?

I tried in clean 16.04 just now. It works well, but I found the code in mavgenerate.py which determine import tkinter or Tkinter is not working. Always it imports tkinter even if it is in Python2.7. 

And I also found the directory of "tkinter" is different between clean 16.04 and the environment I tried first. 
The former is in "/home/tn/.local/lib/python2.7/site-packages/" , the latter is in /usr/lib/python2.7/dist-packages/.  
I can't remember why this difference is happened. 
And, It seems there are some difference in how bridge tkinter to Tkinter when tkinter is imported in python2.7 code.

hamishwillee, 2018-04-20:
Thanks! 
@julianoes Are you OK with the suggested code change in the main description: https://github.com/mavlink/mavlink/issues/893#issue-316094456

While I don't understand how the Python 3 tkinter can be in a Python 2.7 environment, clearly the solution given will work, and I believe is cross-platform.

closed 2018-04-20 by tnpd

ShockvaWe, 2019-03-01:
I changed the code like in the #893 but I do still get : 
`Traceback (most recent call last):
  File "mavgenerate.py", line 207, in <module>
    app = Application()
  File "mavgenerate.py", line 56, in __init__
    self.createWidgets()
  File "mavgenerate.py", line 128, in createWidgets
    self.strict_units_value.set(mavgen.DEFAULT_STRICT_UNITS)
AttributeError: 'module' object has no attribute 'DEFAULT_STRICT_UNITS'`
Error.
OS : Windows 10 Python : 2.7.15

hamishwillee, 2019-03-04:
@ShockvaWe This error was closed in 2018 and does not appear related to your problem. If you want us to follow this up, please remove all your code changes and follow the instructions here: https://mavlink.io/en/getting_started/installation.html

If you still get the problem, raise a new issue

commit b50c7c0c (2018-04-25): Improve Py 2/3 compatibility to fix #893