mavlink/mavlink #898: MAVLink set_home_position in local frame coordinate system

alexx600, 2018-05-01:
Hello,
I would like to ask if it is possible to change home position using local NED x,y not global lattitude, longitude coordinate?

According to http://mavlink.org/messages/common#SET_HOME_POSITION “The global and local positions encode the position in the respective coordinate frames”.
What shoud i send as a lattitude and longitute if i would like to move home position in local frame (for example 2 m forwad)?
I tried set lattituede and longitude to zero, but this caused reseting home position. Is it possible to change home position in local system?

shubham9436, 2019-07-21:
Commenting to get the notification

julianoes, 2019-07-22:
Is this a question for the MAVLink protocol or PX4 or ArduPilot?

hamishwillee, 2019-07-23:
@julianoes FWIW I don't think PX4 handles this message at all. But that said, it isn't clear (to me) how this should work - and that has to be clarified.

What are X, Y, Z? I think they are NED of home position relative to current vehicle position - correct? In that case we would minimally need to clarify whether Z is positive up.

Can you just specify X, Y, Z or just lat, lon? It looks like you have to specify everything, which makes this a bit useless for indoor use.

Last of all, this should note that the value of HOME can be changed underneath you by the system.

hamishwillee, 2019-08-21:
@julianoes  How do we progress this. Shubham appears to have lost interest.

I would still quite like to understand how the message is used: https://github.com/mavlink/mavlink/issues/898#issuecomment-514027281 - 
IN particular, related to local frames, and whether both frames need to be set (unlikely). Local frame is particularly unclear because I don't know "what" local frame - quite a lot have been defined e.g. https://mavlink.io/en/messages/common.html#MAV_FRAME_LOCAL_FRD. Perhaps it doesn't matter, or that is something that is defined elsewhere?

julianoes, 2019-08-21:
Wouldn't this position handled like any other position with frame in `COMMAND_INT`?

hamishwillee, 2019-08-22:
COMMAND_INT allows you to specify a frame; this doesn't. Or are you saying something else?

julianoes, 2019-08-22:
Oh, nevermind. Could we add the frame as an extension field?

hamishwillee, 2019-08-23:
> Oh, nevermind. Could we add the frame as an extension field?

Yes, and I have done that in https://github.com/mavlink/mavlink/pull/1216

That still doesn't fix everything though. I added my additional question in above PR.

julianoes, 2019-08-23:
Probably only one global and one local frame will be supported at first. If this was a command you could just nack the request with `UNSUPPORTED`. By having this as a separate message, I'm not sure how it will work.