mavlink/mavlink #901: Add temperature sensor instance reporting

jaxxzer, 2018-05-07:
Suggestion for a new mavlink message:

id: TEMPERATURE
- char[16] name
- float temperature (degrees C)

I know that the plotting tools do not support 'instance fields' in a mavlink message, but that is another problem, and I think this is the most sensible way to enable reporting temperatures from different components ie 'imu1' 'esc1' 'probe1' in a flexible manner.

WickedShell, 2018-05-07:
Can we do this without a string? (IE I'd rather see a uint8_t instance and a uint8_t that takes a type enum where we can just add values for SENSOR_ESC SENSOR_IMU etc)

jaxxzer, 2018-05-07:
I think the string will add a lot of value for users, hackers, and developers alike

WickedShell, 2018-05-07:
I think strings are a bad choice for a couple of reasons:
1. They are a continual hassle to deal with, an identifying tuple tends to be a lot easier to pass around a GCS.
1. It makes the configuration side on the vehicle painful, as users can't easily input parameters to configure what the name is.
1. It's not a compatible approach across vehicles. IE PX4 and ArduPilot might pick different strings for sending the ESC temperature, which means a GCS needs to understand both, or in turn maintain a list of all temperature probes just mushed together, and it can't attempt to separate them by function easily.

As a different note the message should also be timestamped.

jaxxzer, 2018-05-07:
> As a different note the message should also be timestamped.

Ok, edited.

I think the enum makes more sense in theory, but in practice, idk. For a user, it's a PITA to look up bitfields and enumerations.

I never had any intention of allowing the user to control what was sent in the `name` field. I mean for a user it is easier to just see 'esc0' and know that is the temperature of the imu rather than going to the mavlink spec to look it up. Idk what you mean about a tuple, but from the GCS side, char[] vs int for the key seems to me like a trivial difference to handle. This can all amount to the same thing with different implementations, I'm not against the suggested modifications.

TSC21, 2018-05-07:
This should be discussed on an RFC first.

TSC21, 2018-05-07:
https://mavlink.io/en/contributing/contributing.html

jaxxzer, 2018-05-07:
Does a new message really require such a process? This is not an additional micro-service. It is protocol spec.

`Protocol specification and documentation changes are usually changes with less impact and can be directly raised as pull requests against this repository.`

TSC21, 2018-05-07:
Any new msg is subject to RFC

jaxxzer, 2018-05-07:
cool.

jaxxzer, 2019-02-06:
There is currently no alternative. I think it's entirely desirable to be able to report the temperature of some arbitrary component or probe. I've got no interest in opening an RFC as that obligates me to implement the changes. https://github.com/mavlink/rfcs/blob/master/text/0001-rfc-format.md#detailed-design.

Besides, this document says this should not warrant an RFC: https://mavlink.io/en/contributing/contributing.html

hamishwillee, 2019-02-07:
@jaxxzer I completely agree - this does not require a RFC. Often though it is easier to start with a PR than an issue. @LorenzMeier  just wanted to make sure that this is still of interest before we re-engage. 

I see your point about logs, but to me the uncontrolled nature of the string pointed out by @WickedShell is a bigger problem. 

The simplest option is to do it like [DISTANCE_SENSOR](https://mavlink.io/en/messages/common.html#DISTANCE_SENSOR) - a standalone message with fields

TEMPERATURE SENSOR
* uint_8 id: sensor instance
* uint32 timestamp (us)
* float temperature (degrees C)

Do we need other fields? Are there types of sensors? Is it useful to be able to specify the range at which the sensor can be trusted? etc.

I do not like the idea of doing it generically like this ...

SENSOR
* id: sensor instance
* type: sensor type (enum)
* uint32 timestamp (us)
* float value

Because
- some sensors require other useful information - see DISTANCE_SENSOR 
- you can't define units or bounds for the generic value that will make sense for all cases. Makes things harder for GCS.

hamishwillee, 2020-04-23:
I'm closing. We can reopen though if you actually need it still and plan to implement it.

closed 2020-04-23 by hamishwillee