mavlink/mavlink #920: C++11 main include missing?

hdino, 2018-05-22:
I generated the standard messages for C++11, but cannot figure out which file I should include.

Obviously, `standard.hpp` has to be included somehow. A `find` in the output directory shows that only `standard/gtestsuite.hpp` includes `standard.hpp`. Hence, I tried to include `standard.hpp` directly, but then the macros from `protocol.h` and `MAVLINK_MESSAGE_CRCS` from `standard.h` and `common.h` are missing.

Do I follow a wrong approach or is an appropriate include file missing?

vooon, 2018-05-23:
If you wan't to use C++11, and not C library, then you should only include `<dialect>.hpp` and nothing else.

Also it is required to provide `mavlink::mavlink_get_msg_entry()` in your code.
You may use [libmavconn][1] as an example.

[1]: https://github.com/mavlink/mavros/tree/master/libmavconn

hdino, 2018-05-25:
Thanks for the tipps! Actually, there were just two macros missing:
```
#define MAVLINK_START_SIGN_STREAM(chan)
#define MAVLINK_END_SIGN_STREAM(chan)
```
Now everything works fine.

I had a look at the documentation, but I couldn't find anything about these macros.

closed 2018-05-25 by hdino

vooon, 2018-05-25:
That macroses used for packet signing (extension) in v2. But as i know it implemented only in APM and MAVProxy.

hamishwillee, 2018-05-29:
Hi @hdino ,

Following your experience, would you be able to summarise how to use the libraries in https://mavlink.io/en/getting_started/use_source.html#c11 ?