mavlink/mavlink #921: Trouble with 'runpy.py', can u help me?

nyankobw, 2018-05-22:
When I type `python -m mavgenerate`
'cmd' say that
```
Traceback (most recent call last):
  File "C:\Python27\lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "C:\Python27\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Users\USER\Desktop\Study\MavLink\mavlink\mavgenerate.py", line 44, in <module>
    from pymavlink.generator import mavgen
ImportError: No module named pymavlink.generator
```

So even though I studied 'runpy.py', I didn't make much progress.

Can I get some advice on this?

hamishwillee, 2018-05-22:
Yes, follow the installation instructions here: https://mavlink.io/en/getting_started/installation.html

In particular, note the command to also clone the submodules `git submodule update --init --recursive`

hamishwillee, 2018-05-22:
Oh, and best place for MAVLink support questions is Slack: https://mavlink.io/en/#support

nyankobw, 2018-05-22:
@hamishwillee Thanks to advice. I'll try.

nyankobw, 2018-05-22:
@hamishwillee ho...

Why didn't I do this(`git submodule update --init --recursive`)?
I was very stupid.
thanks...

hamishwillee, 2018-05-23:
Accidents happen :-). Good luck.

closed 2018-05-23 by hamishwillee