mavlink/mavlink #925: Issue/PR closing process

hamishwillee, 2018-05-28:
What is the triage and issue closing policy for this project?

Some projects allow old/incomplete PRs and issues to live forever - which has the problem that it can be hard to see genuine problems. Others ruthlessly close any issue as stale if it is not marked as solved or otherwise commented in a certain time period (e.g. #529) - using activity as a measure of priority/interest/importance.

I don't know the ideal policy, but I do feel that both approaches discourage bug posting. Irrespective of the policy, it should be articulated, and ideally semi automated.

hamishwillee, 2018-05-28:
FWIW I do not like auto-closing of issues as stale. In a community driven project, I think it is better to auto-send a prompt to the originator to ask them if the issue is still a problem for them.  It should be possible to mark an issue in such a way that it will remain open until fixed (without notification).

LorenzMeier, 2018-05-28:
We should set the automation for a double-ping: Initially a warning that the issue will be closed and then an auto-close. Doing that manually is not consistent and time-consuming. A bot can handle it.

Generally, I would like to have stale issues removed because they clearly distract from relevant ones. If someone cares they should remain responsive (or the maintainer adopts the issue).

So in short: We are reasonably ruthless in closing them, but a warning makes sense.

hamishwillee, 2018-05-28:
On further thought, the problem here is actually that the project has not been sufficient responsive and good at driving discussions through to a conclusion (point #529 is a good example).

So yes, works for me, if we can get better at defect triage/discussion ... 
Maybe that is something that could be better shared?

closed 2019-06-04 by hamishwillee