mavlink/mavlink #934: Separate repository for XML files

benbarkay, 2018-06-20:
Keeping the definition XML files in a separate repository would make it easier to git sub-module them rather than copying the files.

hamishwillee, 2018-06-25:
@amilcarlucas @LorenzMeier Do you have an opinion on this one? Personally I think it would be nice if the DTD lived in mavlink repo.

amilcarlucas, 2018-06-25:
mavlink/mavlink already contains mostly .xml files.
Code gets generated by the ardupilot/pymavlink submodule.

This seams to want to create a mavlink/base repo with two submodules:
- mavlink/mavlink_definitions
- ardupilot/pymavlink

That is too much confusion for almost no real added value.
Every time mavlink/mavlink_definitions gets changed we would need to change mavlink/base

What I think would be a REAL improvement would be https://github.com/ArduPilot/pymavlink/pull/7 and https://github.com/mavlink/mavlink/pull/915

Basically add proper structuring to the XML, moving information from comments (where they are not machine parseable) into proper machine parseable information. Those are REAL improvements.

benbarkay, 2018-06-26:
No, that is a false dilemma due to that there is no "mavlink/base", there is just mavlink, and pymavlink:

`pymavlink` *depends* on `mavlink_definitions`, it makes sense that it will submodule to it. You don't need `base` which is the glue to the two other repositories, which are not decoupled from eachother -- `pymavlink` is coupled to `mavlink_definitions`, and `mavlink_definitions` is independent, there is no need for a gluing repository.

hamishwillee, 2020-07-27:
@amilcarlucas I am inclined to agree with Ben above https://github.com/mavlink/mavlink/issues/934#issuecomment-400370414

Pymavlink is just one generator that takes xml as an input, and that xml depends on the xsd. To me the xml and xsd should be together in one repo so that anyone who wants to use it can, without taking everything else.

amilcarlucas, 2020-07-28:
The [current schema](https://github.com/ArduPilot/pymavlink/blob/master/generator/mavschema.xsd) IS part of pymavlink. It makes sure that the genarator validates the .XML files before generating the code.

If the .xsd is not part of pymavlink, and both .xsd and .xml files change outside of the pymavlink repo, then code generation whould potentialy get broken.

Feel free to do an upstream PR for pymavlink removing the .xsd file from it. But I am preety sure that the answer above is what you will get from tridge: The xsd is there to protect the generator from ill formed .xml files.

hamishwillee, 2020-07-29:
@amilcarlucas I know why it is in pymavlink, but neither of these points are valid arguments. _Any_ generator that wants to validate it's inputs needs that XSD. If the XSD and XML were co-located then any generator, including pymavlink could take them and nothing else. The only reason to keep them in pymavlink is to say "screw you" to any other generator. 

I'm happy to PR them out. My only concern is how to manage the dependency - ie can this be in mavlink/mavlink/xsd/ or does it need to be in a submodule mavlink/xsd. What would be the least disruptive?

amilcarlucas, 2020-07-29:
I'm for a mavlink/mavlink/xsd/
No submodule. But please ask @tridge about this.

hamishwillee, 2020-07-30:
@benbarkay @amilcarlucas OK, I spoke to Tridge about this. His "take" on this is that having a more complex module structure (by splitting out the XSD) does add developer friction that is probably not worth the hassle (vs cost of having to pull in pymavlink if you don't need it).

At this point I'm going to close this as WontFix. If someone wants to argue this then the way forward would be to attend the Devcall to explore options going forward.

closed 2020-07-30 by hamishwillee