mavlink/mavlink #935: Remove unused parameters from MAV_CMD xml

benbarkay, 2018-06-21:
Given that `CommandLong` and `CommandInt` already specify the payload structure, there is no need to specify `<param index="N">Empty</param>`. A benefit from removing these is that programs that parse the XML files could tell which params are used and which params aren't.

hamishwillee, 2018-10-11:
This seems simpler to me than parsing params for the text "Empty". Have you tested that the various generators are smart enough to cope with non-defined params (ie don't require a param to be defined?).

@amilcarlucas Gut feeling - good idea or not? IF we decide this is a good idea how would we test it - run the main generators from mavgen before and after and check that generated output doesn't change?

amilcarlucas, 2018-10-11:
Yes, that would be a good test, ... but I have a gut feeling that this will be a problem when param1 is empty and param2 is not.

hamishwillee, 2018-10-12:
Looking at generated Python libraries, you can see that each enum gets a dict of enum values, and each enum value has a dict of parameters.  In the original version we had indexed values for each parameter, and now we're missing all the empty values.
```
enums['MAV_CMD'][30] = EnumEntry('MAV_CMD_NAV_CONTINUE_AND_CHANGE_ALT', '''Continue on the current course and climb/descend to specified altitude.  When the altitude is reached continue to the next command (i.e., don't proceed to the next command until the desired altitude is reached.''')
enums['MAV_CMD'][30].param[1] = '''Climb or Descend (0 = Neutral, command completes when within 5m of this command's altitude, 1 = Climbing, command completes when at or above this command's altitude, 2 = Descending, command completes when at or below this command's altitude. '''
enums['MAV_CMD'][30].param[7] = '''Desired altitude in meters'''
```

Any code that iterates the param dict by index will fail with key errors. I do suspect that we will see code that does this.

hamishwillee, 2018-10-12:
For C. 
The docs get a bit screwed up because they assume all params populated. E.g. for MAV_CMD_CONDITION_GATE
*  | Altitude: 0: ignore altitude| Empty| Empty| Latitude| Longitude| Altitude|  */
* | Altitude: 0: ignore altitude| Latitude| Longitude| Altitude|  */

Other than that, on brief code inspection, it looks like everything should work, because as far as I can see there is no data structure that people might try access index. Could be wrong. I'd want to test a bit.

hamishwillee, 2020-04-23:
We could actually do this now, but I'm still not sure whether or not we want to. 
Closing because if this irritates someone enough they will come back to it with a PR.

closed 2020-04-23 by hamishwillee