mavlink/mavlink #936: Add attributes "name" and "enum" to MAV_CMD.param

benbarkay, 2018-06-21:
I am not so sure whether or not `MAV_CMD` is undergoing deprecation due to the increase in message id size, however, given that it is still being used, adding attribute `name` and `enum` would allow generation of specific functions.

This is especially beneficial in statically typed languages, where this modification would significantly increase discoverability and convenience in using the generated API.

amilcarlucas, 2018-08-07:
This is basically a subset of what ArduPilot/pymavlink#7 wants to achieve. Please add any future comments there

closed 2018-08-07 by amilcarlucas

amilcarlucas, 2019-04-24:
This is now 98% done. If you want to help achieving 100%, review #1105

hamishwillee, 2019-04-30:
And in the interest of completeness, 
1. The name attribute is actually called "label"
1. There is no intention to deprecate MAV_CMD because it is heavily used. We might however consider making it an explicit type rather than a special type of enum.
1. There are other attributes, primarily intended to make display in GCS better: https://mavlink.io/en/guide/xml_schema.html#param