mavlink/mavlink #942: Conditions to get company's MAVLink extension .xml file to this repo

okalachev, 2018-07-02:
Hello!
If I want to extend MAVLink with some specific messages, what should I do to add the .xml file here?
Are any .xml's accepted? Or are there some requirements?

hamishwillee, 2018-07-02:
Hi @okalachev 

Create a Git PR to put it into https://github.com/mavlink/mavlink/tree/master/message_definitions/v1.0

> Are any .xml's accepted? Or are there some requirements?

We'd expect the xml file to successfully build (i.e. be compliant with the DTD) and have comment/description about the platform it supports as a comment (ideally with platform URL/contacts). 

On submission there may be other requirements, but I doubt it.

okalachev, 2018-07-03:
Thanks!
The most significant issue, I was wandering, are there some requirements for the platform itself?
We're developing an autonomous drone system, that requires extensions for MAVLink. This system is under active experimental usage, but I hardly can say, that this it's in production, or that it is commercially available, for example.

magicrub, 2018-07-03:
@okalachev This whole repo is full of pull requests of people submitting changes. Please make your own request and we can discuss it from there

hamishwillee, 2018-07-03:
@okalachev Yes, as above we can discuss this once you have a PR and iterate. 

We don't require great detail from you about the system. What is important is that people can find it if needed, won't accidentally use it if not needed, and that we can have someone to contact in future if we want to (say) merge the definition into common messages or do clean up/maintenance.

> are there some requirements for the platform itself?

I don't know what you mean by "the platform", and what sort of requirements you might mean. Generally if you're creating your own dialect you will import the common.xml messages and you will provide your own new messages - which can pretty much do whatever you like as they will be interpreted by your own system.

hamishwillee, 2020-04-23:
This is answered. Closing

closed 2020-04-23 by hamishwillee