mavlink/mavlink #943: /usr/bin/python: No module named tools

gin2017, 2018-07-06:
I have  git cloned the mavlink from github.
And then do some steps below to build the source code .
a. mkdir build && cd build
b. cmake  ..
c. make 
Then some questions has occured. Anyone can help me ? Thanks very much.
[image]

gin2017, 2018-07-06:
Anyone can help me ???

hamishwillee, 2018-07-12:
Docs on setup etc are here: https://mavlink.io/en/getting_started/

closed 2018-07-12 by hamishwillee

pjalocha, 2023-02-25:
> git submodule update --init --recursive

This is an absolutely essential step - it must be mentioned in the README - other you walk around trying to solve issues, while nothing works.

hamishwillee, 2023-03-08:
Does no one know how to read? This is covered in the manual. 

I guess the answer is no. I'll add a quick start.