mavlink/mavlink #950: Mission Protocol: Are MAV_MISSION_RESULT enums errors or warnings or mixed?

hamishwillee, 2018-07-16:
The [Mission Protocol](https://mavlink.io/en/protocol/mission.html) and XML messages do not specify whether the [MAV_MISSION_RESULT](https://mavlink.io/en/messages/common.html#MAV_MISSION_RESULT) values other than [MAV_MISSION_ACCEPTED](https://mavlink.io/en/messages/common.html#MAV_MISSION_ACCEPTED) are errors or warnings or mixed.

In most cases it is clear that they are **errors** - they are sent to indicate conditions that the sender cannot recover from (e.g. not understanding a message id). The sender should therefore return to an idle state. This is the implementation for PX4. 

There is however one exception: `INVALID_SEQUENCE`. As reported in  https://github.com/mavlink/qgroundcontrol/pull/6718 ArduPilot will continue processing after returning this Ack (PX4 does not handle/return this value).

So a number of questions:
1. How is ArduPilot able to recover from invalid sequence? (what if the missing element is never sent?) Or is the assumption that the source is just "wrong" and this is an accident not an incorrect mission.
2. Should this be treated as a warning? 
3. Are any of the other enums that are actually warnings and not errors?
4. What do/can GCS do with the information?

My take is that either:
1. They are all treated as errors, and we document that.
2. They are treated by default as errors, and we document in the exceptions (warnings).

WickedShell, 2018-07-17:
1. You can get `INVALID_SEQUENCE` whenever you have a link break down, particularly if you are batching commands to it. The autopilot simply waits for the appropriate sequence. The GCS recovers by either sending the last requested ID it knew the autopilot wanted, or more generally it waits until the autopilot reprompts what waypoint should be sent.
2. `INVALID_SEQUENCE` is probably the only one that is properly a warning.
3. It should be noted that you can actually resend any of the commands that fail (on ArduPilot) until an overall transaction timeout hits (8 seconds for some reason). This can actually be useful to recover from certain situations. For example you might get a `MAV_MISSION_UNSUPPORTED_FRAME` when you sent a `MAV_FRAME_GLOBAL_RELATIVE_ALT_INT`. You could try resending the same command with `MAV_FRAME_GLOBAL_RELATIVE_ALT`. The `ALT_INT` frames were a bad idea. In some places the autopilot will accept it, in some it wont. Given that the frame is dictated by the `MISSION_ITEM` vs `MISSION_ITEM_INT` encoding I just don't see the point.
4. Usually used to create a format string that is more intelligible to the user an presented as a dialog stating which waypoint failed and why. The exsisting descriptions aren't appropriate/useful for directly emitting in a UI.

peterbarker, 2018-07-17:
1. ArduPilot will continue to expect the correct mission item to be uploaded if it returns an invalid-sequence response.  Good luck with that - but it is essentially a warning.

2. You could treat this as a warning as far as AP is concerned.

3. AP only seems to send ACCEPTED, ERROR or INVALID_SEQUENCE

4. Probably throw a warning and wait for the autopilot to re-request the waypoint it actually wants.

WickedShell, 2018-07-17:
> 3. AP only seems to send ACCEPTED, ERROR or INVALID_SEQUENCE

I think you missed some, we send INVALID_PARM1-7, and we send MAV_MISSION_UNSUPPORTED.

peterbarker, 2018-07-17:
@WickedShell has just pointed out several cases I missed.

I think generally running a whitelist of responses that can be ignored might be the way to go - and treat everything else as errors.  AFAICS, INVALID_SEQUENCE is the only one safe-to-ignore.

WickedShell, 2018-07-17:
@peterbarker what about the specific `ALT_INT` examples I brought up earlier? (We accept `ALT_INT` in some places, not in others (particularly missions vs on the fly commands)

hamishwillee, 2018-07-17:
Thank you both for this. 

> It should be noted that you can actually resend any of the commands that fail (on ArduPilot) until an overall transaction timeout hits (8 seconds for some reason). 

Based on this it looks to me like ArduPilot treats all messages as warnings. It essentially sends the response, resets to the state where it was before it received that message (or simply doesn't update it). Either it receives the correct update and continues OR does not receive the update, and then fails on the overall transaction timeout.

Or are there errors cases where a message will always kill the transaction without waiting.

PX4 by contrast treats anything it sends back as an error - ie it ends the current communication and returns to the idle state. However it does not detect or return INVALID_SEQUENCE (possibly a bug). 

5. Does that sound about right?

WickedShell, 2018-07-17:
> or simply doesn't update it
That's correct.

Killing the transaction immediately is not a great approach IMO as the communication streams are far to lossy to simply kill it immediately.

hamishwillee, 2018-07-17:
> Killing the transaction immediately is not a great approach IMO as the communication streams are far to lossy to simply kill it immediately.

6. Doesn't the MAVLink protocol take care of the packet loss using the frame sequence number? Ie Isn't an invalid sequence due to dropped packets handled before it gets to this protocol? [That's just an assumption by the way!]
If so, then the only reason to get an invalid sequence is that an invalid sequence was sent.

I'm not too concerned about ability to resend with new MAV_MISSION_UNSUPPORTED_FRAME discussed above. That is a side issue for this discussion.

WickedShell, 2018-07-17:
6. It depends. The sequence number is the intended way to tell about it, but if you use batching (allows significantly faster transfers) then sequence failures become much more probable. There is also a fundamental race condition on sequence numbers you can hit regardless of that. Consider the following situation:
  - The vehicle requests mission item 5 be uploaded.
  - The GCS is delayed on responding to the request (do to link delay, GCS processing anything)
  - The GCS sends item 5 towards the vehicle
  - The vehicle rerequests item 5 because it hasn't received the timeout yet
  - The vehicle parses item 5, requests item 6
  - The GCS sends the item 5 request
  - The vehicle sends `INVALID_SEQUENCE` as it wanted item 6 not item 5.
In general if you kill a transaction the moment sequences become wrong you have significant issues, as our transports are not reliable, and order is not strictly guaranteed.