mavlink/mavlink #972: Error generating C source files from XML (possibly broken/untested commits)

samacumen, 2018-08-06:
_I identified an issue committed recently (May 17th) which clearly breaks the usual procedure of generating C files using **mavgenerate.py** tool under mavlink directory._

You can reproduce this by:

1. Clone the latest mavlink repository.
2. Go to mavlink directory and do "python mavgenerate.py &"
3. Enter the common.xml directory  path and destination path for "C", "v2.0" header generation.
4. Check on "Validate" checkbox ("Validate Units" DOES NOT work with latest commits)
5. Press on button "Generate"
6. Notice that the _tool crashes_ with no generated headers; produces error messages.

The commit "**3fb5b02fcbe83a41d8327a3401affd71b601f4f4**" introduced some **<deprecated ../> tags** in Common.xml and these tags are introduced in changes for ardupilotmega.xml in "**fbe9cfa688a4f9941876dbf0610b5777ef1a2d27**", etc.

Mavgenerate.py tool does not like the introduction of tag <deprecated../> as it doesn't seem to support it.

You can use mavgenerate.py tool for generate C files using Common.xml with any other previous commit and it works just fine.

`Can someone please look into this?
`

amilcarlucas, 2018-08-07:
Please update the pymavlink submodule

git submodule update

It should fix all your issues.

samacumen, 2018-08-07:
Thanks @amilcarlucas :+1: 

`git submodule update --init --recursive ` on the 'mavlink' directory helped! Looks like all is well now. I will go ahead and close this issue.

closed 2018-08-07 by samacumen