mavlink/mavlink #973: Ideas for working out what messages are supported

hamishwillee, 2018-08-08:
Finding out whether a particular message works as documented on a particular system is hard:
- MAVLink support for determining what features are supported by an autopilot are very course-grained and at protocol level [MAV_PROTOCOL_CAPABILITY](https://mavlink.io/en/messages/common.html#MAV_PROTOCOL_CAPABILITY). 
- Many messages are not implemented.
- Many messages are partially implemented, or only implement some of the fields (particularly in missions)
- It is a common request "does this command work on autopilot X", "does this command work on autopilot X in mode Y".
- Often finding the answers to above is hard even by code inspection - message support or "detailed support" depends on vehicle type, with behaviour being overridden.
- Sometimes messages are deprecated, but support has not yet been added in a system - it would be useful to know that a message is/isn't supported.

Some ideas:
1. Allow XML tag to be added to any message, enum or field specifying system does *not support* feature: `<notsupported enum="MAV_AUTOPILOT" />`
   - Not sure enum works, because perhaps support is at a vehicle type level as well as autopilot
   - Idea is that not supported implies "does not work"
2. Is there any way to write a simple compatibility suite that queries individual messages and can tell if support exists based on response? We might then be able to query support across a range of systems and vehicle types, and generate reports in that way.

Others?

I like the idea of not-supported-by.... and one could use the dialect enums to specify the autopilot type.

WickedShell, 2018-08-08:
> 2. Is there any way to write a simple compatibility suite that queries individual messages and can tell if support exists based on response? We might then be able to query support across a range of systems and vehicle types, and generate reports in that way.

Anything that is streamable you should be able to test for via setting/testing the message interval on it. It's all the other commands that's an issue. (And this won't allow you to rev the message/test what version is supported)

hamishwillee, 2018-12-12:
Closing. This is probably best handled by test suite.

closed 2018-12-12 by hamishwillee