mavlink/mavlink #979: Lat lon field for sim_state should be int32_t

khancyr, 2018-08-21:
see https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L3851

llschloesser, 2022-11-01:
@khancyr, why was this closed? We would like to see this fixed. Switching to int32_t DegE7 would increase accuracy 100x, right? From roughly 1.1m to 0.01m.

khancyr, 2022-11-01:
No idea...
If you are using ArduPilot then we have 
Simstate message that got the position as uint32.

Otherwise this message will need an extension to add the uint lat lon version

hamishwillee, 2022-11-02:
@llschloesser What flight stack are you using?

Just to clarify, I think you are asking to change the lat, lon fields on [SIM_STATE](https://mavlink.io/en/messages/common.html#SIM_STATE) in common.xml? What @khancyr is saying is that the ArduPilot dialect uses (emits?) the [SIMSTATE](https://mavlink.io/en/messages/ardupilotmega.html#SIMSTATE) message which has the fields as `int32`,

The fields should be in `int32` in the "common" message (`SIM_STATE`), but we can't just change them without breaking existing deployments. Options:
- Add lat/lon values as `int32` extensions.
- Port the ArduPilot message across to common (if it is an acceptable replacement) and deprecate the current version.
- Port the ArduPilot message across to common, extending it if needed,
- completely new message
- Live with it.

One of the first three is most likely, all have challenges. Whether we do this depends on what flight stack you're working on @llschloesser - because there is no point making changes unless someone is interested in implementing them.

jasonir129, 2022-11-02:
> @llschloesser What flight stack are you using?
>
> Just to clarify, I think you are asking to change the lat, lon fields on [SIM_STATE](https://mavlink.io/en/messages/common.html#SIM_STATE) in [...]

I work with @llschloesser , we're using a modified version of ardupilot that is sending both SIMSTATE and SIM_STATE.  We don't like SIMSTATE because it doesn't have altitude for some odd reason, so we're in this predicament.  I guess we could use lat/lon from SIMSTATE and altitude from SIM_STATE, but we should just fix this and do it the right way for the future.  I don't know why there's an effort to minimize packet size for something like SIM_STATE when it's generally for running locally on a full sized machine rather than pixhawk etc. where we might be more resource constrained.

hamishwillee, 2022-11-03:
@julianoes @auturgy There are two version of the sim state message 
[SIM_STATE](https://mavlink.io/en/messages/common.html#SIM_STATE)) (common.xml) and [SIMSTATE](https://mavlink.io/en/messages/ardupilotmega.html#SIMSTATE)  (ardupilotmega.xml). ArduPilot supports both but it would be better if there could just be one in common.xml. 

Unfortunately the common version uses float rather than `int32` for lat/lon.
It is otherwise a superset of the ArduPilot version with additional fields: q1 to q4, std_dev_horz, std_dev_vert, vn, ve, vd

I am assuming that the additional fields in the common SIM_STATE are actually useful, otherwise we could just move the ardupilotmega version across and deprecate the one in common?

IMO next best option is to add extension fields for the lat/lon in int32 format. Normally I'd set fields to 0 or NaN to indicate "not sent" but perhaps we could suggest both values be sent and let the recipient decide based on whether it gets non-zero values in this field? That way a sender does not have to update to be useful to a recipient still?

- `latitude` (int32): degE7: Latitude. If not supported this should latch to 0.
- `longitude` (int32): degE7: Longitude. If not supported this should latch to 0.

Another option is to create `SIM_STATE_V2` with the lat/lon changed. That creates a path to allow us to eventually create a cleaner final message.

Thoughts?

khancyr, 2022-11-03:
[SIMSTATE](https://mavlink.io/en/messages/ardupilotmega.html#SIMSTATE) is definitily not the way to got for simulation state report, SIM_STATE is much more uselfull.
@peterbarker @tridge any opinon on @hamishwillee proposal ?

peterbarker, 2022-11-03:
I actually had to revert us changing to use `SIMSTATE` for our autotest - but I definitely did want us to go that way:

```
commit 540a56adb8593148c1fdf4f48e6a2c10af512c8a
Author: Peter Barker <pbarker@barker.dropbear.id.au>
Date:   Mon Jun 28 13:33:34 2021 +1000

    autotest: use SIM_STATE (common.xml) instead of SIMSTATE (ardupilotmega.xml)
    
    It is a useful superset
```

I like the idea of a `SIM_STATE_V2` using integer lat/lng.

peterbarker, 2022-11-03:
I actually had to revert us changing to use `SIMSTATE` for our autotest - but I definitely did want us to go that way:

```
commit 540a56adb8593148c1fdf4f48e6a2c10af512c8a
Author: Peter Barker <pbarker@barker.dropbear.id.au>
Date:   Mon Jun 28 13:33:34 2021 +1000

    autotest: use SIM_STATE (common.xml) instead of SIMSTATE (ardupilotmega.xml)
    
    It is a useful superset
```

I like the idea of a `SIM_STATE_V2` using integer lat/lng, but the extensions are probably a better idea.  Maybe after we've tacked on a another couple of extensions before coming up with a new message.  THe alternate semantics above - if new fields are zero then use original fields - seems good to me.

peterbarker, 2022-11-08:
We discussed this at the DevCall today.

tridge is happy either way - either add lat_int/lng_int to SIM_STATE or all of the extra fields to SIMSTATE.

Does anybody have any preferences between the two?

hamishwillee, 2022-11-08:
@julianoes Do you have an opinion on this ^^^

The benefit of adding the fields to SIM_STATE is that nothing breaks for stuff that is already reliant on it - there is literally no work needed unless you want to do something. The down side of this is that it might be confusing to users as to why we have both fields.

The benefit of adding the values to SIMSTATE is that we have a cleaner API that is closer to where we eventually want to be, less confusing message, slightly slighter weight. But  means a change for any existing deployments of common.xml.

I think SIM_STATE is likely the least churn/easiest to manage if people are already emitting it.

hamishwillee, 2022-11-08:
I'll discuss in the mav call and see if we can get some consensus.

jasonir129, 2022-11-09:
I just wanted to add that we’re really appreciative how receptive you guys have been to improving this message! We use arduplane extensively and it’s great to see how active and open to feedback/criticism the community is. Looking forward to hearing the outcome from the mav call.

hamishwillee, 2022-11-10:
OK, this is what we ended up with: https://github.com/mavlink/mavlink/pull/1916

hamishwillee, 2022-11-16:
All merged. Happy days :-)

closed 2022-11-16 by hamishwillee