mavlink/mavlink #981: RC_CHANNELS_OVERRIDE - does it meet required use cases

hamishwillee, 2018-08-30:
Cross posted from https://github.com/mavlink/mavlink/pull/970

hamishwillee, 2018-08-30:
The conversation so far ...

## @WickedShell 

A slight tangent I'd like to discuss as well, there are 3 states I'd like to be able to do with the RC_CHANNELS_OVERRIDE message per channel.

1. Set the channel to the override value
2. Ignore the RC_CHANNELS_OVERRIDE message and don't update the value on the channel
3. Return control to the RC receiver physically connected

1 is what everyone does with the message already, and per the spec number 2 can be achieved by sending UINT16_MAX. I'd loosely like to propose that the value of 0 means return control to the RC receiver, but I haven't given it a lot of thought, and it means that the upper 8 channels would always be under RC control unless you were sending it as MAVLink 2 with the extensions.

## @MaEtUgR 

@WickedShell Don't you automatically take the RC receivers values if the override doesn't provide any? Ah I see it's per channel and you want to mix controls up between override and not? You would like to do that to the channels 1-8 as well then right?

In our use case we don't use 1000-2000 values but rather 0-4095... so zero could be a value and we have to be careful with the channels 9-18.

Would it be possible to use something like (UINT16_MAX - 1) to mix in the receivers value for that specific channel?

**EDIT:** Isn't `A value of 0 means control of that channel should be released back to the RC radio.` already what you propose?

## @julianoes 

I made this change initially:
https://github.com/mavlink/mavlink/commit/101cca833dc0727f38b455d1e61c893b1b9cf4a8

I said it was for compatibility with mavlink 1 but I'm not sure about what that exactly means anymore.

What is for sure is that we can save bandwidth by not having to set all channels to `UINT16_MAX` but can let them 0 and have them trimmed.

@WickedShell I'm trying to think through your 3 cases:
1. Naturally, makes sense.
2. Do you ever mix and match RC channels with `RC_CHANNELS_OVERRIDE`? I haven't considered that.
3. What about just stopping to send the message at that point.

julianoes, 2020-04-23:
Looks like this was addressed in #970.

closed 2020-04-23 by julianoes