mavlink/mavlink #982: Typical connection workflow

michprev, 2018-09-02:
Hi,
I am new to mavlink. I would like to implement it on my quad but I am having quite a hard time understanding how even connection is estabilished.

From dev guide I have read I would expect it to be https://mavlink.io/en/guide/mavlink_version.html#version_handshaking

But after running DroneCode take off example (https://sdk.dronecode.org/en/examples/takeoff_and_land.html) it seems to me that GCS waits to receive the first message from the drone (which does not make much sense to me).

Is there any standard workflow how to connect a drone / any UAV?

I have also read that HEARTBEAT is very often sent at 1 Hz. I know that this is not protocol specific and anyone can implement it by their needs but I believe it might be a good idea to place recommendations / typical usage somewhere on mavlink guide webpage.

hamishwillee, 2018-09-02:
@michprev You're quite right - the docs do not cover this well. That is my job to fix. Currently I am working more on the deep detail for generator implementers and people wanting to create new messages. This is one of the things that I hope to work on in the coming weeks.

So the simple answer is that I believe I understand how this works, but I have not yet verified with test code. Here goes:
1. MAVLink has a number of default ports that are listened connected to by default ([PX4 info here, but other systems same](https://dev.px4.io/en/simulation/#default-px4-mavlink-udp-ports)
1. Specifically 
   - autopilots broadcast a HEARTBEAT on 14540 to communicate with offboard APIs. Offboard APIs are expected to listen for connections on this port.
   - autopilots broadcast a HEARTBEAT on port 14550 for communication with ground control stations. GCS are expected to listen for connections on this port. QGroundControl listens to this port by default.
1. So yes, GCS and Developer APIs (e.g. dronekit, SDK) listen for the autopilot hearbeat. Receiving the heartbeat on a regular basis is what counts as a connection. The old docs on connections state that HEARTBEAT should (not "must") be sent at least once every 10seconds, but PX4 and ArduPilot both send at 1Hz. I suspect the rate depends on the channel - a high latency connection might not send it at all. 
1. Autopilots also have a listener port but this does not appear to be standardised (not yet confirmed by me!). So it is possible to initiate a connection from a developer API or (in theory) a GCS. This is what ROS scripts do but as far as I know, no GCS do.
1. Check out the examples - this one covers UDP: https://mavlink.io/en/examples/c_udp.html . Note that this does not actually "connect" got QGroundControl but does communicate with it. I have not yet debugged this, but I believe that for QGC to recognise something as an autopilot and connect to it the other end needs to implement the parameter protocol.  There is another example showing serial connection - I believe semantics are the same - either end can initiate connection, but GCS typically waits.
1. https://mavlink.io/en/guide/mavlink_version.html#version_handshaking is about working out if you have MAVLInk 1 or 2. 
1. Yes, absolutely to ... 
   > I have also read that HEARTBEAT is very often sent at 1 Hz. I know that this is not protocol specific and anyone can implement it by their needs but I believe it might be a good idea to place recommendations / typical usage somewhere on mavlink guide webpage.

   As I said, this is one of the next docs for me to do. I will need to get those examples working perfectly first. I plan to add the information above (verified and complete) as the first item in the protocols section: https://mavlink.io/en/protocol/overview.html

There is a simpler guide, that you may also find of interest here: https://discuss.ardupilot.org/t/mavlink-step-by-step/9629

Hope that helps get you started!

.

hamishwillee, 2018-09-02:
PS Every node should broadcast heartbeat, so developer API and GCS also broadcast.

hamishwillee, 2018-09-02:
Also, I assume you are writing your own flight stack? Sorry, should have asked that to start with!

michprev, 2018-09-04:
@hamishwillee thank you very much for such a comprehensive reply.

> Also, I assume you are writing your own flight stack? Sorry, should have asked that to start with!

Yes! It is just a small hobby project. It is based on dual-core ESP32 microcontroller with Wi-Fi (without RC receiver) and MAVLink seems like a perfect option for it.

hamishwillee, 2018-09-04:
That seems pretty cool! 
Once you get past the basics you'll want to implement some of the sub-protocols ("microservices"). There is some documentation in the library, but often looking at the existing implementations in ArduPilot, QGC and PX4 will be necessary to understand the detail. 

One thing to be aware of is that because the specification is not tightly enough written, there are some areas where the protocol implementations are different in the main flight stacks. Specifically ArduPilot and PX4 have slightly different behaviour in terms of mission handling and significant differences in terms of parameter handling.

One thing you can do as a form of validation is use DroneKit or Dronecode SDK to test your implementation - if these work you've matched an existing implementation :-)

I'll close this now.

hamishwillee, 2020-04-23:
Closing. The question was answered with links above in docs.

closed 2020-04-23 by hamishwillee