mavlink/mavlink #987: Mavlink protocol level

Huginn87, 2018-09-18:
Good morning. I wanted to ask, kindly, what kind of protocol was mavlink if on a physical level, data-link level or network level. Thank you so much in advance for your kind reply

hamishwillee, 2018-09-19:
Not physical layer, but it impacts across the other layers. 

The lower level definition for MAVLink is the [over the wire serialization format](https://mavlink.io/en/guide/serialization.html) - ie what packets look like. The packets include source ids and may include destination ids, but the actual routing behaviour is a higher level concept. 

In addition to message format the protocol defines a number of [sub protocols/microservices](https://mavlink.io/en/protocol/overview.html). These define specific types of messages, their sequence, packet resending etc - and so are a high level protocol definition.

Hope that helps.

Huginn87, 2018-09-19:
If I understand correctly it should be similar to the MQTT protocol. So, in addition to staying above the other levels, it has its own routing technique that associates the id of the component with the physical address of the component itself, right?

hamishwillee, 2018-09-19:
I don't know the MQTT protocol. 

>  it has its own routing technique that associates the id of the component with the physical address of the component itself, right?

Not quite. Consider the case where multiple GCS and autopilots are on a wifi network. Autopilots broadcast to known UDP ports. All messages on this network go to all autopilots/GCS. However if the message has a target id that does not match the current system then it will ignore the message. 

The routing rules cover how messages will cross interfaces. So consider the case in the above where one of the autopilots connects to a companion computer via a serial link. In this case the autopilot would be expected to route broadcast messages to the companion computer. It would also be expected to route the messages that are directed at that companion computer. Improved docs on routing are in progress - see this PR: https://github.com/mavlink/mavlink-devguide/pull/113