mavlink/mavlink #993: Global GPS position

RohanKurane, 2018-09-24:
Hi,

I want to get the global gps position of the UAV using mavlink in a loop. 

But when I tried to read the messages coming from the PX4 flight controller, I did not see this data returned all the time (only after certain intervals). Is there a way to get this data at a constant rate or continuously ? 
@zehortigoza @LorenzMeier

zehortigoza, 2018-09-24:
GPS modules usually works at 5Hz so you will get one message per 200ms, is that what are you getting?

RohanKurane, 2018-09-24:
I will take a look at the exact interval.

RohanKurane, 2018-09-27:
Hi @zehortigoza 

If I want the latitude , longitude and altitude for UAV, which mavlink message should I be looking for ? 

Should I look for
MAVLINK_MSG_ID_GPS_RAW_INT  (24)
or
MAVLINK_MSG_ID_GLOBAL_POSITION_INT (33)
or 
some other message ? 

I ran Mavlink in a loop for 3 mins and did not see the MAVLINK_MSG_ID_GLOBAL_POSITION_INT message returned. 

I did see the MAVLINK_MSG_ID_GPS_RAW_INT message returned once in a while. 

P.S: I was running this indoors in my apartment. Would that be the reason I did not see MAVLINK_MSG_ID_GLOBAL_POSITION_INT returned ?

RohanKurane, 2018-09-27:
Its possible I need the MAVLINK_MSG_ID_GPS_RAW_INT (24) message from MAVLINK for this. 

Can you confirm ?

Gps Latitude: 
Gps Longitude: 
Gps Altitude: 
Gps HDOP (horizontal dilution of position): 
Gps VDOP (vertical dilution of position): 
Gps Ground Speed: 
Satellites visible: 

Also, I tried reading a bit online about what HDOP and VDOP are. Are they the linear and circular precision errors for the GPS coordinates ?

hamishwillee, 2018-09-27:
I'm looking at ArduPilot in simulator and seeing GLOBAL_POSITION_INT at 4Hz.

RohanKurane, 2018-09-27:
Hi @hamishwillee 

Could you help answer this question ? 

f I want the latitude , longitude and altitude for UAV, which mavlink message should I be looking for ?

Should I look for
MAVLINK_MSG_ID_GPS_RAW_INT (24)
or
MAVLINK_MSG_ID_GLOBAL_POSITION_INT (33)
or
some other message ?

When I ran a loop to scan for MAVLINK messages, I did not see GLOBAL_POSITION_INT show up but I did see GPS_RAW_INT every 200ms or so show up.

hamishwillee, 2018-09-27:
I would use GLOBAL_POSITION_INT for lat, lon, alt. It is filtered and a smaller packet. If you specifically need to look at GPS errors then consider the RAW int - most people don't.

What are you ultimately trying to do?

hamishwillee, 2018-09-27:
I assume you are working on PX4? Try the mavlink inspector widget I link above. If you get different results from that, then your code is probably wrong. If they are the same, then we can look further into this - probably moving to the PX4 discussion boards.

RohanKurane, 2018-09-27:
@hamishwillee  Yes I am using PX4. 

I can try the widget.

I basically want to locate the GPS coordinates of my UAV and then send it to a map to show the UAV on it based on its coordinates. 
I am not very familiar with the values they both return, but is GLOBAL_POSITION_INT more accurate than GPS_RAW_INT for UAV position ? Is that why you suggest using it ?  The size of the packet is not an issue for my application.

hamishwillee, 2018-09-27:
I don't know the practical implications of using the RAW value. I am suggesting using the other value because it is delivered at a higher rate and I "assume" that if filtering wasn't useful, we'd just supply the raw value. 

Have you considered using the SDK: https://sdk.dronecode.org/en/ ?

It will use the right message so you don't need to think about it.

hamishwillee, 2018-09-27:
And if you don't want to use the SDK, look at its source, and you can see what it is using!

RohanKurane, 2018-09-27:
Great. Thanks. I'll take a look at the SDK source. 

Also, the rate of delivery is not something I am considering at this moment.

RohanKurane, 2018-09-27:
@hamishwillee 
Looks like the SDK uses GLOBAL_POSITION_INT

Telemetry::Result TelemetryImpl::set_rate_position(double rate_hz)
{
    _position_rate_hz = rate_hz;
    double max_rate_hz = std::max(_position_rate_hz, _ground_speed_ned_rate_hz);

    return telemetry_result_from_command_result(
        _parent->set_msg_rate(MAVLINK_MSG_ID_GLOBAL_POSITION_INT, max_rate_hz));
}

set_rate_position has this - 
        const Telemetry::Position &position = telemetry->position();
        std::cout << "Position: " << std::endl
                  << "Absolute altitude: " << position.absolute_altitude_m << " m" << std::endl
                  << "Relative altitude: " << position.relative_altitude_m << " m" << std::endl
                  << "Latitude: " << position.latitude_deg << " deg"
                  << ", longitude: " << position.longitude_deg << " deg" << std::endl;

To see why I am not receiving the GLOBAL_POSITION_INT message you suggest I use the widget ? 

I am also trying this indoors. Could that be a reason I don't see the GLOABL_POSITION_INT message ?  

With 50Hz rate I should see this message every 20ms or so I would imagine. I am not sure if anything is wrong with the code since I see a bunch of other messages during the 3min loop I run to scan for mavlink messages.

RohanKurane, 2018-09-27:
Yes I do plan to use this outside.

RohanKurane, 2018-09-27:
@hamishwillee @zehortigoza 

I tried this outside today and still did not see the GLOBAL_POSITION_INT message returned. 

I saw a bunch of other mavlink messages returned. 

Do I need to calibrate the GPS in order to see the GLOBAL_POSITION_INT message ? 

P.S: The UAV is not flying. Just sitting on the ground.

RohanKurane, 2018-09-27:
Here is the relevant code - 

                            n = recvfrom(tcp_fd, buf, sizeof(buf), 0, (struct sockaddr *)&sockaddr, &addrlen);
                                             if (n == -1)
                                             {
                                                printf("Error reading socket %d\n", errno);                                                
                                                return false;                                                
                                             }
                                             // decode mavlink messages  in a loop
                                             loop:
                                             {
                                                for (i = 0; i < n; i++)
                                                {
                                                    if (mavlink_parse_char(MAVLINK_COMM_0, buf[i], &msg, &status))  
                                                    {
                                                      if (msg.msgid == MAVLINK_MSG_ID_GLOBAL_POSITION_INT )
                                                      {
                                                       mavlink_global_position_int_t global = {};
                                                       mavlink_msg_global_position_int_decode(&msg, &global);
                                                       printf("latitude: %d:\n", global.lat);
                                                       printf("longitude: %d:\n", global.lon);
                                                       printf("altitude: %d:\n", global.alt);
                                                       printf("relative altitude: %d:\n", global.relative_alt);
                                                       printf("Ground XSpeed: %d:\n", global.vx);
                                                       printf("Ground YSpeed: %d:\n", global.vy);
                                                       printf("Ground ZSpeed: %d:\n", global.vz);
                                                       printf("--------------\n");
                                                     }

zehortigoza, 2018-09-27:
You need to start search for the information by yourself, a quick search you would know what is the difference between  MAVLINK_MSG_ID_GPS_RAW_INT and MAVLINK_MSG_ID_GLOBAL_POSITION_INT. Are you not getting the message? Why not check in the flight stack code? What is the conditions to flight stack send that message?

RohanKurane, 2018-09-27:
According to @hamishwillee  and the Dronecode SDK source code I need the MAVLINK_MSG_ID_GLOBAL_POSITION_INT message. 
I do see the  MAVLINK_MSG_ID_GPS_RAW_INT message returned. 

My question is - Do I need to have the GPS calibrated to get the MAVLINK_MSG_ID_GLOBAL_POSITION_INT message ? 

I'll look in the PX4 flight stack code to see how it sends the MAVLINK_MSG_ID_GLOBAL_POSITION_INT message

RohanKurane, 2018-09-27:
Looking in the code, there is nothing that shows any conditions (Except for position subscription which I am not sure how to set in my code). A callback is set to get the global position whenever it is available at the specified rate. So unless I am missing something in the open source code, I should be getting the message. 
I'll check the QGC logs to see if there is anything there

hamishwillee, 2018-09-27:
> My question is - Do I need to have the GPS calibrated 

GPS is plug'n'play. As long as it is working you should be getting messages. 

I don't really want to think about your code until you have looked at mavlink inspector or logs and confirmed that the messages really aren't being sent.

RohanKurane, 2018-09-27:
I am going to look at the widget now and will update you

RohanKurane, 2018-09-27:
I have attached a screen shot of the mavlink inspector widget. You can see that I am not getting the GLOBAL_POSITION_INT message. I do see the GPS_RAW_INT message as I mentioned yesterday amongst few others. 
[image]

hamishwillee, 2018-09-27:
@RohanKurane So for now you can work with GPS_RAW_INT - if your code is also getting it at 5Hz you know your code is OK. 

The fact that you are not getting GLOBAL_POSITION_INT is concerning. As this is a PX4 specific query I suggest you post on http://discuss.px4.io/ with the above information and ask for debugging suggestions. YOu can also cross-post here. 

IMO this is no longer a "MAVLink" issue, so this item should be closed.

RohanKurane, 2018-09-27:
Thanks @hamishwillee  for your help. 

I'll post an issue on PX4 about the GLOBAL_POSITION_INT. 
And use GPS_RAW_INT for now.

closed 2018-09-27 by RohanKurane

RohanKurane, 2018-09-27:
@hamishwillee 
I opened 
http://discuss.px4.io/t/px4-fc-does-not-return-mavlink-message-for-global-position-int/7940

I dont know how to cross-post here.

hamishwillee, 2018-09-27:
What you have done is what I mean by cross post - a link to the issue from here, and a link to here from the issue :-)

RohanKurane, 2018-09-27:
oh great