mavlink/mavlink #1000: Landing_target message?

alexx600, 2018-10-07:
Hello, i have got a companion computer connected to the autopilot. Odroid is equipped with digital camera and pattern reckogniction algorithm. I want to send message https://mavlink.io/en/messages/common.html#LANDING_TARGET . Algorithm gives me offset X and Y relative to image center. I do not understand why there is angle X angle Y and dostanie values? Why there is no standard polar coorfinates? How i can transform my parameter to message definition variables?

Maybe this id more complicaged situation? Maybe two angled are necessery to desrcibe unit Vector on 3D space( X Y and altitude )?how does It works? I shoukd use some Euler angles? How Axis are located? IT is right handed system?

hamishwillee, 2018-10-10:
Hi @alexx600,

On PX4 the angle/distance parameters aren't used - the system ONLY accepts the x,y,z values in the `MAV_FRAME_LOCAL_NED` frame (you can see this [here](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L2222))

The x,y,z params are "new" in MAVLink 2, which implies that in MAVLink 1 the angle/distance were sent to the vehicle and the vehicle used them to work out how to move.

> How i can transform my parameter to message definition variables?

I don't understand what you mean?

@bkueng Who would be good person to talk to about this message? Specifically how the angle_x, angle_y, distance, size_x, size_y are supposed to be used? 
(My trigonometry is a bit rusty, but I would guess that with the two angles and the distance to the target you can then work out the local x and y distance to the target ... but then:
- the angles are in the horizontal plane and the distance presumably includes a vertical to-ground component
- I don't see how size_x, size_y would be in radians. )

bkueng, 2018-10-10:
> @bkueng Who would be good person to talk to about this message?

Maybe @mhkabir or @okalachev ?

TSC21, 2018-10-10:
Those angles are the offset relative to the camera center and they are computed in radians. To get cartesian coordinates, you need the FOV and the distance from the camera to the target.
Now it's not clear if @alexx600 is using APM or PX4. The autopilot firmwares support this message differently.

TSC21, 2018-10-10:
Just bumped into this: https://discuss.ardupilot.org/t/landing-target-messge/33772. So it's already answered and I guess we can close this issue.

hamishwillee, 2018-10-11:
Thanks @TSC21 . I'm more concerned about whether we can improve [LANDING_TARGET](https://mavlink.io/en/messages/common.html#LANDING_TARGET) so that we don't have to answer this again. A user shouldn't have to know the algorithm but they should be able to work out what the parameters actually are. 

If you look at the screenshot of my "image" are angle_x, angle_y as shown? How would you represent/draw size_x, size_y - basically I don't understand what an "x" or y means in radians.
I assume the distance is the real distance from the camera to the target - is this from a rangefinder?

[image]

alexx600, 2018-10-11:
Hi @hamishwillee 
> Hi @alexx600,
> 
> On PX4 the angle/distance parameters aren't used - the system ONLY accepts the x,y,z values in the `MAV_FRAME_LOCAL_NED` frame (you can see this [here](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L2222))
> 
Thank You very much for response. I didn't know that PX4 accept only `x`,`y` not `angle_x `and `angle_y`.
I have tested it on ArduCopter SITL. Unfortunatelly i am using pymavlink, message there doesn't support `x`,`y` only `angle x` and `angle y `
`def landing_target_encode(self, time_usec, target_num, frame, angle_x, angle_y, distance, size_x, size_y`

@TSC21
>Just bumped into this: https://discuss.ardupilot.org/t/landing-target-messge/33772. So it's already >answered and I guess we can close this issue.

This is my topic. I was wonder how to change image x and y to angle_x and angle_y. I didn't know about fov and formula.

@alexx600 hamishwillee 
>If you look at the screenshot of my "image" are angle_x, angle_y as shown? 
If the data was represented in this way, only one `angle` would be enought

ndepal, 2018-10-11:
> If you look at the screenshot of my "image" are angle_x, angle_y as shown?

This is my understanding: Imagine a ray pointing from the camera's principal point to the target. The x angle is the angle spanned by that ray and the x axis of the image plane. The same holds for the y angle respectively. I.e. the x angle is a function of the x pixel coordinate of the target (denoted u̅ in the image below), the y angle is a function of the y pixel coordinate (denoted v̅ in the image below) . Using the angle rather than u̅/v̅ pixel coordinates has the advantage that the effect of the camera lens is already accounted for. Otherwise the receiver of the message would need to know the camera FOV etc.

[image]

The sizes in x and y direction are analogous. They describe the angle between the smallest and biggest pixel in x/y direction respectively of the target as seen in the image.

> To get cartesian coordinates, you need the FOV, the image size

@TSC21 I believe this to be incorrect.
> To get cartesian coordinates, you need ... the distance from the camera to the target

agreed

TSC21, 2018-10-11:
@ndepal https://github.com/goodrobots/vision_landing/blob/master/src/track_targets.cpp#L637

ndepal, 2018-10-11:
@TSC21 the code you link calculates the offset in radians from pixel measurements, right? For that you definitely need the FOV (as I stated in my previous comment). To get from radians to cartesians you do not need the FOV anymore.

hamishwillee, 2018-10-11:
@alexx600 A point of clarification:

> I have tested it on ArduCopter SITL. Unfortunately i am using pymavlink, message there doesn't support x,y only angle x and angle y
   ```
   def landing_target_encode(self, time_usec, target_num, frame, angle_x, angle_y, distance, size_x, size_y
   ```

The message above is for MAVLink 1 - you should be using mavlink 2. If you look at the [message definition](https://mavlink.io/en/messages/common.html#LANDING_TARGET) you can see the [message extensions](https://mavlink.io/en/guide/mavlink_2.html#message_extensions) added for mavlink 2 are shown in blue.

If you do, then the message signature is as shown below (includes x, y, z). Assuming you are using mavutil to set up the connection then you select the appropriate signature using environment variables.
```
def landing_target_encode(self, time_usec, target_num, frame, angle_x, angle_y, distance, size_x, size_y, x=0, y=0, z=0, q=0, type=0, position_valid=0):
```
Work in progress docs on using pymavlink [here](https://hamishwillee.gitbooks.io/ham_mavdevguide/content/v/pymavlink/en/mavgen_python/) include setting environment variables.

Noting of course that x, y, z will only work for PX4 and the angle_x... only work on ArduPilot.

hamishwillee, 2018-10-12:
@ndepal @TSC21 Thank you both very much.

We should capture this info forever more. I have added as a sub-protocol here: https://github.com/mavlink/mavlink-devguide/pull/122
Can you both review/answer the open questions there?