mavlink/mavlink #1027: RSSI in RC channels has not consistent across implementations.

hamishwillee, 2018-11-20:
Discussion originates here: https://github.com/mavlink/qgroundcontrol/issues/7013

There are a bunch of messages that define an `rssi` field. 
* [RC_CHANNELS](https://mavlink.io/en/messages/common.html#RC_CHANNELS)
* [RC_CHANNELS_RAW](https://mavlink.io/en/messages/common.html#RC_CHANNELS_RAW)
* [RC_CHANNELS_SCALED](https://mavlink.io/en/messages/common.html#RC_CHANNELS_SCALED)
* [RADIO_STATUS](https://mavlink.io/en/messages/common.html#RADIO_STATUS)

Spec defines field as:
* `rssi` - `uint8_t` - `%` - Receive signal strength indicator. Values: [0-100], 255: invalid/unknown.

It is (anecodotally) implemented in ArduPilot, minimOSD, MavlinktoFrsky and many other projects using 0-255 is the whole range (and presumably NaN for invalid).

Can/should we revert this to the implementation used by others? If not, we should add this to list of things to "version" going forward.

hamishwillee, 2018-11-20:
[RADIO_STATUS](https://mavlink.io/en/messages/common.html#RADIO_STATUS) has no documentation. [According to Gus](https://github.com/mavlink/qgroundcontrol/issues/7013#issuecomment-440119354), the QGC code for this for Sik Radio is [here](https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L1654-L1671)

...

RSSI is expressed in (negative) dBm (decibels relative to a milliwatt), where <= -120dBm is basically no signal and >= -50dBm is 100%.

In addition to the actual value, you need a way to tell you simply don't have it. On a 0 - 255 scale, do you consider 0 as no signal or no RSSI availability?

hamishwillee, 2018-11-20:
Options for discussion
1. Spec is reverted to old version, and we convince PX4 to move to this implementation
2. Spec stays the same, and we try encourage everyone to move to the new behaviour. What is cost/impact if they won't?

hamishwillee, 2018-12-14:
@LorenzMeier @DonLakeFlyer 

We discussed the `[RC_CHANNELS](https://mavlink.io/en/messages/common.html#RC_CHANNELS).rssi` (and friends) fields on the [Dev Call](https://github.com/mavlink/mavlink/wiki/20181212-Dev-Meeting).

The general feeling is that the current spec should be reverted from a percentage value:
> Receive signal strength indicator. Values: [0-100], 255: invalid/unknown.

TO 
> Values:  0-255 value, where 0 is invalid/unknown.

The reason are:
- This reflects the original can current implementation of many systems. The spec was (anecdotally) updated to the current format without consultation.
- The 0-255 value gives greater resolution. This does not matter for GCS users/humans, but it will to antenna tracker and similar systems.
- A GCS can infer a percentage from either format for display.
- Currently PX4 and ArduPilot do this differently, so by making this change we can reduce the specific code in QGC.

Thoughts? 
If you agree, I can make a PR to make this change in mavlink. Can @LorenzMeier and @DonLakeFlyer then co-ordinate the associated changes that might need to be made in QGC/PX4?

DonLakeFlyer, 2018-12-14:
> Can @LorenzMeier and @DonLakeFlyer then co-ordinate the associated changes that might need to be made in QGC/PX4?

I don't understand what would need to change in QGC? This is just a doc change isn't it.

hamishwillee, 2018-12-16:
> I don't understand what would need to change in QGC? This is just a doc change isn't it.

No. If we change this then PX4 will start sending full range RSSI values with values from 0-255 rather than % (0-100). This would make PX4 the same as ArduPilot. It is the reverse of the problem you fixed in https://github.com/mavlink/qgroundcontrol/issues/7013

DonLakeFlyer, 2018-12-17:
Oh, I see. That would mean that QGC will need version based decoding of this message. @dogmaphobic I'm can't remember how you implemented ArduPilot chnge. But it means it would need to be reversed and special handling on PX4 side instread based on version.

hamishwillee, 2018-12-17:
@DonLakeFlyer Yes, precisely. Initially this would be a docs change and the code change would need to be co-ordinated with Firmware update later.

@LorenzMeier Can you confirm we can/will make the change in Firmware? 

Note that if we use them, this will affect these types of messages: RC_CHANNELS, RC_CHANNELS_RAW, RC_CHANNELS_SCALED, RADIO_STATUS. 

The last one (RADIO_STATUS) is different definition of RSSI again, and I'm not sure we can change it because it is SIK radio: https://github.com/mavlink/mavlink/issues/1027#issuecomment-440123426
For that I'd just document it as SIK implements it. (OK)?

dogmaphobic, 2018-12-17:
> I'm can't remember how you implemented ArduPilot chnge.

Just a different code path when receiving/parsing the value. For Ardupilot, the code converts 0-255 to 0-100. 

I still think it should be 0-254 and 255 means No RSSI. If there is no RRSI, the control is not shown. If 0 means both, it's ambiguous. As is, the control is always there showing 0% even if you have no RSSI value.

hamishwillee, 2018-12-17:
@dogmaphobic Is RRSI above a typo?

I personally much prefer what you are proposing, but doesn't this effectively introduce a "third implementation" such that now PX4, ArduPilot, minimOSD, MavlinktoFrsky and many other projects will be broken for very high RSSI?
Will you be able to "live" with "0"?

@auturgy Could/would ArduPilot consider moving to using 255 for "not used" rather than 0 as it makes things much clearer for a GCS? Do we have contacts in miniumOSD, MavlinktoFrsky  that we could push this through to the broader community. Guess at the cost/impact of doing this without broad agreement?

dogmaphobic, 2018-12-17:
>@dogmaphobic Is RRSI above a typo?

Yup

> Will you be able to "live" with "0"?

It's mostly wishful thinking. I would not expect anybody to change anything.

hamishwillee, 2018-12-18:
@dogmaphobic One day we'll have decent versioning. At that point, we can make the change and deprecate the old version. Ha ha ha.

hamishwillee, 2018-12-18:
> QGC knows the firmware version. So the code should check that and do the right thing.

Yes, I know. I mean MAVLink will have decent versioning so  *anyone* can properly manage change.

tridge, 2018-12-19:
@hamishwillee changing doc to match behavior seems right to me
using full range for max resolution also makes sense

hamishwillee, 2018-12-19:
@tridge @auturgy Great - thanks. Do we have any connections in minimOSD, MavlinktoFrsky etc to get them to make the change if we set 255 to zero? 
ie It is IMO the right thing for MAVLink  to set 255 as indicator for disable. However if we can't get broad commitment for everyone to change this do we make the change anyway or continue as planned (make the message match what everyone currently does?).

hamishwillee, 2019-01-10:
History as per github. 
- Feb10 2012 - Commit that added the v1.0 message tree (very first in tree) is [this](https://github.com/mavlink/mavlink/commit/c894393440952d07bbcce38e2f42cca8fcbc98e1#diff-ceacc1e2390186650146a45be8fcb691) which created [this common.xml](https://github.com/mavlink/mavlink/blob/c894393440952d07bbcce38e2f42cca8fcbc98e1/message_definitions/v1.0/common.xml#L1052). It had
  - [RC_CHANNELS_SCALED - 34](https://github.com/mavlink/mavlink/blob/c894393440952d07bbcce38e2f42cca8fcbc98e1/message_definitions/v1.0/common.xml#L1052).rssi - as per this PR - 0:255 = 0:100%
  - [RC_CHANNELS_RAW - 35](https://github.com/mavlink/mavlink/blob/c894393440952d07bbcce38e2f42cca8fcbc98e1/message_definitions/v1.0/common.xml#L1066).rssi - as per this PR
  - *RC_CHANNELS* - not defined.
- Sept 6, 2012 - [This commit](https://github.com/mavlink/mavlink/commit/4b5a966169ab4ca2179ba263e8f827907f1b668b) shortly after (2012) changed [RC_CHANNELS_SCALED](https://github.com/mavlink/mavlink/commit/4b5a966169ab4ca2179ba263e8f827907f1b668b#diff-ceacc1e2390186650146a45be8fcb691R1094).rssi and `RC_CHANNELS_RAW .rssi` to use the [0-100] (%) with 255 for "not used".
- March 25, 2014 This [commit](https://github.com/mavlink/mavlink/commit/3909ff755a62cae67e2b5bd5e0729aa35e88078d) `RC_CHANNELS` was created from the `RC_CHANNELS_RAW` - `.rssi` copied too using format  [0-100] (%) with 255 for "not used", and adding support for 18 channels.

@LorenzMeier so the end result is that `RC_CHANNELS` itself was always the way it is for MAVLink 2, but it derives from RC_CHANNELS_SCALED and RC_CHANNELS_RAW that were formerly the old way.

Either way we have a confusing compatibility story. We also have a reasonable argument that a bigger range is better for some use cases, and contrasting that we have the argument that the current way is better in that it has a mechanism for showing "not supported".

If it were up to me and we could fix compatibility I'd be tempted towards either `[0 - 254]=[0-100%], 255=not used` OR move to a new message using an absolute measurement for signal strength.

hamishwillee, 2019-02-04:
OK, so looking at the history, the message was effectively changed to the current behaviour in 2012 using whatever the process was at the time. This has always been the way it is in MAVLink 2 timeframes. So any systems that chose not to update in the last 5 years have either missed the change or chosen to ignore it. 

It is also not clear that using the full range 0-256 is any more useful than 0-100 because of noise and other factors - all most users are interested in is "good signal", "not good signal".
It is clear that using 256 as a not-used value *is* better that not having an explicit "not used" value.

Had a chat to @LorenzMeier about this. PX4 would consider `0-255` range with `256=not used`. The assumption is that this will work OK with systems that aren't updated to specify 256 as not used - because no real system using `0-256` will ever _really_ output the max value. Is that reasonable?

We would hope ArduPilot would move to this too - otherwise there isn't much point having a standard.

Thoughts? @auturgy  @DonLakeFlyer

hamishwillee, 2019-02-07:
@DonLakeFlyer et al. Decision in devcall was to move rssi to the [0-254] format, with 255 indicating not used. I'll update in bits. The first one is `RADIO_STATUS` - https://github.com/mavlink/mavlink/pull/1078. I'm trying to work out what the other fields do too.

edwinhayes, 2019-02-10:
RSSI measurements have units.  And (if you ignore the sign), real world values for RSSI are typically in a range which can be represented easily in eight bits: -20 to -120 being pretty typical.  Why wouldn't you just use the actual measurement?

edwinhayes, 2019-02-10:
Further, I realise that an obvious objection is that, a: it's non-trivial to obtain an easy '0-100%' reading to present to users; and b: it's inconsistent with existing generators of the message.

This would be readily resolved on the GCS side though by having a pair of config parameters for 'full' and 'empty' which then allow a linear transformation which is compatible with any radio.  (Flame wars could then be conducted regarding whether sensible default values would be 20-120 or 0-254 etc)

auturgy, 2019-02-10:
This really is a bit of a vague area, as RC RSSI, SiK radio RSSI, and other links such as 802.11 (wifi) all represent RSSI differently.
The 802.11 spec uses 0 - 255, but allows manufacturers to define their own max within that range (Cisco use 0-100, Atheros uses 0-60 for example). The SiK radios use a formula based on dBm with a multiplier and offset. Crossfire can output rssi as dB or Link Quality (0-300%).
I think using 0-254 in mavlink covers most bases. A parameter based on the particular radios fitted might be useful to aid specific meaning for some.

hamishwillee, 2019-02-10:
> I think using 0-254 in mavlink covers most bases. A parameter based on the particular radios fitted might be useful to aid specific meaning for some.

@auturgy So you're suggesting we might have a `rssi_type` extension to explain that a particular radio instead is using dbm say for this value? Interesting idea, for which I'd think a new message was safer.

auturgy, 2019-02-11:
It doesn’t necessarily have to be an extension or message field (although that might useful). A parameter or gcs option would achieve the same result, and documentation to increase the likelihood of a user understanding what the rssi value means might be enough.
Basically, because rssi isn’t universally the same, interpretation by/presentation to the user is problematic, and would benefit from a means of clarification. As we don’t currently do anything other than present a number to the user, I don’t think that should hold up this change though.

Regards,

James

hamishwillee, 2019-02-12:
@auturgy Thanks. For RADIO_STATUS the updated docs now say 
`<field type="uint8_t" name="rssi">Local (message sender) received signal strength indication. Values: [0-254], 255: invalid/unknown.</field>`
So yes, we could make it more clear that these are  device dependent. How about:
`<field type="uint8_t" name="rssi">Local (message sender) received signal strength indication in device-dependent units/scale. Values: [0-254], 255: invalid/unknown.</field>`

dogmaphobic, 2019-02-12:
Just keep in mind that the great majority of users don't think "RSSI". They think x _bars_ out of 5. That fits in 3 bits with room to spare. Any RSSI _value_ shown to the user is highly dependent on the hardware in use and unless you have a degree in EE, it is bound to be meaningless.

hamishwillee, 2019-02-12:
@dogmaphobic True, but we're stuck in a world where this is already used and has a device-specific "definition" - and also where a few things do need greater resolution than the 5 bars.

So while I completely agree, I don't see how I can take that suggestion as "direction".Any suggestions on better wording much appreciated!

edwinhayes, 2019-02-12:
Well, it's only "x bars out of _five_" if you're using a ground control package which has a signal strength widget which has five segments; the format which is sent over the air should be independent of the manner in which the data is presented to the user.

I think that ground control software having parameters for calibrating what constitutes 'full reception' and 'low reception' is the best approach, since not only does this allow for handling different devices, it also caters to the situation that some users might want to change the scaling to provide better resolution in their expected operating environment.

hamishwillee, 2019-02-13:
@edwinhayes Gus was being very generous - arguably most users actually want only 2 options: "good enough", "not good enough" :-) 
In any case, we can't modify these messages as suggested because they already have meaning.

@auturgy above is more or less suggesting the same thing as you: https://github.com/mavlink/mavlink/issues/1027#issuecomment-462203263 - a way to work out what the rssi actually means for a particular device. That discussion is well worth having here, but wont hold up the changes I am currently making.

edwinhayes, 2019-02-13:
@hamishwillee - well, based on assessment of some of the flights we see customers do, that could possibly be simplified down further to a single value of "it's probably still ok". :-)  As you note, the actual change to the message definition now seems fine.

PR #1083 (2019-02-21): Update RSSI values to 0-254, 255=not used
Update RSSI values in  [RC_CHANNELS](https://mavlink.io/en/messages/common.html#RC_CHANNELS), [RC_CHANNELS_RAW](https://mavlink.io/en/messages/common.html#RC_CHANNELS_RAW), [RC_CHANNELS_SCALED](https://mavlink.io/en/messages/common.html#RC_CHANNELS_SCALED) to same definition discussed in #1027 (note, RADIO_STATUS already updated).

Fixes #1027

@LorenzMeier @auturgy Please confirm you are happy with this. @auturgy as discussed, this will need to be updated in both PX4 and ArduPilot.
closed 2019-02-21 by hamishwillee