mavlink/mavlink #1035: Increase the output frequency of pixhawk using mavros/Unable to connect to mavlink

utkarsh447, 2018-12-10:
## Hi, I got two issues to discuss,
### 1. I am trying to access pixhawk using GPIO via mavlink, but i cannot connect,
I get following error:
```
pi@pi:~$ mavproxy.py --master=/dev/ttyS0 --baudrate 57600 --aircraft aa1
WARNING: You should uninstall ModemManager as it conflicts with APM and Pixhawk
Connect /dev/ttyS0 source_system=255
no script aa1/mavinit.scr
Log Directory: aa1/logs/2018-12-10/flight1
Telemetry log: aa1/logs/2018-12-10/flight1/flight.tlog
Waiting for heartbeat from /dev/ttyS0
 MAV> Exception in thread main_loop:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/bin/mavproxy.py", line 811, in main_loop
    master.wait_heartbeat(timeout=0.1)
TypeError: wait_heartbeat() got an unexpected keyword argument 'timeout'

```

My mavlink library versions are as follows:
```
Name: pymavlink
Version: 2.0.6
Summary: Python MAVLink code
Home-page: http://github.com/mavlink/mavlink
Author: UNKNOWN
Author-email: UNKNOWN
License: LGPLv3
Location: /usr/local/lib/python2.7/dist-packages
Requires: 
Required-by: MAVProxy, dronekit, droneapi
```
```
Name: MAVProxy
Version: 1.7.7
Summary: MAVProxy MAVLink ground station
Home-page: https://github.com/ArduPilot/MAVProxy
Author: Andrew Tridgell
Author-email: andrew@tridgell.net
License: GPLv3
Location: /usr/local/lib/python2.7/dist-packages
Requires: pymavlink, pyserial
Required-by: droneapi
```

Whereas when i try to connect via mavros using command:
`roslaunch mavros px4.launch fcu_url:=/dev/ttyS0:57600 tgt_system:=1`
I get the heartbeat with following log:
```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

*** TOPIC LISTS ***

NODES
  /
    mavros (mavros/mavros_node)

auto-starting new master
process[master]: started with pid [25229]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7da38bfe-fc54-11e8-ae5c-b827ebb6c53f
[... 105 lines cut ...]

### 2. I am trying to increase the output frequency of pixhawk using mavros:
I want to increase the frequency of IMU data output.
I am following PX4 option of the answer mentioned in the link,
[increase-output-frequency-of-pixhawk-using-mavros](https://answers.ros.org/question/213390/increase-output-frequency-of-pixhawk-using-mavros/), 
I am using Sanddisk ultra 16 GB class 10 Micro SD card and doing the following steps:
1. Formatted the SD card and inserted that card in Pixhawk, 
2. Ran pixhawk, performed basic commands, closed pixhawk, and check the contents of SD card, I got logs in that. (I can conclude SD card is working).
3. I created /etc/extras.txt folder in root with this code snippet: 
```
#!nsh
#
# MAVLink start
#

mavlink start -d /dev/ttyS0 -b 57600
# Enable a number of interesting streams we want via USB
mavlink stream -d /dev/ttyS0 -s ATTITUDE -r 200
mavlink stream -d /dev/ttyS0 -s HIGHRES_IMU -r 200

# Exit shell to make it available to MAVLink
exit
```
Inserted sdcard back and ran the pixhawk. With this, i am unable to get heartbeat. 
Please suggest the proper way to increase the output frequency from pixhawk?

amilcarlucas, 2019-02-26:
update the pymavlink version to 2.3.4, update the mavros version to latest. and retry

amilcarlucas, 2019-03-26:
If we get no answer, I vote we close this one.

hamishwillee, 2019-03-26:
The first is failing because it can't see heartbeats on that input serial port. It may be failing here due to some other misconfiguration - i.e. there is no output param.
The second problem is PX4 specific and should be raised on the PX4 discussion boards. 

I'm closing this now thanks @amilcarlucas - because it is a mavproxy and/or PX4 problem, not a MAVLink problem *at this point*. @utkarsh447 Feel free to repost on PX4 discussion boards.

closed 2019-03-26 by hamishwillee