mavlink/mavlink #1051: CI failing Python 3.3 tests due to lmxl 3.6 incompatibility

hamishwillee, 2019-01-07:
CI for Python 3.3 failing all PRs in the environment setup stage because "This lxml version requires Python 2.7, 3.4 or later.". Tests pass on 2.7 and 3.5. The version does not appear to have changed for sometime in ArduPilot/pymavlink, so no idea what is going on.

Failing example here: https://travis-ci.org/mavlink/mavlink/jobs/476173108#L432 (from https://github.com/mavlink/mavlink/pull/1048)

dagar, 2019-01-07:
I'm not sure what to suggest, lxml dropped support for Python 3.3 in the current version (4.3.0). You could update the requirements.txt to specify a specific version of lxml.

[image]

What are we trying to achieve with mavlink CI? If the requirement (or desire) is to test a wide range of python versions I'm wondering if we'd be better off leaning on docker for the environments.

hamishwillee, 2019-01-09:
@amilcarlucas Thanks.

> What are we trying to achieve with mavlink CI? If the requirement (or desire) is to test a wide range of python versions I'm wondering if we'd be better off leaning on docker for the environments.

@dagar I've created #1056 to remove the Python 3.3 build from travis. We don't really need it to work on any particular version of Python because this is a build toolchain, not a runtime toolchain. As long as it runs on near recent Python 2 and 3 I'm OK (ideally the versions shipped with Ubuntu LTS)

PR #1056 (2019-01-09): Remove tests on Python 3.3
Fixes #1051

Tests are failing on Python 3.3 due to lxml dependency no longer supporting that version. IMO 3.3 is a legacy test platform superseded by Python 3.5 - there is no real need to continue to support it.
closed 2019-01-09 by LorenzMeier