mavlink/mavlink #1055: Component ID for Video Streams

dogmaphobic, 2019-01-08:
There are no assigned component IDs for Video Streams. How do we go about creating them?

This also brings up how to arrive at a solution for video stream discovery.

Camera discovery for instance, relies on a camera component sending heartbeats. If the GCS supports it, it will notice the heartbeats and start sending camera information requests to the camera (using the camera component ID found within the camera's heartbeat message). In other words, the GCS doesn't really care what the component ID is. The allocation is there simply to create some sort of discipline on its use.

Video discovery however, uses broadcasts. That is, the GCS initiates the discovery by sending video streaming info requests to component ID 0 (broadcast). If a vehicle supports is, it would respond, if not, it would ignore it and the GCS would assume no video streaming is supported (to the extent that automatic video streaming discovery is supported).

The problem with broadcasting it is that the autopilot will also receive this command and complain about not knowing what it is (beeping, flashing LED lights, etc.) The solution seems to be to do the same as camera discovery. Have the vehicle (its video stream component) send heartbeats using a "Video Stream Component ID".

Does this make sense?

hamishwillee, 2019-01-09:
> There are no assigned component IDs for Video Streams. How do we go about creating them?

Not sure what level of detail you are after. Normally you'd just create a PR to add a value to MAV_COMPONENT in https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L463

Usually just create one value like MAV_COMP_ID_VIDEO_STREAMER with any free id from further up the range. If a system might have multiple streamers that need to be separately addressed you'd create MAV_COMP_ID_VIDEO_STREAMER2 (and so on) and give them sequential values

hamishwillee, 2019-01-09:
What you are already doing with broadcast sounds fine. 
I wasn't aware that an autopilot or other system would  have a tantrum if it got a command it doesn't understand a message - I presume by "does not understand" you mean if the autopilot gets a message that was not in common.xml when it was built - so this only applies to systems that are using older versions of mavlink? If not, what causes this?

In any case, a component does not in theory have to broadcast its heartbeat for this to work. I *think* that if you give the streaming component a component ID and set the autopilot to accept messages from that component all will be well. The component heartbeat would only _have_ to be broadcast if the streamer is on another network interface than a receiving system (e.g. if autopilot gets the message via telemetry raido, and forwards it to streamer component on companion computer via serial cable) because otherwise the message would not be routed.

hamishwillee, 2019-01-09:
Setting closed - discussed offline.

closed 2019-01-09 by hamishwillee