mavlink/mavlink #1063: GPS_RAW_INT: eph, epv and HDOP/VDOP mixed-up? Which values does it report?

rodizio1, 2019-01-14:
Someting seems to be mixed-up there, quoting from:
https://mavlink.io/en/messages/common.html

> eph | uint16_t |   |   | GPS HDOP horizontal dilution of position (unitless). If unknown, set to: UINT16_MAX
> -- | -- | -- | -- | --
> epv | uint16_t |   |   | GPS VDOP vertical dilution of position (unitless). If unknown, set to: UINT16_MAX
> 

AFAIK, eph and epv are _NOT_ the same as HDOP and VDOP.  eph and epv is not unitless (it uses "meter" as a unit), only HDOP and VDOP are.

Question would be, which values are really reported? I'd be interested in epv and eph, as VDOP/HDOP do not really show GPS degradation due to interference (only takes the geometrical positions of the satellites into account).

hamishwillee, 2019-01-15:
The message description is correct and the fields are miss-named. 

They really should be renamed to hdop and vdop, but that can't be done without breaking message compatibility in the generated libraries, and with the deployment in PX4, ArduPilot, and many existing GPS systems.

We could [extend](https://mavlink.io/en/guide/define_xml_element.html#message_extensions) the message to add a real eph/epv. If you need this, I propose you create a PR for this.

@LorenzMeier Is it worth creating a new message for this? I imagine having eph and "really_eph" in one message will be very ugly. Better to have eph and hdop. 

FYI: There is quite a bit of history on this message - most of which can be read in, or is is linked from, https://github.com/dronekit/dronekit-python/issues/412

closed 2019-01-15 by hamishwillee

rodizio1, 2019-01-15:
Thanks for the info.