mavlink/mavlink #1086: [C library] mavlink_parse_char argument mavlink_status_t * misleading

michprev, 2019-02-15:
https://github.com/mavlink/c_library_v2/blob/d240d0986710045663894aebcea89e71ce981ee4/mavlink_helpers.h#L931-L971

First of all code example is obsolete - it does not pass 4th argument (mavlink_status_t *). https://github.com/mavlink/c_library_v2/blob/d240d0986710045663894aebcea89e71ce981ee4/mavlink_helpers.h#L962

Some members of mavlink_status_t are never assigned - for example `msg_received`, `buffer_overrun`, `parse_error` (this is very misleading!). Value from `parse_error` is assigned to `packet_rx_drop_count` - https://github.com/mavlink/c_library_v2/blob/d240d0986710045663894aebcea89e71ce981ee4/mavlink_helpers.h#L832

Please let me know if you need more info / code example.

hamishwillee, 2019-02-18:
@michprev Thanks for your post. I have posted a partial PR for this in https://github.com/ArduPilot/pymavlink/pull/271 that fixes the problem with the missing parameter - and some other incorrect names for those parameters (generally ArduPilot/pymavlink is the right place for generator issues).

I'm not sure about the last point - I think those assignments are made indirectly in functions called by `mavlink_parse_char ()`

hamishwillee, 2019-02-18:
Oh yes, and appreciate your confirmation that it does look like a good fix!

michprev, 2019-02-26:
Thank you for forwarding me to the correct repository. The fix is perfect.
About the last point - I will create an issue in https://github.com/ArduPilot/pymavlink so this one can be closed.

closed 2019-02-26 by michprev