mavlink/mavlink #1088: SERVO_OUTPUT_RAW streaming rate can't be changed

raffaele590, 2019-02-18:
Hello, I am using a Pixhawk cube with the latest firmware on it, uploaded using Qgroundcontrol.
Using the Mavlink Inspector Widget I can give a look at the frequencies at which the different massages are streamed, and using the command    mavlink stream -d /dev/ttyACM0 -s ALTITUDE -r 40 I am able to change the rate of every message. The weird thing is that using the command mavlink stream -d /dev/ttyAMC0 -s SERVO_OUTPUT_RAW -r 20 the related message doesn't change at all.

That's the same issue I have when using the "Mathworks Pilot Support Package (PSP)" I create a custom firmware. In that case though the frequency is only 1Hz and I really need to increase it in order to do HIL. I have tried using the same command "mavlink stream -d /dev/ttyAMC0 -s SERVO_OUTPUT_RAW -r 20" but nothing. I thought it was an issue related to the PSP, but even using a regular firmware downloaded from QGC that rate can't be changed.  

Therefore I thoiught that's a general Mavlink problem.

Thank you so much for your help.

hamishwillee, 2019-02-18:
This is not a MAVLink defect - the project provides message definitions and libraries for processing these messages, but it is up to a GCS, flight stack or other system to handle them.

Depending on the flight stack you might be able to change the rate by sending the MAVLink message [MAV_CMD_SET_MESSAGE_INTERVAL](https://mavlink.io/en/messages/common.html#MAV_CMD_SET_MESSAGE_INTERVAL)

closed 2019-02-18 by hamishwillee

raffaele590, 2019-02-18:
Thanks for clarifying, that helps.

How do I send that particular Malvink Message? 
I have a PX4 flight Stack v 1.8.2 Stable Release.

Thank you!!

hamishwillee, 2019-02-18:
@raffaele590 YOu might do it with the Dronecode SDK: https://sdk.dronecode.org/en/api_reference/classdronecode__sdk_1_1_mavlink_passthrough.html

Before doing that though I would ask your question on the discussion boards http://discuss.px4.io/ - there might be an easier way to do this.

raffaele590, 2019-02-18:
Thank you for the hints, really appreciate.