mavlink/mavlink #1090: COMMAND_INT/MISSION_ITEM_INT definition danger spot

MaEtUgR, 2019-02-26:
[`COMMAND_INT`](https://mavlink.io/en/messages/common.html#COMMAND_INT) and [`MISSION_ITEM_INT`](https://mavlink.io/en/messages/common.html#MISSION_ITEM_INT) both state that param 5 and 6 are of type `int32` and contain `PARAM5 / local: x position in meters * 1e4, global: latitude in degrees * 10^7`.

Now this definition according to my understanding says:
- Either it's a global coordinate a fixed 10^7 exponent that allows for higher accuracy than a float because the exponent doesn't need to be sent, it's in the message definition. This is presumably the main reason for coming up with COMMAND_INT in the first place.
- Or it's an arbitrary parameter of integer type. (Least sure about this one)
- Or it's a local coordinate with a fixed exponent of 10^4

The problems are:
- Integer being incompatible with certain definitions of commands that expect to send a float in param 5/6 like defined in [`COMMAND_LONG`](https://mavlink.io/en/messages/common.html#COMMAND_LONG) like e.g. [`MAV_CMD_DO_MOUNT_CONFIGURE`](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_MOUNT_CONFIGURE) or [`MAV_CMD_DO_SET_ROI_WPNEXT_OFFSET`](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ROI_WPNEXT_OFFSET)
- The implementation of distinguishing the different cases does not scale. You need to have an extra case for every new message to decide if it's a coordinate or not.
- To support the last point I refer to current implementations on QGC (mission item int [send](https://github.com/mavlink/qgroundcontrol/blob/6b641477ac1310039dcf379426b00fc161273a52/src/MissionManager/PlanManager.cc#L549-L550), [receive](https://github.com/mavlink/qgroundcontrol/blob/6b641477ac1310039dcf379426b00fc161273a52/src/MissionManager/PlanManager.cc#L400-L401)) and PX4 (command int [receive](https://github.com/PX4/Firmware/blob/f229444722132a2e71d92d425597faaadcb480c4/src/modules/mavlink/mavlink_receiver.cpp#L499-L500) which always use the 10^7 exponent for the INT versions

So now the question is how do we resolve this? Did I interpret this correctly? Is there actually an implementation following the definition interpretation I wrote above? Can we not simply always use 10^7 exponent to mimic a float? What was the [initial discussion and intention](https://github.com/mavlink/mavlink/pull/230/files) @LorenzMeier @tridge ?

In any case I think the definition needs to be improved even if it's just clarification.

hamishwillee, 2019-02-26:
@MaEtUgR 

Your interpretation of param 5 and 6 in the *_INT* variants is correct. If the command is positional, the frame field tells you if it is local or global and it is interpreted as indicated. 
If it is not positional then it is just an int32 - and I in this case if you need it to be interpreted as 10^7 exponent you would add `units="degE7"` to the MAV_CMD param.

Historically there was no way to tell that the message is positional or not except from description, but now you could apply `hasLocation="false"` in the schema to indicate this (@DonLakeFlyer I assume if hasLocation is set false, the default of isDestination=true will be ignored?)

As you say, there is still a problem with any message that explicitly assumes either a float in param 5 and 6 - it can't be reliably sent using the *_INT variants. Similarly, anything that assumes an int for reasons of precision can't be sent in the *_LONG variants.

So you're right, this is worth discussing in dev call next week. Minimally a clear way to know what message a command can be sent it, but ideally a standard way that a recipient can interpret the data if received in either message.

@auturgy FYI

MaEtUgR, 2019-02-28:
@hamishwillee Thanks for the explanation! ok, so the problem is valid and we need a clear documented distinction for the commands that is hopefully not too complicated to implement to reduce the error probability.

DonLakeFlyer, 2019-02-28:
> Can we not simply always use 10^7 exponent to mimic a float?

Which is basically what QGC always does/expects.

MaEtUgR, 2019-02-28:
@DonLakeFlyer The command part of PX4 does this as well. Only for the mission items there's a distinction which doesn't cover all commands.

But using the 10^7 exponent results in a maximum parameter value of `214.7483647` because of the integer limit. It's probably still best if it's defined within MAVLink which commands are used with `COMMAND_INT` because they use the parameters for a position, profit from `MAV_FRAME` and high accuracy and which ones with `COMMAND_LONG`.

hamishwillee, 2020-06-23:
As an aside, the current version of the mission protocol has dropped MISSION_ITEM, so only MISSION_ITEM_INT exists. 

For commands I agree, it would still be great to know what commands can be used with a command_int, or some alternative.