mavlink/mavlink #1094: Reduce memory usage of Mavlink library

tuloski, 2019-03-04:
I'm running mavlink on an arduino pro mini, and it only has 2048 bytes of memory.
I noticed that mavlink library takes the most of that memory. Without any optimization it was taking more than 2048 bytes only mavlink.
The functions that add lots of memory are: mavlink_parse_char() and mavlink_msg_to_send_buffer().

I did the following to optimize the memory, but still I'm at the limit:
* Set MAVLINK_COMM_NUM_BUFFERS to 1
* Set MAVLINK_MAX_PAYLOAD_LEN to 70 (because I'm using only a set of messages with payload < 70 bytes)

Is there anything else I can do?

hamishwillee, 2019-03-04:
@LorenzMeier @peterbarker @auturgy Any advice on optimising the MAVLink C library for very low memory systems?

hamishwillee, 2019-03-04:
One of the best things you can do is reduce the number of messages in your dialect. Each one generates additional code.

vooon, 2019-03-05:
@hamishwillee deleting messages shouldn't help, because all unused functions should be stripped by linker.

@tuloski perhaps you may find some optimizations in old APM code (for AVR mega).
One of possible thing to optimize - move message CRC array to PROGMEM.

tuloski, 2019-03-05:
Actually I had a big buffer to store the message for mavlink_msg_to_send_buffer().
After reducing that and reducing MAVLINK_COMM_NUM_BUFFERS and MAVLINK_MAX_PAYLOAD_LEN it can fit in SRAM of pro mini. Only 50% used (1kB).

amilcarlucas, 2019-04-05:
@tuloski can you please post a PR?
That way future users can profit from your code changes.

tuloski, 2019-04-09:
@amilcarlucas what I did was a hack for my very particular situation. I do not reccommend to use my modifications, because I assume to have only one comm link and a maximum payload of 70 bytes which might not be true for other applications.

amilcarlucas, 2019-04-09:
If you add that information into the PR than it becomes usefull :)

hamishwillee, 2020-06-23:
Closing this now. 

@tuloski Should you wish, creating a PR to add instructions on how to create a very small variant, covering what you did/do (https://github.com/mavlink/mavlink/issues/1094#issuecomment-469797991) , and the constraints (https://github.com/mavlink/mavlink/issues/1094#issuecomment-481147524) of your solution, it would be very welcome :-)

closed 2020-06-23 by hamishwillee