mavlink/mavlink #1095: message range request

climr, 2019-03-12:
Requesting reservation of a group of 1000 message IDs for Horizon31’s BLOS communication system.  Company info: URL: http://www.horizon31.com, email: brad@horizon31.com

hamishwillee, 2019-03-12:
Hi @climr ,

Sure. You can use 43000 - 43999

When do you plan on creating a definition file and pushing a version to master. I ask so that I can reference it and make the allocation clear in https://mavlink.io/en/guide/define_xml_element.html#message_id_ranges

If you don't plan on submitting I can still reserve the range.

hamishwillee, 2019-03-12:
Note however, ideally your system would try to use messages in common.xml rather than create your own dialect. The main reason to create your own dialect is if you want to talk to a close ecosystem.

climr, 2019-03-12:
Thank you, yes we currently have a fairly large message set that enable and will work only with our closed system.  We had not planned on publishing the dialect, but we much appreciate the reservation of 43000-43999 to ensure we maintain compatability.  Thanks!

hamishwillee, 2019-03-13:
Well it won't actually maintain compatibility, but I guess it will ensure that if you use the messages with other dialects there won't be clashes :-)

@LorenzMeier Do you think it worth creating a "user range" for message ids - essentially a range that anyone can use for their private networks? That way we wouldn't need to create private ranges for cases like this unless they need to interwork with other private software that uses the range - at which point they are no longer "private"

hamishwillee, 2020-06-24:
@climr Gosh, lots of time since this.

Had a chat to other interested parties in the dev call. Upshot is that we won't allocate private ranges for now. I assume you have not run into any problems in the last few years so this is not a "real" problem. If it proves to be in future, reopen and we will allocate you one.

closed 2020-06-24 by hamishwillee