mavlink/mavlink #1098: Common Test Requirements Missing

podhrmic, 2019-03-15:
Hi all,

based on the discussion with @hamishwillee and @tstellanova regarding [Rust-mavlink](https://github.com/3drobotics/rust-mavlink) I would like to get some sense of what tests are required to make a new language implementation compliant. Please let me know if there is a better place for this issue.

Looking at the mavgen generated (C) code, there seems to be single encode/decode test for each message with presumably random values for the payload.  Is that correct?

My understanding is, we want the following:
- any message with arbitrary field values can be successfully encoded
- any message with arbitrary field values can be successfully decoded
- encoding/decoding either succeeds or gracefully fails (i.e. no segfaults or exceptions)
- anything else?

These things are not clear to me:
- when parsing an enum that doesn't have a defined value (let's say `MAV_AUTOPILOT`), should the parser fail (`"unknown enum value"`) or use some default value instead? Let's say I am parsing `MAV_AUTOPILOT=99`. Is the result an `Error`, or `MAV_AUTOPILOT_GENERIC`?
- if an enum doesn't have a defined numeric value (e.g. `MAV_CMD_ACK`), should the values be generated, and if so should they start from 0 or 1?

Such tests can be handled by using for example [quickcheck](https://github.com/BurntSushi/quickcheck) for Rust implementation, or [Hypothesis](https://hypothesis.works/) for Python. I would be very happy to explore how to use quickcheck for rust-mavlink, once the test requirements are clear.

hamishwillee, 2019-03-18:
> when parsing an enum that doesn't have a defined value (let's say MAV_AUTOPILOT), should the parser fail ("unknown enum value") or use some default value instead? Let's say I am parsing MAV_AUTOPILOT=99. Is the result an Error, or MAV_AUTOPILOT_GENERIC?

If you're asking about a case like [MAV_STATE](https://mavlink.io/en/messages/common.html#MAV_STATE) where only the number of the first enum value is defined (0) then all other enum values are automatically auto-numbered with sequential values.

Or are you asking about what a library function should do if someone passes in a number that has no associated enum value - ie MAV_AUTOPILOT = 99? Library users should be passing in enum values rather than numbers, but either way, the documentationdoesn't say anything about this case and the library functions don't do any range checking. IMO that is the right thing:
- the library is supposed to be lightweight
- allowing currently unsupported values means that adding a new value does not automatically break miss-matched implementations.

Or did you mean something else?

hamishwillee, 2019-03-18:
> if an enum doesn't have a defined numeric value (e.g. MAV_CMD_ACK), should the values be generated, and if so should they start from 0 or 1?

Yes, they should be autonumbered from 1 - see C implementation: https://github.com/mavlink/c_library_v2/blob/master/common/common.h#L628

hamishwillee, 2019-03-18:
Regarding the rest of your question, I have added some other reviewers and hope to discuss in dev call this week. If you don't hear back Thurs/Friday please ping me/us.

hamishwillee, 2019-03-18:
A note back from the person who probably knows most about the pymavlink-based tests:
"Pretty sure pymavlink really only does the first two, really. Fuzz testing would not be a bad idea."

podhrmic, 2019-03-18:
Thanks for the response @hamishwillee 

> Or are you asking about what a library function should do if someone passes in a number that has no associated enum value - ie MAV_AUTOPILOT = 99? Library users should be passing in enum values [...]

Thanks for the clarification. I don't think "range checking" on enums would bring any significant overhead for strongly typed languages (like Rust), but it would definitely require all supported values to be defined in the XML file. On the other hand, this makes fuzz testing much easier:-)

hamishwillee, 2019-03-18:
Let's see if anyone else has anything to add ...

podhrmic, 2019-03-19:
Just FYI, what I got from fuzz testing so far:

For example, this input `[254, 1, 5, 254, 104, 75, 246, 0, 0, 8, 0, 246, 8, 104, 0, 254]` causes `panic!` in the current Rust implementation, and raises error (`raise MAVError('Unable to unpack MAVLink header: %s' % emsg)`) in Python implementation.

I suspect that raising an error is fine, but segfault/panic is not:-)

hamishwillee, 2019-03-19:
> I suspect that raising an error is fine, but segfault/panic is not:-)

Yes :-). 
As a complete guess, I suspect you've set the compatibility bit for a signature but do not have the correct length for the message. With a value of 5 for the incompatibility bits I expect no messages should actually be decodeable, but as you say they shouldn't segfault.

hamishwillee, 2019-03-21:
G'day, 
Raised this in the MAVLink devcall last night. No one had much to add except:
1. We can't _tell_ you to do more than what the mavgen testing in ArduPilot/pymavlink does. Basically that means all [official dialect XML files](https://github.com/mavlink/mavlink/tree/master/message_definitions/v1.0) should parse, validate, and be able to encode/decode for all the messages
1. Failures should be properly handled - not "crashes".
1. Your testing sounds fine/good. We love the idea of fuzz testing but we don't do for mavgen, so wouldn't see it as "compulsary" (see first point above).
1. It would be _interesting_ to run a test of your library output into pymavlink, and pymavlink output into your library to see if anything breaks. Currently the pymavlink tests are looped back, so this is not possible. If you are able/have time would be cool to enable them to be decouple-able (even if still decoupled). Someone might do that at some point - feel free to take it on, but see point 1 above :-)

Thoughts?

hamishwillee, 2020-06-22:
Hi @podhrmic 

Long time no see. I'm cleaning up the stale items. 

Any reason to keep this open?

Do you think perhaps capturing the comments on testing from  https://github.com/mavlink/mavlink/issues/1098#issuecomment-475106150 in the contribution section ([below here](https://mavlink.io/en/contributing/contributing.html#how-to-contribute-complex-changes would be useful))?

podhrmic, 2020-06-22:
Hello @hamishwillee - indeed!
I agree we can close this. Yes, it would be great to add a few notes to the contribution section.

hamishwillee, 2020-07-15:
Thanks @podhrmic . Closing this.

Would appreciate your review of "the fix" in https://github.com/mavlink/mavlink-devguide/pull/263

closed 2020-07-15 by hamishwillee