mavlink/mavlink #1099: Programmatically reset the EKF

walid19, 2019-03-18:
how can i creat mavlink "reset" command that tells SITL to stop and start the estimators (attitude/position) ?

hamishwillee, 2019-03-18:
Why do you need it? / What is the problem you are trying to solve? / What flight control stack are you targeting?

walid19, 2019-03-18:
I am working on a reinforcement learning algorithm  using  px4 stack 1.80 and gazebo SITL as a simulator. I want to reset the UAV location at the starting point while UAV hit something(collision count plus one).  when I do so the quadrotor lose the attitude control drift and go far away due to the bad state estimation, I need to reset the EKF and the drone location at the same time.

hamishwillee, 2019-03-19:
This does not sound very "generic" to me - ie the message is really just for your particular research. As such you probably should create your own dialect and modify the code to handle your particular messages. There is some information on this in http://dev.px4.io/en/middleware/mavlink.html#receiving-custom-mavlink-messages

LorenzMeier, 2019-04-12:
This is not a real MAVLink question - closing. Please ask this here: http://discuss.px4.io

closed 2019-04-12 by LorenzMeier

ZhangZiyu1216, 2025-10-24:
It might be a bit late to answer this issue, but I recently encountered the same problem.

After starting PX4 and MAVROS, I checked `/mavros/local_position/local` and discovered that the z-altitude had drifted to over -600 meters. However, the drone was still able to take off normally in Position mode, and the altitude-relative data also appeared correct. After searching through a lot of documentation without success, my only explanation is that the EKF2 initializes too early upon PX4 startup, before the GPS position has converged. By the time the GPS data is stable, the drone's localization is already based on an incorrect origin. To fix this, I needed a way to programmatically reset the EKF2 origin before my ROS application starts.

I found some discussions suggesting that this could be done by sending `ekf2 stop` and `ekf2 start` in the QGC MAVLink Console, but doing this manually every time is inconvenient. Fortunately, I found [an official PX4 example script that provides access to the MAVLink shell from a terminal](https://docs.px4.io/main/zh/debug/mavlink_shell). I realized that with a few modifications, I could turn this into a script to programmatically restart the EKF2.

Here is my script:

```python
#!/usr/bin/env python3

from __future__ import print_function
import sys
import time
from argparse import ArgumentParser

try:
    from pymavlink import mavutil
except ImportError as e:
    print("Failed to import pymavlink: " + str(e))
    print("\nYou may need to install it with:\npip3 install --user pymavlink\n")
    sys.exit(1)

try:
    import serial
except ImportError as e:
    print("Failed to import pyserial: " + str(e))
    print("\nYou may need to install it with:\npip3 install --user pyserial\n")
    sys.exit(1)

class MavlinkSerialPort():
    '''an object that looks like a serial port, but
    transmits using mavlink SERIAL_CONTROL packets'''
    def __init__(self, portname, baudrate, devnum=0, debug=0):
        self.baudrate = 0
        self._debug = debug
        self.buf = ''
        self.port = devnum
        self.debug("Connecting with MAVLink to %s ..." % portname)
        self.mav = mavutil.mavlink_connection(portname, autoreconnect=True, baud=baudrate)
        self.mav.mav.heartbeat_send(mavutil.mavlink.MAV_TYPE_GENERIC, mavutil.mavlink.MAV_AUTOPILOT_INVALID, 0, 0, 0)
        self.mav.wait_heartbeat()
        self.debug("HEARTBEAT OK\n")
        self.debug("Locked serial device\n")

    def debug(self, s, level=1):
        '''write some debug text'''
        if self._debug >= level:
            print(s)

    def write(self, b):
        '''write some bytes'''
        if isinstance(b, str):
            b = b.encode('utf-8')
        
        self.debug("sending '%s' of len %u\n" % (b, len(b)), 2)
        while len(b) > 0:
            n = len(b)
            if n > 70: n = 70
            chunk = b[:n]
            buf = list(chunk)
            buf.extend([0]*(70-len(buf)))
            self.mav.mav.serial_control_send(self.port,
                                             mavutil.mavlink.SERIAL_CONTROL_FLAG_EXCLUSIVE |
                                             mavutil.mavlink.SERIAL_CONTROL_FLAG_RESPOND,
                                             0, 0, len(chunk), buf)
            b = b[n:]

    def close(self):
        self.mav.mav.serial_control_send(self.port, 0, 0, 0, 0, [0]*70)
[... 71 lines cut ...]
        mav_serialport.write('ekf2 start\n')
        
        # Read the response from the start command.
        response = mav_serialport.read_response(timeout=1.0)
        if response:
            sys.stdout.write(response)
            sys.stdout.flush()

        print("\n--- All commands sent successfully. ---")

    except Exception as e:
        print(f"\nAn error occurred: {e}")

    finally:
        if mav_serialport:
            mav_serialport.close()
            print("Connection closed.")

if __name__ == '__main__':
    main()
```

I'm not sure if this will still be helpful to you now (it being 2025 and all), but I hope it can help others who run into this issue, as it troubled me for a long time.

By the way, I am using PX4 v1.16.