mavlink/mavlink #1109: Is Mavlink feasible for real time control of drones?

purplehorse, 2019-04-01:
I have read several times that "MAVLink is a protocol for communication between drones and/or ground control stations." but I always read it in the context of autopilot and missions. Is it possible to use Mavlink as a real time controller? For example, How would be possible to perform this kind of communication without sending a complete mission:

- Create connection
- Send order (e.g. initiate flight)
- Send order (e.g go left X seconds)
- Send order (e.g. land)
- Close connection.

hamishwillee, 2019-04-01:
Yes. Basically there are two ways to do this. 
- You can monitor telemetry and missions, and then upload a new mission as circumstances change.
- You can use the command protocol to send instructions to the vehicle - just as you have outlined above.

The main thing to remember is that this is not "real time" in the sense of guaranteed and very fast delivery of messages. You can use it to do the sort of sequence you outlined above though.

Typically people us a higher level API to do this sort of control, because some of the work is removed. If you're using ArduPilot flight stack then consider using [DroneKit-Python](http://python.dronekit.io/). If you're using PX4 flight stack, then consider using the [Dronecode SDK](https://sdk.dronecode.org/en/)

closed 2019-04-01 by hamishwillee

purplehorse, 2019-04-02:
Thank you for your response.

I assume a full manual control could be provided by the use of MANUAL_CONTROL (#69) and a more relaxed kind of manual control could be provided by SET_POSITION_TAGET_LOCAL_NED (#84)? Correct me if I am wrong. 

Thank you also for your recommendation of APIs, however, I am interested on working on Mavlink level to try to reduce latency as much as possible (that is why I am asking about manual control means, because I assume they are faster to changes than missions).

According to guaranteed / fast deliver it also depends on the transport protocol Mavlink relies, isn't it? If I am not mistaken, Mavlink works over transport layer so I could deploy it over TCP, UDP, etc.

Sorry for the possible inconveniences caused by my questions, any help would be highly appreciated.

hamishwillee, 2019-04-02:
> I assume a full manual control could be provided by the use of MANUAL_CONTROL (#69) and a more relaxed kind of manual control could be provided by SET_POSITION_TARGET_LOCAL_NED (#84)? Correct me if I am wrong.

NOT quite. There is no guarantee any MAVLink message will actually be supported by the flight stack. Generally the examples for DroneKit show some supported commands for ArduPilot, while examples for SDK show some of what works on PX4. 

 MANUAL_CONTROL is intended for a ground control station to send joystick like commands to a vehicle - e.g. for on-screen joysticks or to integrate with a real joystick. It is not usually used for control by a script.

SET_POSITION_TARGET_LOCAL_NED is commonly used for low level control of position or velocity. If you just want to go to a position you might use https://mavlink.io/en/messages/common.html#MAV_CMD_DO_REPOSITION in the command protocol. But again, easier to use a high level API for that. 

> I am interested on working on Mavlink level to try to reduce latency as much as possible (that is why I am asking about manual control means, because I assume they are faster to changes than missions).

What are you trying to achieve? Typically the higher level APIs like dronekit and the SDK will introduce a little latency, but they effectively do what you will have to code yourself anyway - it you will add this level of latency and probably more if you try to use mavlink directly. 

> According to guaranteed / fast deliver it also depends on the transport protocol Mavlink relies, isn't it? If I am not mistaken, Mavlink works over transport layer so I could deploy it over TCP, UDP, etc.

Yes, you can deploy it over a faster transport layer than a radio telemetry set. Often people connect a serial link from an on board computer to drive the flight stack. Whether Wifi will be fast enough for what you want depends on what you are trying to achieve.

purplehorse, 2019-04-04:
I am trying to create my own API for a proyect. The idea is to control a drone over a wireless connection (as wifi but no point to point) over a transport protocol. I read that Mavlink is one of the most extended protocols to control drones but I do not want to use any autopilot, as I want to test the capabilities of my own API sending and receiving messages controlling the drone. In this experimental way, do you think Mavlink is a good choice? What commands would you use to control the drone as manual as possible?