mavlink/mavlink #1116: MavLink Global position coordinates

RohanKurane, 2019-04-12:
Hi,

@lucasdemarchi  @LorenzMeier @zehortigoza 

Here is a snippet of my code - 
```                                      
  <<< establish connection to PX4 FC >>>

  uint8_t buf[MAVLINK_MAX_PACKET_LEN];
  mn = recvfrom(tcp_fd, buf, sizeof(buf), 0, (struct sockaddr *)&px4_sockaddr, &addrlen);
  if (mn == -1)
        return false;
else
 {
       for (mi = 0; mi < mn; mi++)
       {
            printf("total bytes : %d\n", mn);
            main_loop:
            {
                if (mavlink_parse_char(MAVLINK_COMM_0, buf[mi], &msg, &status))  // UART 0/1 ?
                {
                     if (msg.msgid == MAVLINK_MSG_ID_GLOBAL_POSITION_INT )
                     {
                          // process GLOBAL POSITION of sUAS
                          mavlink_global_position_int_t global = {};
                          mavlink_msg_global_position_int_decode(&msg, &global);
                          printf("latitude: %d:\n", global.lat);
                          printf("longitude: %d:\n", global.lon);
                          printf("altitude: %d:\n", global.alt);
                     }
               }  // error on parse_char
       }   // process all bytes in for loop  
}
```

My question is, if the sUAS is flying, the global coordinates that I see in the MavLink inspector widget on QGC do not match the coordinates I see in the above code. 

The altitude seems to change above and match the widget but not the latitude and longitude fields. 

Is there anything I am doing wrong above ?

Thanks for your help

Rohan

LorenzMeier, 2019-04-12:
I would check your print arguments to make sure you're getting the right number of digits out of your system. Please don't use the MAVLink issue tracker to answer user-code questions. This is to track real MAVLink issues, so I'm closing this.

closed 2019-04-12 by LorenzMeier

RohanKurane, 2019-04-12:
@LorenzMeier  
Thanks, 
What is the right number of digits ?  
This is a bit important so re-opening for now. Im sorry.

RohanKurane, 2019-04-12:
Before closing please provide an answer @LorenzMeier

LorenzMeier, 2019-04-12:
Feel free to ask in the forum: http://discuss.px4.io - we need to keep the issue tracker for issues and free of usage questions. MAVLink is developed by volunteers and these developers are overwhelmed with support questions. If you share the forum link to your post here, I'd be happy to continue the conversation there.

RohanKurane, 2019-04-12:
Thanks @LorenzMeier 
Here you go - http://discuss.px4.io/t/mavlink-global-position-coordinates-do-not-match-with-qgc-widget/10460