mavlink/mavlink #1121: Microservice and direction attributes

amilcarlucas, 2019-04-24:
The messages should have two optional enum attributes:

1. ```service``` should be an enum of: 

-    Heartbeat
-     Mission
-     Parameter
-     Command
-     Camera
-     Gimbal
-     ArmAuthorization
-     ImageTransmission
-     FTP
-     LandingTarget
-     Ping
-     PathPlanning

2.  ```direction``` should be an enum of:

- MAV_TO_GCS
- GCS_TO_MAV
- MAV_TO_MAV
- MAV_AND_GCS_BIDIR
- GCS_TO_PAYLOAD
- PAYLOAD_TO_GCS
- PAYLOAD_TO_MAV
- MAV_TO_PAYLOAD

or one could add two attributes ````source``` and ```destination```, both a bitmask of

- MAV bit0
- GCS bit1
- PAYLOAD bit2
...

This would allow to fully define the way the message is meant to be used. For example GCSs would be able to filter the messages that they can send and generate a GUI only for the messages and commands that they can send.

DonLakeFlyer, 2019-04-24:
That just seems like too much detail trying to be respresented in meta data.

hamishwillee, 2019-04-30:
Interestingly though, it is a slice of the [MAVLink microservices versioning](https://docs.google.com/document/d/1LkfhOn-FPNYnKmSZw0mf7gyc3A8GI-191DUi_yC-Fxk/edit?usp=sharing) doc - note, this is a read-only *copy*.

For that the intent is more to be able to query for support for a service. I think that approach makes more sense than this. In particular don't really agree with the direction idea, because some messages can go any old direction (both in theory and practise), which defeats the purpose.

So I lean towards - "should we look at the versioning protocol proposal again"?.

hamishwillee, 2020-04-23:
Closing. This is good, but there are more recent efforts ongoing to address.

amilcarlucas, 2020-04-23:
I still think that adding two attributes: source and destination; both a bitmask of;

-     MAV bit0
-     GCS bit1
-     PAYLOAD bit2

Whould be a good idea!! The GCS would then know what to do wth the messages.

hamishwillee, 2020-06-22:
@amilcarlucas I see why you suggested this, and I was initially keen on the idea, but it really is only of theoretical benefit.

1. The GCS still does not know what to do with the messages. Or more precisely it knows what to do for some messages, but this does not capture a microservice sequence and is hence of limited use. I also think it is very GCS/UAV centric. 
2. There is no pull for this from any stakeholder/GCS.

 Closing again. We can reopen and discuss if a GCS or payload vendor is requesting this and plans to implement.

closed 2020-06-22 by hamishwillee