mavlink/mavlink #1122: bufferIndex variable in mavlink_frame_char_buffer() superflous

olliw42, 2019-04-26:
in the function mavlink_frame_char_buffer() in mavlink_helpers.h a variable bufferIndex is used, which appears to have no function, and thus should be removed IMHO.

```
MAVLINK_HELPER uint8_t mavlink_frame_char_buffer(mavlink_message_t* rxmsg, 
                                                 mavlink_status_t* status,
                                                 uint8_t c, 
                                                 mavlink_message_t* r_message, 
                                                 mavlink_status_t* r_mavlink_status)
{
...
	int bufferIndex = 0;
...
	bufferIndex++;
...
}
```

It would be inconsistent to define it as int given that everywhere else uintXX is used anyhow.

I'm sorry to not do a PR, but this file isn't present in the repo, but generated, and I have no idea about that part.

hamishwillee, 2019-04-30:
Thanks. I think you're right.

> I'm sorry to not do a PR, but this file isn't present in the repo, but generated, and I have no idea about that part.

The code is imported as a submodule from https://github.com/ArduPilot/pymavlink - the correct place to make the change is in this file in that repo: https://github.com/ArduPilot/pymavlink/blob/master/generator/C/include_v2.0/mavlink_helpers.h
It would be great if you could do that, but if not, let me know and I'll take care of it.

amilcarlucas, 2019-05-03:
This should be reported upstream at https://github.com/ArduPilot/pymavlink

hamishwillee, 2026-08-25:
Closing as "wrong repo".

closed 2026-08-25 by hamishwillee