mavlink/mavlink #1123: including dialects as described in the wiki can't work (IMHO)

olliw42, 2019-04-26:
in the wiki it is said:

> If you support multiple independent dialects you can include these separately:
> 
> #include <common/mavlink.h>
> #include <your_dialect/mavlink.h>
> #include <another_dialect/mavlink.h>

(https://mavlink.io/en/mavgen_c/)

This doesn't, and from looking at the code, can't work as described. Once common/mavlink.h is included, all following dialect/mavlink.h would be ignored since each mavlink.h is protected by the same MAVLINK_H, e.g. for ardupilotmega 

```
#pragma once
#ifndef MAVLINK_H
#define MAVLINK_H
...
#include "version.h"
#include "ardupilotmega.h"

#endif // MAVLINK_H
```

such that any further includes are not executed, in this case the include of ardupilotmega.h

Several corrections/soultions come to mind, since I don't know which one you had in mind, no suggestions from my side.

hamishwillee, 2019-04-30:
olliw42 Thanks, you're right. 

@julianoes Can you advise here on best-practice for including multiple dialects? We might advise:

----

If you support multiple independent dialects you should import mavlink.h first (for any dialect in order to get a set of standard `#define` values), and then include the remaining dialect-named header files:
```
#include <common/mavlink.h>
#include <your_dialect/your_dialect.h>
#include <another_dialect/another_dialect.h>
```

olliw42, 2019-04-30:
I would conclude that it is essentially NOT possible (in a sufficiently easy way) to include multiple dialects.

The part which prevents this is the MAVLINK_MESSAGE_CRCS.

In some way I think this restriction even makes sense, given that dialects might have overlapping usage of message-IDs.

So I think the wiki simply should say that (with the lib as is) only one dialect can be imported, and that this is done by including dialect/dialect.h. 

The canonical approach for multi-dialects would then be on the .xml level, which I think one can live with and also in some way makes sense. (i.e. to make one new .xml which includes both dialects)(it might be that the xml parser then needs a mechanism to include identical xml's only once, if it doesn't have this already, since dialects might e.g. share the common xml)

IMHO

hamishwillee, 2019-04-30:
@olliw42 The toolchain allows you to build multiple dialects at once and will let you know (if validation is enabled) if you've duplicate message or enum ids. It also generates a [combined](https://github.com/ArduPilot/pymavlink/blob/master/generator/mavgen.py#L90) MAVLINK_MESSAGE_CRCS for all the dialects you specify.

While I tend to agree with you that creating a dialect that includes all the message sets you're interested in is a better approach, the multiple independent dialect mechanism should work and if it does, should be documented. As an aside, until recently the toolchain only supported a single level of dialect inclusion, so it may be that this approach was the only one you could use in some cases.

olliw42, 2019-05-06:
> The toolchain allows you to build multiple dialects at once and will let you know (if validation is enabled) if you've duplicate message or enum ids. It also generates a combined MAVLINK_MESSAGE_CRCS for all the dialects you specify.

yeah ... but that's just a variation of what I denoted as canonical procedure. It works (can work) whenever it's created in a coordinated way.

My point was that you can't add together C files once created, in contrast to what is implied by the wiki. This is so for various reasons, and essentially all of them can be solved except the MAVLINK_MESSAGE_CRCS thing. MAVLINK_MESSAGE_CRCS is exclusive and not additive, hence the "multiple independent dialect mechanism" does not work. Pl check for yourself, you'll find this to be correct.

In my estimation it won't be easily made to work, so we are left with the canonical procedure. And the wiki needs to be corrected. The comment was thus a polite hint of how to most easily resolve things :)

have fun :)

hamishwillee, 2019-05-07:
@olliw42 Curse it, you're right. Thank you.

hamishwillee, 2019-05-07:
Fixed in https://github.com/mavlink/mavlink-devguide/pull/181
@olliw42 If you have time, would appreciate your review.

hamishwillee, 2019-05-08:
Fix merged. Thank for this.

closed 2019-05-09 by hamishwillee