mavlink/mavlink #1124: suggestion: support r_message = NULL in mavlink_frame_char_buffer()

olliw42, 2019-04-26:
the function mavlink_frame_char_buffer() in mavlink_helpers.h returns a variable mavlink_message_t* r_message, which is filled internally by a memcpy from rxmsg to r_message.

`memcpy(r_message, rxmsg, sizeof(mavlink_message_t));`

(this line occurs twice in the function code)

The suggestion is to allow r_message = NULL, by changing the two occurrences to

`if (NULL != r_message) { memcpy(r_message, rxmsg, sizeof(mavlink_message_t)); }`

I note that r_message = NULL seems to be explicitly allowed already, given that such if's can be found in code lines further down. 

I also note that it might be that memcpy would handle a NULL correctly, in which case the suggestion would be superfluous. However, using google seem to indicate that this might be so but might also not be so, in which case it would make it safe.

I lastly note that the idea here is to avoid having to have two mavlink_message_t buffers. This is useful on not-so-RAM-rich systems. This might be also relevant in context of issue https://github.com/mavlink/mavlink/issues/1094.

olliw42, 2019-04-29:
ok, have seen that it had been added to pymavlink already weeks ago: https://github.com/ArduPilot/pymavlink/commit/136d0e4b73b2a6e11909889e4e77c5ed568756e2

mavlink just doesn't  yet use the latest pymavlink module :)

thus closed.

closed 2019-04-29 by olliw42

hamishwillee, 2019-04-30:
Thanks. I've created PR to pull latest fixes in https://github.com/mavlink/mavlink/pull/1126