mavlink/mavlink #1132: Geofence breach status missing

julianoes, 2019-05-08:
We have an enum `FENCE_BREACH` to report the geofence breach status but no message to actually send it. I wonder if it made sense to add a message to transmit the geofence status, or if this would rather be part of the events interface. @bkueng could you comment?

`FENCE_BREACH` had initially been [moved over from ardupilotmega.xml](https://github.com/mavlink/mavlink/commit/8458b6ce66cb677c5c7fb55e4450f741b12fcc47) 5 years ago.

bkueng, 2019-05-08:
This fits better into the events interface. But depends on how fast you need it.

julianoes, 2019-05-08:
Ok, I can wait :smile:.

julianoes, 2019-05-15:
@auturgy would be great to get your comment.

hamishwillee, 2019-07-04:
@auturgy ^^^^

auturgy, 2019-07-04:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Fence.cpp
Fence status is definitely sent by ArduPilot. Of course the fence protocol is different.

hamishwillee, 2019-07-04:
Thanks @auturgy - @julianoes there you have it :-)

julianoes, 2019-07-04:
@hamishwillee would it be possible to move [FENCE_STATUS](https://github.com/ArduPilot/mavlink/blob/4e566bba4b466118f41650122e84fc8c12bb1351/message_definitions/v1.0/ardupilotmega.xml#L1124-L1130) to common?

hamishwillee, 2019-07-04:
I assume that because we hope to replace this with events interface at some point you don't particularly want to revisit the design of the message?

If not, sure, but I have not moved anything before and I don't know the best way to co-ordinate it because normally apm changes come from the dialect fork as a PR, and we push common changes down.

I *think* we just make both changes to mavlink/mavlink, and the push that down to ardupilot as a separate effort.

@auturgy Can you advise?

auturgy, 2019-07-04:
I’d suggest pull it in to common keeping the message unchanged (including ID).
ArduPilot can then choose to sync, or not, via the normal sync cycle managed by @amilcarlucas
I think (?) that the build system will catch the duplication and reject it from the included XML, so it shouldn’t matter. CI might tell us. If it doesn’t, might be an interesting test to add (ie building a test.xml that has all dialects as includes, or something along those lines).

Regards,

James

PR #1173 (2019-07-07): Move FENCE_STATUS to common
Fixes #1132.
closed 2019-07-07 by LorenzMeier