mavlink/mavlink #1133: Proposal for Improving User-fiendliness of Component Handling

olliw42, 2019-05-08:
Components are a significant part of a Mavlink ecosystem, but their support by the GCSes, from a user perspective, is somewhat modest and not 2019. A main issue is that relevant meta data to the component's parameters, which can help the user tremendously in handling the things properly, is missing. For the autopilot in contrast, GCSes such as MissionPlanner or QGC provide such meta data, for an obvious good reasons I would argue. MissionPlanner is in fact quite user-friendly in that aspect, as it provides detailed info on the parameter ranges/values as well as some descriptive text. Also QGC has such user-friendliness and provides short descriptions.

For cameras a similar problem has been identified, in that cameras can differ quite substantially by their modes, shutter speeds, options, and so on, which needs to be handled by a GCS. The camera micro-service has solved that problem quite neatly, via the CAMERA_INFORMATION message, which allows a GCS to ask for a URI to a camera definition file, which the GCS then can read and digest to present a nice and useful interface to the user.

The proposal here is to generalize/extend that mechanism to all components, i.e., to define a message which allows a GCS to request for information from the component, which among others would include an URI to a component definition file. This file would be also xml, and may contain all sorts of info, and among it the meta data to the component's parameters.

There would be several ways to do that. For instance, one could rename the MAV_CMD_REQUEST_CAMERA_INFORMATION to MAV_CMD_REQUEST_COMPONENT_INFORMATION, and the component would respond with a message appropriate for the component. That is, a camera would respond with the existing CAMERA_INFORMATION message, while a typical component could respond with a yet to create COMPONENT_INFORMATION message. Another approach could be to create a MAV_CMD_REQUEST_COMPONENT_INFORMATION command along with a COMPONENT_INFORMATION message. There might be further approaches.

Similar to the camera micro-service, a GCS would call that request when it connects to a component, read the component information file and could present the user with a user-friendly interface. Given that the component handling and the camera micro-service is already in place, I would think that it should not be very difficult to implement such a scheme into GCSes (they are free to choose how much/little they want to make use of this anyhow).

I'd like to emphasize that this concept could open the door to further "nice" things. It e.g. could be carried over to UAVCAN nodes if there is a Mavlink bridge. That is, the UAVCAN component could similarly respond to a request, which the GCS then could use to provide a nice GUI. The definition file could also include additional info. For instance, it could provide a "translation table" from the Mavlink parameter name to the component's native parameter name. That way, the GCS could display the parameters with using names the user would be familiar with already from the native GUI (e.g. the parameter names in the UAVCAN GUI Tool of a Zubaxx gnss, or the parameter names of a STorM32 gimbal in the STorM32's GUI Tool)(for UAVCAN nodes this could be considered a "solution" to the different parameter length issue). It also could be linked with some plugin mechanism to show some buttons and such. And so on and forth. In short, since that concept is very flexible, and generic, one can expect other great applications to come along in future, once the mechanism is established.

Cheers, and always have fun,
Olli

(I've checked the open issues and couldn't find a similar topic, hopefully I haven't overlooked something)

olliw42, 2019-05-08:
@julianoes
it is obvious that my perspective is pretty narrow and that I didn't have any clue about the fuller picture you were painting. So, I'm a bit hesitant to answer since any of my thoughts necessarily will be somewhat basic, but that's what is in my mind:

(1) I'm surprised by you calling it complex, since I thought the camera spec is really neat and simple. One provides an uri for an xml file with the specs, which the GCS downloads and processes. How much simpler could it be. It hardly can be harder than a programming language, right. But as said, I obviously don't have a full picture. 

(2) I would have thought a relatively simple version number scheme would be perfectly fine, but again I obviously have a simple-minded picture. To ensure that there is an up-to-date xml file available is really the responsibility of the component maintainer, not yours, I would say.  

(3+5) My general attitude would be to say that much of what you indicate wouldn't be your problem. In my simple perspective a specification's purpose is to make possible certain things and precisely define how that is to be done. And the spec's main concern is to be a good spec. End of story. So, I don't see (3) and (5) as a concern. One can't prevent folks from using things in ways they want, and even in ways which breaks a specification. Their decision. Fragmentation in the parameters and the related issues for SDKs is unavoidable. The overlap in parameters will always be small, I can't see how it could be different. It is in fact the primary job of a specification to define a subset of functionality and how that should work, and this can then be handled by a SDK and be used by developers. Every parameter/function beyond that scope is - well - outside of that scope. The kind of fragmentation you describe in (3) I actually would consider a good thing. I mean, how cool is it to have established a mechanism which allows folks to do all sorts of things WITHOUT YOU having to do anything!

Since that's the situation how, in my mind, it is and ever will be (unless you freeze innovation), there are only two reasonable possibilities: The parameters&features beyond the scope of the specification are either not handled or are handled in a generic way. Component specific handling could of course also be done, but that's a pain and not something a standard should attempt.

(4) This I don't understand. I might have gotten the docs wrong, but in my understanding a component also just could give an internet url, from which a GCS would download the definition file. Maybe you are only concerned about the situation where folks want to use the system with new/updated components but the GCS doesn't have an internet connection. If so, I would not be concerned at all. I mean, you can't use google maps offline if you haven't downloaded the maps beforehand. I would think that this is a concept everyone could accept.

The GCS might help by voluntarily checking the "known" xml files for updates (similar to what they appear to do with e.g. the autopilot firmware anyhow)

So, in summary, I guess I largely don't see the issues, but as said before, it's obvious that I don't see the full picture.

>Regarding the changing of the messages, we need to find a way to be backwards compatible. 

I think both approaches I suggested are backwards compatible. The 2nd obviously so, but also the first, since it's still the same command as before, and any component knows if it's a camera or not. From my perspective it's more a matter of taste. I understand your argument to not rename but to just re-purpose. 

> I think it would be good to discuss this on the MAVLink dev call in order to also get the feedback from the APM side. If you could join the call next week Wednesday morning (see calendar) that would be fantastic.

thx for the invitation but I'm generally not great at using this modern communication stuff. No idea how simple/difficult it would be. Anyways, I checked the schedule and I'm busy unfortunately Wednesdays from 9 to 12:30. Maybe I could free some time from 9:00 to 9:30, but then things must really go swiftly, i.e., there would be no time for fooling around with getting connected to the dev call and such things.

cheers, Olli

hamishwillee, 2019-05-09:
@dogmaphobic FYI, a proposal to generalise the camera definition file approach so it can be used by any component.

jkflying, 2019-05-09:
Just to add my 2 cents, I'd also be interested in something similar to this for components running on a companion computer. Right now they need to be configured through a side-channel (ROS, ssh etc) or somehow try to piggyback off of flight controller parameters. It would be great to have a mechanism to configure well described parameters directly from the GCS.

I would also suggest on trying to make the mechanism of distribution standardized, so there aren't too many quirks. I guess MAVLink FTP would be best there, especially if a library could be provided.

hamishwillee, 2019-05-09:
> I would also suggest on trying to make the mechanism of distribution standardized, so there aren't too many quirks. I guess MAVLink FTP would be best there, especially if a library could be provided.

MAVLink FTP is already supported in QGC and PX4: https://mavlink.io/en/services/ftp.html . IMO this microservice needs to be revisited as the payload section is opaque and vendor specific. 

Just to be pedantic, I don't think we'd standardise the mechanism of distribution in the mavlink definition - a  generalised URI is much more flexible. We could however provide recommendations for methods that will sort out the problem.

dogmaphobic, 2019-05-10:
> FYI, a proposal to generalise the camera definition file approach so it can be used by any component.

That makes sense. Until now, only cameras had this need, hence, the camera API. As to transport methods (for the XML) file, good question. Anything other than TCP/IP would require an entire new (transport layer) protocol, not to mention most non TCP/IP links are too slow and/or cumbersome. You'd be better off serving it off a standard web server, which is how this was designed to be.

jkflying, 2019-05-10:
Just some more thoughts - sorry it got a bit long.

Having this parameter metadata served by something on the internet has reliability implications. The component is now dependent on some file being hosted at some address in perpetuity. Looking at the number of broken links I find for sites from just a few years ago, I'm not confident in the longevity of components dependent on this. Also, the one time I need to reconfigure my __ will be the time I'm not connected to the internet because I'm in a different country, on a boat, in a rural area, ran out of data, etc. which is a frustrating user experience.

There's also the privacy/security perspective. Any component could make the GCS leak information by requesting some customized URL with information encoded in it. Anything could be in the URL - locations the camera was triggered, other component IDs, parameters from the flight controller. The design shouldn't allow every component in the system on getting implicit (admittedly 1-way) access to the internet without the user's knowledge.

Hence, I'm more in favor of the MAVLink FTP, which is entirely internal to the system, is guaranteed to be available as long as the component is running, and is self contained thus there is one less place for things to go wrong, eg. out of sync versions. Basically, if it works once it should continue to work, and shouldn't require continuing maintenance.

I know it puts more pressure on the low bandwidth links, but hopefully it doesn't need to be refreshed too often.

julianoes, 2019-05-10:
@dogmaphobic I vote to use MAVLink FTP wherever possible. QGC and PX4 already support it, the Dronecode SDK should follow soon. 

@olliw42 
(1) Haha, that's relative! :smile: But I'm glad you see it that way.

(2) That's true, it's up to the maintainer, and we could show warnings in QGC etc. if it doesn't match.

(3), (5)  Yes, you're right it's also very much a good thing for adoption.

(4) Yes you can give it a URL but often when you fly a drone you might not have a good connection because you're out in the field, or your phone doesn't have a connection because it's connected to some Wifi in order to communicate to the drone. Therefore, I'd recommend MAVLink FTP as @jkflying said.

> From my perspective it's more a matter of taste. I understand your argument to not rename but to just re-purpose.

Yes, I'd say we discuss that in the call. If you join cool, otherwise we'll post an update here afterwards.

olliw42, 2019-05-15:
@julianoes
I was really considering showing up at the dev call, but it seems you are using zoom and it seems it's commercial, with only a free 30day trial version, and doing all that effort to get into it with knowing it's good for only 30 days didn't make sense to me. So, sorry for my absence.

@LorenzMeier
>could you point me towards a full setup (controller + gimbal) that we could order and put onto one of our test aircraft?

I'm aware of only 4 comerically available ready-to use gimbals which use the STorM32, but they all unfortunately use I2C and run on outdated firmware, and can't be used as such without modding. However, there are these options:

Mike Kelly, who is a well-known and regular user at rcgroups and ardupilot discuss, and I also have seen him at gitter, has offered to put together a working and ready-to-use test gimbal setup (based on the HAKRC gimbal)  and donate it to you or whoever of you should get it. If that is an option you like to chose I can get you into contact if needed.

However, in order to test and play around with all that MaAvlink stuff one doesn't actually need a fully working STorM32 gimbal, just the electronics would be sufficient for all except of practical real-world tests. This would open up a heck of options:
* Probably the easiest would be to get a v3.3 STorM32 controller (with USB, option 5) and a NT Imu form Gapey: https://t-storm32.jimdofree.com/assembled-pcb-for-beta-testers/ and https://t-storm32.jimdofree.com/assembled-pcb/. Here one just needs a proper cable to connect the two in addition
* The v1.3 NT set from banggood, https://www.banggood.com/de/Storm32BGC-NT-V1_3_F103RC-32-bit-Brushless-Gimbal-Control-Board-p-1105106.html?akmClientCountry=DE&rmmds=search&cur_warehouse=CN, which is also available (but a bit pricey) in Germany here: https://powerparts-uav.de/produkte/gimbal-controller/406/storm32bgc-nt-v1.3-f103rc-32-bit-brushless-gimbal-control-board. With this set one needs to spend a bit more effort to make a cable to connect the controller and imu boards.
* I could do a firmware version which doesn't need a IMU at all, which would allow one to use essentially any board with a STM32F103RC chip, i.e., on could use just a STorM32 controller board for 20-25$ without a camera imu, but also dev boards like those cheap Asian F103RC boards. Chances are you guys have such a thing rotting somewhere in some drawer :)

That's all I can suggest.

:)

hamishwillee, 2019-05-15:
@olliw42 You don't need to sign up for a trial to join Zoom meetings, just to create them. If you click the meeting link it should download the needed client software. In addition, there are phone numbers you can use: https://mavlink.io/en/about/support.html#schedule

The meeting is in about 2 hours and 10 mins

olliw42, 2019-05-15:
ah, oh ... 
so the phone number thing means I just grab any conventional phone and call that phone number and then I would hear you all and you would hear me?? ... I shouldn't be too old for doing this LOL ...

auturgy, 2019-05-15:
fyi mavftp is in work for ardupilot https://github.com/ArduPilot/ardupilot/pull/11183

olliw42, 2019-05-16:
@julianoes
as follow up to the devcall discussion, I'd like to share my thoughts on using MavlinkFTP for the purpose at hand. If you feel that this discussion would be better placed in its own issue, just tell.

Let me first present a simple math as an example showing the kind of challenge components like a STorM32 and potentially many others (without a SD card) might face: The STorM32 has more than 250 parameters. A description for one might take on average, if one keeps them brief, let's say, 50 characters, in addition there is the data for ranges, and ideally also a translation table from the MAVLink parameter name to the native name. In addition xml is very verbose. So, counting 100 bytes per parameter is probably conservative, which makes it 25kB of flash at minimum. And no cool other features yet. And this is just one xml, a device like the STorM32 will have to store internally several versions to react properly on some parameter settings. It's not only lots of flash, but then all that erasing, copying, creating needs RAM to handle. And once one has started this, it will very likely grow and grow since more wishes come up. In short, MAVLinkFTP, as cool as it is, is a mightmare for such kind of systems.

I see two approaches: Either a simple set of new cmds/messages which accomplish what at the end is a simple task, namely to transfer a junk of data from the component to the GCS. Or a minimal functional MAVLinkFTP, which essentially only supports reading, and has only one file in its "directory".

My initial thought was along the first option. The mechanism could be very similar to reading the parameter list. I.e. a CMD_REQUEST_DEFINITIONFILE send out by the GCS, to which the component responds with messages  MSG_DEFINITIONFILE_JUNK, where the EOF would be indicated by a data size length shorter than the maximal junk size (similar to how UAVCAN is doing it). One could add another request to reread a particular junk, to allow a GCS to get missed junks, similar to how missed parameters are handled. Simple, easy, and efficient.
For instance, it would make it easy to store the xml data in a more compressed format (i.e without the xml verbosity), and to create the xml file "on the fly", which should massively reduce flash requirements.

I've looked now again deeper into the MAVLinkFTP spec, and think it also could be fine, if some restrictions are acceptable:
* Only TerminateSession, ResetSessions, OpenFileRO, ReadFile would be supported. All others would be NACKed with a FAIL (or a newly created error OPCODE_NOT_SUPPORTED).
* Given that it holds only one file, supporting ListDirectory would also not be a biggy, but I don't see any use of it. That is, a GCS simply could attempt a OpenFileRO with the desired filename, and give up if that is NACked.
* no idea about CalcFileCRC32
* I have not yet fully penetrated all implications of the timeout/resending mechanism, but in case it would require to keep substantial previous info in RAM to make it to work according to spec, I would suggest to find ways to lessen the requirements. It should be sufficient to be able to create the data junk on the fly.
* One would specify that any component only has to support these MAVLinkFTP parts to comply with the spec as regards to the definition file thing. A GCS should not attempt to use other functionalities, and if it does so it should not have a problem with getting along happily with the NACKs.
* Probably not really required, but it could have advantages to add a capability field to the ACK response to OpenFileRO (and other "initiator" opcodes), so that a GCS could determine which set of opcodes is supported.

After having looked at it again, I now actually don't have a significant preference for the one or other approach. So, please just tell me what you think, or what better alternative you see. I would then start to implement the scheme in STorM32.

As a note of potential relevance: The STorM32 controller is not only a gimbal controller but also a camera controller (always has been). I have in fact already implemented the camera micro-services (not in the official version, the latest alpha), and it already allows using a cameradefiniton file, which however is currently stored on my server. So, what this means is that we can work out and test the definitionfile thing very quickly, since it's going to be the same for camera as for any other component. We just use the STorM32 as camera and not as gimbal. And QGC has lots of camera stuff already too. I was serious, if we are quick with decisions we will have done it by the end of the weekend LOL :) :) :)

julianoes, 2019-05-16:
> And this is just one xml, a device like the STorM32 will have to store internally several versions to react properly on some parameter settings.

I think only the one version matching the current firmware is required.

> Only TerminateSession, ResetSessions, OpenFileRO, ReadFile would be supported. All others would be NACKed with a FAIL (or a newly created error OPCODE_NOT_SUPPORTED).

I think that makes sense.

> Given that it holds only one file, supporting ListDirectory would also not be a biggy, but I don't see any use of it. That is, a GCS simply could attempt a OpenFileRO with the desired filename, and give up if that is NACked.

I don't think we need it because we just pass the URL in the definition message: `mavlinkftp:///definition.xml`

> I have not yet fully penetrated all implications of the timeout/resending mechanism, but in case it would require to keep substantial previous info in RAM to make it to work according to spec

You don't need to have it in RAM I think, you can just `seek` around the file, no?

I'm not familiar with MAVLink FTP yet, so if you want you can go ahead and try it out. Otherwise, you'd have to wait for me to look into it. I mostly need to add support in the Dronecode SDK for it to, that's to download files, and while testing it I should probably add the server part as well.

olliw42, 2019-05-16:
> I think only the one version matching the current firmware is required.

I wasn't clear, sorry. The STorM32 is both a gimbal controller and a camera controller. For the gimbal part only one xml is need for the firmware version, as you say. For the camera part at least a 2nd different is needed, and since STorM32 supports several cameras it needs several. I currently will need 2 xmls: 1x gimbal, 1x runcam split. I hope to also get the gopro better integrated in some future => 3. And also the sonys => 4. And so on and forth.
It's already exploding beyond the STorM32's flash space :)

> You don't need to have it in RAM I think, you can just seek around the file, no?

this is my current understanding too, but I feel unsure since I don't have yet the feeling I fully realize all edge cases. 

> if you want you can go ahead and try it out.

I'm not sure how to read it: Are you implicitly saying that you would want to decide for the moment for the minimal MAVLinkFTP approach over a new REQUEST_DEFINITIONFILE message set? A simple yes/no would be great :)

> you'd have to wait for me to look into it

I think it's then more depending on the QGroundControl guys, since that's what I would operate with :)

thx sir

julianoes, 2019-05-16:
> Are you implicitly saying that you would want to decide for the moment for the minimal MAVLinkFTP approach over a new REQUEST_DEFINITIONFILE message set?

Yes. Sorry that's what I meant. QGC has implemented MAVLink FTP already but it would need to use it to download the xml file. This second part is missing.

olliw42, 2019-05-16:
> Yes. Sorry that's what I meant.

great, so I'll go along it.

> QGC has implemented MAVLink FTP already but it would need to use it to download the xml file. This second part is missing.

yeah, I realized there is already lots there but not yet directly usable for the purpose. As much as I understand it also has MAVLinkFTP enabled only for PX4 but not ArduPilot (I'm on ArduPilot, more for historical reasons than by heart though; I've tried PX4 last weekend but couldn't get the MAVLink port yet working LOL). But eventually QGC will want to have that all working too, so I hope for an open ear on the QGC side. I'll simply ask them and see :)

julianoes, 2019-05-16:
> But eventually QGC will want to have that all working too, so I hope for an open ear on the QGC side. I'll simply ask them and see :)

Best to talk to @dogmaphobic I'd say since he was driving the camera definition protocol and is working on QGC.

olliw42, 2019-05-20:
it seems that QGC is not an immediate option, as you may have seen here, https://github.com/mavlink/qgroundcontrol/issues/7453. Maybe most relevant is that QGC is currently in the process of switching to QtQuick-only, which will take some more time I believe.

So, I'm not sure how to best proceed with the above. 

I have seen that MP has a recent MAVLinkFTP commit, so I might test MP in that regard.  

However, the above requires MAVLinkFTP only in second place, as it's just one option for fetching the xml. The main task would be to define proper MAVLink, messages, proper xml fields, and to have them digested by a GCS to produce a GUI, for which QGC was the obvious candidate. 

So, any better idea for what to do than just waiting?

:)

julianoes, 2019-05-20:
As an alternative I'm planning to add Mavlink FTP to the SDK, so hopefully that won't take as long as the QGC one.

olliw42, 2019-05-20:
I think you indicated that before, but I'm not sure it's an alternative, in the meaning of the word.

MAVLinkFTP would be just a secondary feature, i.e., one always could, in a first step, easily rely on fetching the xml from a web uri, as it's done by QGC currently (and which is working fine). That is, even if MAVLinkFTP would be fully available on all sides, an extended component GUI would be as far away as it would be without it.

Unless I misunderstand what "add Mavlink FTP to the SDK" means. Does the SDK provide some GUI-ish features?

julianoes, 2019-05-20:
> MAVLinkFTP would be just a secondary feature, i.e., one always could, in a first step, easily rely on fetching the xml from a web uri, as it's done by QGC currently (and which is working fine).

Sure that's fine for a first step.

> Does the SDK provide some GUI-ish features?

Not it's just a C++ library and optional language wrappers.

olliw42, 2019-05-20:
>Not it's just a C++ library and optional language wrappers.

this was my understanding, thx for clarifying

it certainly would be fantastic if the SDK would have MAVLinkFTP support, but I'm afraid this wouldn't help much in the above matter. So, unless you see any better approach, I guess we'll have to postpone and just wait for the QGC guys. :)

hamishwillee, 2019-05-23:
> However, the above requires MAVLinkFTP only in second place, as it's just one option for fetching the xml. The main task would be to define proper MAVLink, messages, proper xml fields, and to have them digested by a GCS to produce a GUI, for which QGC was the obvious candidate.

FWIW I think defining the messages and xml definition could be done in parallel. You know what you want to do with it for gimbal and you also know that you'd like it to be still compatible/usable by camera.

I "guess" @julianoes wants to do MAVlink FTP in the SDK first, because he'd like to prototype the implementation of the messages end to end in the SDK rather than QGC.

Either way, the only reason not to do this in parallel is "insufficient resource". Right?

julianoes, 2019-05-28:
> Either way, the only reason not to do this in parallel is "insufficient resource". Right?

MAVLink FTP is on my and @MatejFranceskin's todo list but low priority (aka maybe sometime).

> FWIW I think defining the messages and xml definition could be done in parallel.

Yes definitely, again I'm reluctant to pick it up because I'm already doing other things and I should also focus.

hamishwillee, 2019-05-29:
> ... I'm already doing other things and I should also focus do other things I promised first.

Fair enough. I think reasonable to ETA when you might be able to look at this so that other interested parties aren't making assumptions. Alternatively to push back and ask them to define RFC/possible messages for review - review is usually easier to find time for that original thinking.

olliw42, 2019-05-29:
I maybe should reemphasize

the discussion in this thread is certainly highly interesting and touched lots of good points, but might have blurred a bit the original proposal and what's required for it. So, let's step back a bit. 

The original idea of this proposal was to establish Mavlink messages which allow a GCS to retrieve meta data on the parameters of a component, which the GCS can use to build a more user-friendly GUI for the handling of that component. The way of providing that meta data to the GCS would be a url to a definition xml file, analogously to the camera thing.

So, from the mavlink side, i.e. you, this actually only asks for (1) to express that such a mechanism is desirable and that Mavlink wants to support it, and (2) defining some messages. It does not ask for some hard coding work, like some MavlinkFTP SDK stuff.
 
My understanding is that point (1) has been answered with "yes", and as regards point (2) it was decided that we start with any message definition and converge it to a final while progressing.

So, and this is what I was trying to say in the last posts but might have miserably failed, at this point in time I don't see any need for any action as regards this proposal Mavlink wise. The next required steps are, in some ways, between QGroundControl and me (and whoever wants to join in). 

In short, no need for you guys to pick up anything. Sit back, and watch :)

:)

julianoes, 2019-05-29:
@olliw42 I agree with the plan, sounds great. Let me get the popcorn.

olliw42, 2019-06-06:
@julianoes just FYI, I've created a corresponding issue in the QGC repo to start the discussion with them (as you may also see from the reference I see is github has put up here)

:)

olliw42, 2019-06-07:
@LorenzMeier 
>Would you mind building out a full mini-example for a gimbal? 

@julianoes 
>The next required steps are, in some ways, between QGroundControl and me

just as status update
as you may see from the accompanying QGC issue, I've mocked up a component information request command and message, added support of it to the STorM32 gimbal controller, and produced component definition xml files. So, the first step in building an example and the first step of the plan decided on in the dev call has been done. I feel the ball is now in QGC's field. 
I've also produced a tentative page in the STorM32 wiki for reference, so the current capabilities could be looked up (http://www.olliw.eu/storm32bgc-wiki/MAVLink_Communication).
Finally, I also feel that it could be useful if I would show up at next week's dev call, also since I have seen that MatejF is mentioned who I came to believe is associated with QGC, but I will be in vacation until Fr (Pfingsten :):) ) and not able to attend. Hence this status update.
Have fun, guys. I will LOL.

julianoes, 2019-06-11:
@olliw42 nice thanks for the update! I need to have a closer look sometime soon!

hamishwillee, 2022-02-24:
Closing. This ended up being implemented as the Component information protocol.

closed 2022-02-24 by hamishwillee