mavlink/mavlink #1139: Parameter protocol: Behavior for empty lists not specified

olliw42, 2019-05-21:
The parameter protocol micro service has no specified behavior for components which do not have any parameters. 

Such components aren't as strange as it may sound. To just give a recent example I stumbled across: A mavlink camera which wants to use the camera microservice in it's most elementary form. 

I note that QGroundControl and MissionPlanner handle such components very differently, which also points towards that this loophole should be closed.

(GCS behaves gracefully while MP tries to seek params for nearly endless)(MP can be made to be happy, but this is then not according to spec)

hamishwillee, 2019-05-21:
You're right, this case is not covered. 

As it is now I would expect that no param value would be sent and the timeout should trigger. The diagram is missing the timeout shown in red (probably) and the explanation in the doc omits what happens after timeout (essentially retry on the request a number of times before failing).

[image]

What would be "right" is IMO, as you say, responding with param_count=0 if there are no params. 

@DonLakeFlyer does QGC ever request params off something that might not have params - what does it do?

DonLakeFlyer, 2019-06-04:
QGC sends PARAM_REQUEST_LIST to component id = 0 which means all components should respond. QGC always expects the main flight controller component (MAV_COMP_ID_AUTOPILOT1) to have at least 1 or more parameters. It will timeout and pop and error if this happens. For other components it doesn't care.

olliw42, 2019-06-05:
thx for the details on QGC
as mentioned, QGC behaves gracefully for parameter-less components, while MP doesn't

guys, simply just define how the zero-parameter case should be handled. GCS maintainers should then just implement this. End of story.

:)

(I also could produce text for why this is a text book example of what MAVLink is still missing to become a standard, and for why you just should define how the zero-parameter case should be handled ... LOL)

hamishwillee, 2019-06-11:
> guys, simply just define how the zero-parameter case should be handled. GCS maintainers should then just implement this. End of story.

It is not as "simple" as that. We need to be aware of the potential cost of breaking existing implementations. 

I have created a docs fix in https://github.com/mavlink/mavlink-devguide/pull/189 which generalised what QGC does. In summary, components should ignore the request for parameters if they have none, and GCS should timeout if it gets no parameters at all from any addressed component, and reset the state machine. It may also notify a user or whatever, but I see that as "implementation specific".

olliw42, 2019-06-18:
>It is not as "simple" as that. We need to be aware of the potential cost of breaking existing implementations.

frankly, the argument isn't thought out well

If "existing implementations" is suposed to mean ALL existing implementations, specifically QGC and MP, then your comment fails since the only way to make MP to work is to specify that each component must have at least one param, even if it's just a fake param, which obviously doesn't make sense. So, MP will have to be changed whatever, in contrast to your claimed goal. So, "existing implementations" at best could mean only the existing QGC implementation, in which case your comment fails since QGC already behaves gracefully and any change can be introduced now without any side effect as regards backwards compatibility. Q.E.D.

in drafting standards I would find it useful to make a bit more use of the principles of logical interference

:)

The suggested mechanism would avoid GCSes to hang up, but relying on only timeout isn't a good solution, as the comments in mavlink/mavlink-devguide#189 indicate.

closed 2019-07-01 by hamishwillee