mavlink/mavlink #1140: Responses to PARAM_EXT messages not fully specified

olliw42, 2019-05-21:
From the current spec one may conclude that the `PARAM_EXT_REQUEST_READ` and `PARAM_EXT_REQUEST_LIST` messages should return `PARAM_EXT_VALUE` messages, but this could be clearly spelled out (similarly as it is done for `PARAM_EXT_SET` and `PARAM_EXT_ACK`).

I would suggest to change the descriptions, e.g.

`PARAM_EXT_REQUEST_READ`: (Mavlink2) Request to read the value of a parameter with either the param_id string id or param_index. Responds with a PARAM_VALUE message if a valid parameter was indexed.

`PARAM_EXT_REQUEST_LIST`: (Mavlink2) Request all parameters of this component. After this request, all parameters are emitted via PARAM_VALUE messages.

`PARAM_EXT_VALUE`: (MAVLink 2) Response from a PARAM_EXT_REQUEST_READ or PARAM_EXT_REQUEST_LIST message, emiting the value of a parameter. The inclusion of param_count and param_index in the message allows the recipient to keep track of received parameters and allows them to re-request missing parameters after a loss or timeout.

:)

Btw, a statement like "If the current state is PARAM_ACK_IN_PROGRESS, you will accordingly receive a PARAM_ACK_IN_PROGRESS in response." doesn't read very meaningful to me, reads like "If the car is yellow, then the car is yellow" :)

hamishwillee, 2019-05-22:
Thanks. See #1143.

hamishwillee, 2026-08-25:
Fixed. here has been some iteration, but the work started in https://github.com/mavlink/mavlink/pull/1442

closed 2026-08-25 by hamishwillee