mavlink/mavlink #1141: The latest released version is six years old

amilcarlucas, 2019-05-21:
Please do a "Mavlink 1.0.12" release version. I am working on the product where I need to list the released version of all the software components that I use.
I would realy like to use a recent mavlink version.

I maybe have the permissions to do that myself. Should I try that?

hamishwillee, 2019-05-21:
If you use the current version it is the latest version - 6 years old or otherwise :-). 

I'd like an update too, mostly because there is some tidying we might reasonably do in a new release. But not just "in order to have a release". The trigger I was hoping for was @tridge making his concrete proposal for a new parameter protocol. 

Others might feel differently?

amilcarlucas, 2019-05-22:
I do vote for a release for the sake of releasing.
We can always do another release later.

amilcarlucas, 2019-05-22:
The dev call is today at 18H00 CET right?

hamishwillee, 2019-05-22:
Next week, 5pm AEST. Easiest way to check is the dronecode calendar linked from here: https://mavlink.io/en/about/support.html#schedule

hamishwillee, 2019-05-29:
Hi @amilcarlucas 

Done in https://github.com/mavlink/mavlink/releases/tag/1.0.12 - sorry for the delay.

We really need to clarify the release process and (possibly) co-ordinate source code, built library and subversion updates. That is deferred to #1149.

closed 2019-05-29 by hamishwillee

amilcarlucas, 2019-06-03:
Thanks !!!! that is a great help :)