mavlink/mavlink #1144: c_library_v1 MAVLINK_FRAMING_BAD_CRC Left Shift

tawheeler, 2019-05-22:
There may be an error in the Mavlink c library v1 in the [mavlink_helpers.h file](https://github.com/mavlink/c_library_v1/blob/master/mavlink_helpers.h):

```C
if (status->msg_received == MAVLINK_FRAMING_BAD_CRC) {
	/*
	  the CRC came out wrong. We now need to overwrite the
	  msg CRC with the one on the wire so that if the
	  caller decides to forward the message anyway that
	  mavlink_msg_to_send_buffer() won't overwrite the
	  checksum
	 */
	r_message->checksum = _MAV_PAYLOAD(rxmsg)[status->packet_idx] |(_MAV_PAYLOAD(rxmsg)[status->packet_idx+1]<<8);
}
```

Here, the intention is to overwrite `r_message->checksum` with a 2-btye checksum from `_MAV_PAYLOAD`. The second byte is left-shifted by 8 bits, when it appears it should be right-shifted instead. ([Left-shifting negative values results in undefined behavior](https://stackoverflow.com/questions/3784996/why-does-left-shift-operation-invoke-undefined-behaviour-when-the-left-side-oper)).

It may be sufficient to instead do:
```
r_message->checksum = *reinterpret_cast<const uint16_t*>(_MAV_PAYLOAD(rxmsg)[status->packet_idx]);
```

hamishwillee, 2019-05-22:
The left shift creates a double byte element with the bottom 8 bits zero-filled and the top byte containing the crc[1] byte from the message. This is then OR'd with the bottom byte (crc[0]) to create the full 2-byte CRC. 
We're not using the shift operator to do mathematical operations so I don't *think* the above discussion is relevant.

Thoughts?

tawheeler, 2019-05-22:
Sorry, left shift is correct. Edited the issue to reflect this.

We are getting undefined behavior warnings using the clang UndefinedBehaviorSanitizer on this line. Left shifting negative values is undefined.

hamishwillee, 2019-05-23:
So the problem is the warnings from the generator? Because the actual behaviour should be fine AFAIK.

Can you please repost/move this  issue to  https://github.com/ArduPilot/pymavlink - that is where the generator for this code exists.

tawheeler, 2019-05-23:
Thanks, I found the file in that repo.

closed 2019-05-23 by tawheeler