mavlink/mavlink #1150: common.xml: heartbeat field type for MAVLink version

SweeWarman, 2019-06-01:
I am implementing a MAVLink generator for Haskell and came across the following in common.xml in the definition of the heartbeat message:
 <field type="uint8_t_mavlink_version" name="mavlink_version">MAVLink version, not writable by user, gets added by protocol because of magic data type: uint8_t_mavlink_version</field>

Why is field type not just "uint8_t"? Is this a mistake? Could some one please let me know?

hamishwillee, 2019-06-02:
Because the value must be auto-filled by the prototol generator, and this makes it clear to the system the type of data and the value that is fetched (mavlink_version). The data IS a uint8_t. it is is pulled out of the version field of the top last included into a build.

amilcarlucas, 2019-06-11:
BTW please add that generator to pymavlink upstream repo. That way you do not need to rewrite the xml parser, only the haskell generator.

hamishwillee, 2019-06-27:
I'm closing this as answered. Please reopen if you have further questions

closed 2019-06-27 by hamishwillee

amilcarlucas, 2019-06-27:
The upstream pymavlink repository is https://github.com/ardupilot/pymavlink 
So far you have not done a pull request there. Please do so.

hamishwillee, 2019-06-28:
@amilcarlucas It is quite likely that @SweeWarman has not yet finished the implementation. 

@SweeWarman it is not compulsary in any way to integrate into mavgen, but this is the best way for end users because they know about the tool already, and it is good for you because much of the code for parsing the xml is already written. If you don't want to integrate with mavgen for whatever reason, you should still make sure that your tool gets listed in the docs https://mavlink.io/en/#supported_languages so that other people can find it.

SweeWarman, 2019-06-28:
I have a working version available at https://github.com/SweeWarman/HaskMavlink. I decided not to integrate this into mavgen. This was mainly because I wanted to simplify setting up the Haskell dependencies using the Haskell stack framework. I wrote a simple parser in Haskell. I am still working on adding unit tests and documentation. Will make sure it gets listed in the supported languages once I have the documentation and test cases complete.

hamishwillee, 2019-06-28:
Thanks very much @SweeWarman