mavlink/mavlink #1152: Pl confirm/clarify heartbeat type for components

olliw42, 2019-06-03:
I've pointed to this in the px4 discuss forum (https://discuss.px4.io/t/mavlink-support-for-the-storm32-gimbal-controller-how-to-make-it-compatible/10723) but it seems it hasn't much frequency, so pl allow me to put this up here:

There is a MAV_TYPE_GIMBAL value in the heartbeat enum (https://mavlink.io/en/messages/common.html#MAV_TYPE_GIMBAL), and it appears obvious (at least to me) that for a gimbal this should be used. However, the docu for the heartbeat type field says that 

>type: Type of the system (quadrotor, helicopter, etc.). Components use the same type as their associated system.

I must admit I fail to understand the rational, and would also see real issues associated to this. So:

1) I wonder if that is indeed correct and if it's not "just" the wording which is misleading, and thus would like to ask to check this.

2) If this is indeed how it's supposed to be, I would find it useful if there could be few statements in the docu explaining the rational/intention/purpose, e.g. in the heartbeat microservice section.

Many thx, Olli

hamishwillee, 2019-06-03:
Hi @olliw42 

So `MAV_TYPE` is the *system* type, where a system is made up of components. If the gimbal was a complete "standalone" system, somehow separate of an autopilot (ie not addressed via the autopilot), it would have the type [MAV_TYPE_GIMBAL](https://mavlink.io/en/messages/common.html#MAV_TYPE_GIMBAL).

However if the gimbal is a component of a system (as I believe is the case) then system MAV_TYPE is that of the bigger system - ie autopilot, and the component is derived from the component ID in the mavlink package - ie it has id [MAV_COMP_ID_GIMBAL](https://mavlink.io/en/messages/common.html#MAV_COMP_ID_GIMBAL).

Does that make sense?

And yes, that should absolutely be captured in https://mavlink.io/en/services/heartbeat.html , but I will await your response.

olliw42, 2019-06-04:
thx, sir

this is how I understand what is written in the docs

I can't say though that it makes sense to me. A steering wheel doesn't become a car because it is mounted in a car, it's still steering wheel.

Also, I can't see any reason for why one would want a gimbal to report itself as of type PX4 when it is connected to a PX4. I do see though issues with requiring this. So, if what you describe is indeed what is desired I would find it helpful to better understand the reason for doing so, e.g. what advantage it brings or what practical problem it whelps to solve.

hamishwillee, 2019-06-04:
@olliw42 The type is primarily used by a GCS for top level UI layout configuration for a whole system. It is really intended for vehicles (but there are some stand-alone cameras, antenna trackers, and other "things" that you might use on their own). It should have been named "system type" to make that more clear.

The concern I (now) see with this is that the gimbal probably wouldn't know what vehicle it is used on, and really shouldn't need to know because that would require some query mechanism or configuration in the gimbal itself.

Is this the sort of concern you were raising? Are there others, and if so, what?

olliw42, 2019-06-04:
you'll find some of my concerns in the px4 discussion post I mentioned before in point C (https://discuss.px4.io/t/mavlink-support-for-the-storm32-gimbal-controller-how-to-make-it-compatible/10723). But I guess I better should reproduce them here:

> This would IMHO represent a real issue.
>* First, in case the STorM32 is not connected to a flight controller one would want it to show type = GIMBAL.
>* Second, this requirement would imply that a component must wait for a heartbeat from the system to arrive, which implies that the component must know which heartbeat is the one from the system !!! I can’t see how that could be achieved without additional configuration parameters, which would be most inconvenient from a usability perspective.
>* Third, I can’t see a generic way for the STorM32 to decide if its the first or the second situation, which would imply the need for further configuration parameters, which would be most inconvenient from a usability perspective.
>* Fourth, this requirement actually doesn’t make any sense to me. Why should a component not be allowed to tell what sort of a kind it is. This loss of information can also create issues, e.g. it might be useful to figure out which ID/messages are related to the gimbal, which would not be possible.
>
>My conclusion would be that the type should be GIMBAL in all cases.

_Comment: The 3rd sentence in the fourth point is probably incorrect, I haven't fully understood at this point in time that the unique identification is realized now by the component id._

So, yes, what you describe is the sort of concerns I see.

And again, I don't yet see any good reason for doing the type change. I fully understand that a GCS wants to know what system there is, but I can't see how that would expand to components such as gimbals. I mean, if a GCS would produce a GUI  for let's say a gimbal it would rather want to know if it is a Gremsy, Alexmos, or STorM32. So, your GUI argument - which I haven't considered before - also would rather point to not doing what the spec appears to currently suggest.
(and note, the GCS would know to which autopilot a gimbal is connected to from the system id, so this info is available without a need for a gimbal to announce itself as PX4 or ArduPilot).

hamishwillee, 2019-06-04:
Reproducing relevant comment from discussion board:

----

C. HeartBeat, type field

So far the STorM32 used it’s own value in the heartbeat’s type field. However, there is now a MAV_TYPE_GIMBAL value in the enum (https://mavlink.io/en/messages/common.html#MAV_TYPE_GIMBAL) and it appears obvious that this should be used.

However, the docu also says

type: Type of the system (quadrotor, helicopter, etc.). Components use the same type as their associated system.

This would IMHO represent a real issue.

1. First, in case the STorM32 is not connected to a flight controller one would want it to show type = GIMBAL.
1. Second, this requirement would imply that a component must wait for a heartbeat from the system to arrive, which implies that the component must know which heartbeat is the one from the system !!! I can’t see how that could be achieved without additional configuration parameters, which would be most inconvenient from a usability perspective.
1. Third, I can’t see a generic way for the STorM32 to decide if its the first or the second situation, which would imply the need for further configuration parameters, which would be most inconvenient from a usability perspective.
1. Fourth, this requirement actually doesn’t make any sense to me. Why should a component not be allowed to tell what sort of a kind it is. This loss of information can also create issues, e.g. it might be useful to figure out which ID/messages are related to the gimbal, which would not be possible.

My conclusion would be that the type should be GIMBAL in all cases.

olliw42, 2019-07-09:
comment:
in a longer phone discussion with @julianoes we both concluded (as reported in #1130): 

MAV_TYPE vs componentID for component identification: agreement was reached that the component is identified via MAV_TYPE (it thus must be set properly mandatorily), while the specification for the componentID values are the suggested default values.

in #1130 there is also some more related discussion

comment:
the existing MAV_TYPE_GIMBAL is marked as for standalone gimbals

julianoes, 2019-07-09:
@hamishwillee I agree with @olliw42 and I had also confirmed that talking to @LorenzMeier yesterday.

I think a ground station should be able to identify a drone including an autopilot, gimbal and camera correctly by parsing the 3 heartbeats it receives. Given it's not just a gimbal and not just a camera, it presumably is this "drone thing" but includes the other components.

hamishwillee, 2019-07-10:
@julianoes Just because this discussion is so long, some clarification/confirmation.
1. Currently componentID is one of [MAV_COMPONENT](https://mavlink.io/en/messages/common.html#MAV_COMPONENT) - this does not change, so a gimbal will use MAV_COMP_ID_GIMBAL.
2. Currently [HEARTBEAT.type](https://mavlink.io/en/messages/common.html#HEARTBEAT) ` (of type [MAV_TYPE](https://mavlink.io/en/messages/common.html#MAV_TYPE)  is marked as being the system type, even for components. The change is that every component will emit its own type, not the type of the system.

This makes sense to me based on the previous discussion that a component doesn't know what component it is part of. If the GCS has a problem it can work out the system from the system id and autopilot type.

Is above correct (ie I modify HEARTBEAT.type, MAV_TYPE (and enums, as appropriate)?

olliw42, 2019-07-10:
the above was meant to say:
1. any component MUST set it's MAV_TYPE to what it is 
2. any component MAY use MAV_COMPONENT as component ID, but one is free to choose any other component ID
=> type identification is done by only MAV_TYPE, not component ID 

currently MAV_TYPE_GIMBAL is spec-ed to be for standalone gimbals only. So one either adds a new type for system gimbals or changes that spec so it can indicate any gimbal

hamishwillee, 2019-07-11:
> any component MAY use MAV_COMPONENT as component ID, but one is free to choose any other component ID

> type identification is done by only MAV_TYPE, not component ID

>  " while the specification for the componentID values are the suggested default values."

@julianoes That will break a lot of routing and other code. Component ID is not AFAIK a "suggestion".

hamishwillee, 2019-07-11:
Also, still a bit confused about MAV_TYPE for the FC. My assumption being that this uses the type of the vehicle itself?

So imagine I have a VTOL titlerotor with the following components - is the mapping of MAV_TYPES correct?

COMPONENT | MAV_TYPE
--- | ---
FC | MAV_TYPE_VTOL_TILTROTOR
Companion computer |  MAV_TYPE_ONBOARD_CONTROLLER
Camera | MAV_TYPE_CAMERA
Gimbal | MAV_TYPE_GIMBAL
FLARM | MAV_TYPE_FLARM

What if they are a servo, OSD, VIO, GPS, and other types in [MAV_COMPONENT](https://mavlink.io/en/messages/common.html#MAV_COMPONENT) that aren't in type?

Basically I see the MAV_TYPE as reflecting a type of hardware while the MAV_COMPONENT is a specific instance. The FC is an autopilot, but as the "hub" we call it the actual physical hardware. 

Is that what you were thinking?

julianoes, 2019-07-11:
We only need to fall back to component IDs that are not in the standard if it's really not available. E.g. you have two gimbals and there is only one gimbal component. Of course we should add GIMBAL1, etc. for that but you get my point.

> That will break a lot of routing and other code. Component ID is not AFAIK a "suggestion".

I think we'll have to fix this one by one over time. I can't think of too many instances.

olliw42, 2019-07-11:
> What if they are a servo, OSD, VIO, GPS, and other types

why not add a type SERVO, OSD, VIO, GPS, and other types ??? 
as much as there are many autopilot types, and there also could be a COMPONENT_GENERIC type as much as there is a GENERIC type for autopilots, and so on ...

> We only need to fall back to component IDs that are not in the standard if it's really not available. E.g. you have two gimbals and there is only one gimbal component. Of course we should add GIMBAL1, etc. for that but you get my point.

the problem of which is that it will easily overflow .... 6 cameras, 6 gimbals, 6 servos, 6 osds, 6 vios, 6 gpses, 6 ... ... how much easier and robust and more logical it appears to be to have 1 x camera type, 1 x gimbal type, 1 x servo type, 1 x osd type, 1 x vio type, 1 x gps type, and to assign each **actually present** component in the system an ID ...

it's certainly good to suggest default IDs as it will make setup simpler, more plug&play, in many cases, but it can't be a substitute for proper type-ing.

hamishwillee, 2019-07-12:
@julianoes @olliw42 Can you confirm my mapping in the table above: https://github.com/mavlink/mavlink/issues/1152#issuecomment-510304148

> the problem of which is that it will easily overflow .... 6 cameras, 6 gimbals, 14 servos, 6 osds, 6 vios, 6 gpses, 6 imus, 6 ... ... how much easier and robust and more logical it appears to be to have 1 x camera type, 1 x gimbal type, 1 x servo type, 1 x osd type, 1 x vio type, 1 x gps type, and to assign each actually present component in the system an ID ...

This I agree with. A type, and a unique component id that indicates the instance. This then means that you have as many components of a type as you like, provided the max doesn't go over 255.

I'm also OK with the fundamental idea:
*I think a ground station should be able to identify a drone including an autopilot, gimbal and camera correctly by parsing the 3 heartbeats it receives. Given it's not just a gimbal and not just a camera, it presumably is this "drone thing" but includes the other components.*

The main problem I have is using MAV_TYPE. It is clearly intended for vehicle frames, and is being used for that purpose. HOw do you differentiate the autopilot? If you do differentiate the autopilot, how do you differentiate the frame? Have you spoken to Don about what this will do to QGC?

It feels like the right thing to do is to change "Type" as you wish, but have a new field for the frame type.

julianoes, 2019-07-12:
> @julianoes @olliw42 Can you confirm my mapping in the table above: #1152 (comment)

I'd say yes.

> It feels like the right thing to do is to change "Type" as you wish, but have a new field for the frame type.

I don't understand what you mean exactly. And I don't think we can have a new field at that point.

My point is that a camera doesn't know that it is part of a drone and neither does the gimbal, so how would it know to send the type accordingly?

hamishwillee, 2019-07-15:
> My point is that a camera doesn't know that it is part of a drone and neither does the gimbal, so how would it know to send the type accordingly?

Why? You **have to configure** the system ID and component ID in the component - how is telling the camera what system and ids it has to use any different from telling it that also must use a particular type?

I'm not saying that it is desirable, I just don't understand why it is a justification for such a big change in behaviour.

I don't like that MAV_TYPE is inconsistent/different for some components than others - ie for a gimbal you are a gimbal, for a camera you're a camera, but for an autopilot you're a VTOL, plane, copter, blah blah blah. But we could live with this it if it seen to make sense. We just need to be very clear that it is inconsistent. 

We do need to confirm how instances will be handled.

I'm uncomfortable with the idea of getting rid of instance ids in MAV_COMPONENT or making the whole set of ids randomly allocated. I can't help thinking there will be a case where we will want to be able to know that SERVO3 component is SERVO3. 

Specifically, how will [component IDs](https://mavlink.io/en/messages/common.html#MAV_COMPONENT) and routing work?
  - Is it that they will be randomly allocated, or are they still as is for the currently defined versions?
  - How is routing to all vs a particular instance to be handled? Ie right now if I want to route a message to a particular camera instance I use its component ID. In future I think you're suggesting that I will have to accumulate all the component ids that are cameras based on MAV_TYPE in heartbeat and assign them all some instance number and then address them? If I want to send to everything perhaps I just use "0" as currently (that works for me!)

> I don't understand what you mean exactly. And I don't think we can have a new field at that point.

This is a bit of an aside. Because I don't like the idea of MAV_TYPE being inconsistent I was proposing that we split the concepts of "airframe type" (which is what MAV_TYPE has historically been used for) and "component type". So we might have `HEARTBEAT.component_type`, and `HEARTBEAT.airframe_type`. Airframe could be optional if it isn't relevant. 
Appreciate this is a massive break to do "right" at the heart of the protocol. 

In any case, before we move ahead we need to confirm **exactly** how this will work. This is a significant enough change that I'd say an RFC is needed.

I'd also want the GCS guys to have input on this - in particular @DonLakeFlyer.

julianoes, 2019-07-15:
> I'm uncomfortable with the idea of getting rid of instance ids in MAV_COMPONENT or making the whole set of ids randomly allocated. I can't help thinking there will be a case where we will want to be able to know that SERVO3 component is SERVO3.

That's not what I'm trying to say. It all stands as it is. My comment only applies if the component ID is already taken and there is only one and you want to add a second one. So this would be the obscure case of two charging stations or something else that we don't (want to) cover in the component spec.

> Why? You have to configure the system ID and component ID in the component - how is telling the camera what system and ids it has to use any different from telling it that also must use a particular type?

I would configure a camera to be COMP_CAMERA0 and that should work for most systems that I integrate it into. However, when working on the camera firmware I don't yet know if the camera will be attached to a quadcopter or a plane.

I would think that a QGS can easily show the UI for a standalone camera if it receives heartbeats from a camera only and none from an autopilot. Once it receives the heartbeats for the autopilot as well, then it knows it's connected to an integrated thing given the components have the same system ID.

hamishwillee, 2019-07-16:
> I would configure a camera to be COMP_CAMERA0 and that should work for most systems that I integrate it into. However, when working on the camera firmware I don't yet know if the camera will be attached to a quadcopter or a plane.

So what are you going to do if it **has to be** camera1-6? How are you going to set the system ID? The point is that when you're working on the firmware you don't know about the system ID or the component ID either. But you can still set them - so why not the parent?

A good counter argument for this change is that there is now less for an end user to configure. By setting the default component ID to the first ID then *in the most common case* they only need to set a system ID. That is indeed better.

Under the current definition you can always get the parent type from the heartbeat and the component type from the component ID. You also immediately know if something is a component or a standalone system.

Under the proposed definition you have to accumulate heartbeats to get the parent type, and to determine if an item is a component or a standalone system. There are tradeoffs, and they aren't all great.

> I would think that a QGS can easily show the UI for a standalone camera if it receives heartbeats from a camera only and none from an autopilot. Once it receives the heartbeats for the autopilot as [...]

I'm sure it can, but I suspect with more complication. But I wouldn't make the change without confirmation from @DonLakeFlyer 

Anyway, can you confirm the summary below:

----

1. `HEARTBEAT.type` and `MAV_TYPE` are the type of the component sending the message (previously the type of the airframe/system owning the component).
   - A flight controller/stack (PX4) exports its type as an ID for the aiframe - e.g. `MAV_TYPE_VTOL_TILTROTOR` rather than a specific type for an autopilot. This is how QGC determines the type of the whole vehicle for layout. See table below for example.
   - A GCS is expected to work out something is a component if it discovers other components with the same system id, and that it is standalone if it doesn't discover components with the same system id.
   - There are some missing types we will probably need to add for components that are not in MAV_TYPE - e.g. SERVO, OSD, VIO, GPS
   - There are no instance types. Something is a gimbal or a camera. To get whether it is camera 1 or whatever you have to check the component id.

COMPONENT | MAV_TYPE
--- | ---
FC | MAV_TYPE_VTOL_TILTROTOR
Companion computer |  MAV_TYPE_ONBOARD_CONTROLLER
Camera | MAV_TYPE_CAMERA
Gimbal | MAV_TYPE_GIMBAL
FLARM | MAV_TYPE_FLARM

2. System ids are still used and must be allocatable to the component (no change)
3. Component IDs **must** still use the values defined in MAV_COMPONENT (no change). Components are expected to set the first instance by default (that is new text)

olliw42, 2019-07-16:
with all due respect, this is a very weird discussion, going in circles, overthrowing insights&arguments which were gained earlier later on, clouding discussion by bringing in all sorts of things which don't relate, and so on. I really don't see that things converge towards something better, but rather goes from flawed to flawed. The general problem is that thinking is not in terms of general concepts but in terms of a bucket of wishes, from which a collection of wishes is draw like in lottery.
I would wish I could be helpful in anyway ... but I don't see that there could be anything one could say and admit I feel helpless. So, I drop out and regret having raised that issue at all. 
Probably you should just NOT CHANGE anything and leave it as it is. Currently this looks to be the MUCH better solution.
I appologize.

hamishwillee, 2019-07-16:
@olliw42 This is frustrating for me too. But the suggested changes and "insights" look half thought out. There are dependencies beyond us that are being ignored - ie QGC.

I may just be too stupid to understand why this is better, but if I don't understand something I won't merge it. 

@julianoes Perhaps the best solution is that you take this forward without me. If you're convinced it is the right thing to do, perhaps create your own PRs and ask Lorenz to merge them? I'm hands off.

julianoes, 2019-07-18:
I've added the label dev call, that's probably the only way to resolve this.

> So what are you going to do if it has to be camera1-6? How are you going to set the system ID? The point is that when you're working on the firmware you don't know about the system ID or the component ID either. But you can still set them - so why not the parent?

I assume with parent you mean `MAV_TYPE`. Let's say you make a camera which can be added to a quadcopter or a VTOL.
- If all vehicles in your fleet have sysid 1 it's easy. You set sysid 1 and compid CAMERA1. `MAV_TYPE` you can't know because you don't know what it is mounted on.
- If the sysid is set random it's a bit harder. Either you need to make sure the camera has the same sysid when you add it, or the camera listens to the heartbeat from the autopilot and uses the same sysid. It could also adapt `MAV_TYPE` at this point. However, this is more to think of when writing this part of the code. And you need to make sure that you don't re-use a `MAV_TYPE` of something that you're not supposed to. E.g. the camera should not copy the companion computer's `MAV_TYPE`. Also, what is the camera supposed to do when the autopilot is updating and not sending heartbeats anymore for a while? Is the camera now a standalone camera and supposed to change `MAV_TYPE`, or does it keep the previous one? This all seems very clunky to me and therefore hard to communicate to gimbal/camera manufacturers.

> Under the proposed definition you have to accumulate heartbeats to get the parent type, and to determine if an item is a component or a standalone system. 

Correct. I think that's what is done already. It's just currently based on compid but it might as well be done based on MAV_TYPE in the future.

olliw42, 2019-07-18:
@hamishwillee
I hope you are not taking my comments personally, they are not meant that way. I just look at things. Like then I say "The earth is a sphere and not flat" it's not meant as to insult the flatearthers, but as a statement of what I perceive as correct.
I'm well aware that also my comments aren't always perfectly thought out either, I however prefer discussions which bring things closer towards "fully thought out", I'm just this sort of person, an dthe present one I unfortunately don't see that way. That's all. 

Over time I somewhat lost confidence that this mostly written communication is always helpfull, I actually believe that many of this sort of discussions would be quite easily concluded if one were talking to each other face to face, maybe with a good beer nearby ... especially with a good beer nearby ...
so, hope to see you for a beer some time
https://www.rclineforum.de/forum/wcf/images/smilies/prost.gif (you have to click on this)

julianoes, 2019-07-18:
I agree and that's why I suggest to talk it through in the dev call.
And for other sync (or almost sync) conversations we usually use our Slack group. We could talk about it in the #mavlink channel there but I myself probably prefer GitHub and the call.

hamishwillee, 2019-07-18:
@olliw42 I was taking your comments personally, but that says more about my frustration with all the other things I am trying to do at the moment, and the time this is taking than anything you are doing. Need to take a pause and a deep breath. 

@julianoes Yes, dev call. Can we get @olliw42 and @DonLakeFlyer there? 
Re  https://github.com/mavlink/mavlink/issues/1152#issuecomment-512769907 you're aware that sysid can be reallocated by the GCS at any time right?

julianoes, 2019-07-19:
> Re #1152 (comment) does this make sense given the sysid of any system an be reallocated by the GCS at any time and that their can be multiple sysids in a network.

Yes, either all parts need to be changed manually or only the autopilot and then the camera adapts.

DonLakeFlyer, 2019-07-19:
So I tried reading through this and I'm pretty lost trying to piece together the end result of what is the proposed changes. Just from a QGC standpoint it only looks at MAV_TYPE on the main HEARTBEAT which has a component id of MAV_COMP_ID_AUTOPILOT1. This then sets the vehicle type for ui.

There is also a hack in QGC to identify a PX4 Flow.

```
void MultiVehicleManager::_vehicleHeartbeatInfo(LinkInterface* link, int vehicleId, int componentId, int vehicleFirmwareType, int vehicleType)
{
    // Special case PX4 Flow since depending on firmware it can have different settings. We force to the PX4 Firmware settings.
    if (link->isPX4Flow()) {
        vehicleId = 81;
        componentId = 50;//MAV_COMP_ID_AUTOPILOT1;
        vehicleFirmwareType = MAV_AUTOPILOT_GENERIC;
        vehicleType = 0;
    }

    if (componentId != MAV_COMP_ID_AUTOPILOT1) {
        // Special case for PX4 Flow
        if (vehicleId != 81 || componentId != 50) {
            // Don't create vehicles for components other than the autopilot
            qCDebug(MultiVehicleManagerLog()) << "Ignoring heartbeat from unknown component "
                                              << link->getName()
                                              << vehicleId
                                              << componentId
                                              << vehicleFirmwareType
                                              << vehicleType;
            return;
        }
    }

    if (_vehicles.count() > 0 && !qgcApp()->toolbox()->corePlugin()->options()->multiVehicleEnabled()) {
        return;
    }
    if (_ignoreVehicleIds.contains(vehicleId) || getVehicleById(vehicleId) || vehicleId == 0) {
        return;
    }

    switch (vehicleType) {
    case MAV_TYPE_GCS:
    case MAV_TYPE_ONBOARD_CONTROLLER:
    case MAV_TYPE_GIMBAL:
    case MAV_TYPE_ADSB:
        // These are not vehicles, so don't create a vehicle for them
        return;
    default:
        // All other MAV_TYPEs create vehicles
        break;
    }

    qCDebug(MultiVehicleManagerLog()) << "Adding new vehicle link:vehicleId:componentId:vehicleFirmwareType:vehicleType "
                                      << link->getName()
                                      << vehicleId
                                      << componentId
                                      << vehicleFirmwareType
                                      << vehicleType;
```

julianoes, 2019-07-22:
> Just from a QGC standpoint it only looks at MAV_TYPE on the main HEARTBEAT which has a component id of MAV_COMP_ID_AUTOPILOT1. This then sets the vehicle type for ui.

This means to me that the proposed change would not affect QGC.

hamishwillee, 2019-08-01:
> This means to me that the proposed change would not affect QGC.

Me too @julianoes 

I have created https://github.com/mavlink/mavlink/pull/1188 to get this in. Merge if you're happy.

olliw42, 2019-08-02:
thx for the changes and PR.

I'm happy to see that it says "All other components must report a value appropriate for their type", with emphasis on the word "must". 

I'd like to emphasize again that a strict corollary of this is that e.g. a GCS must use only MAV_TYPE for determining the type in order to e.g. adjust UI features and such things, but must NOT rely on the ComponentID in any way. That is the rule is:

MAV_TYPE overrules ComponentID.

Since this might not be immediately obvious even though it's a corollary, I think this should be documented at some place, e.g. in the heartbeat/connection protocol page.

I also would add in the appropriate place(s) that a non-flight controller component should have MAV_AUTOPILOT set to MAV_AUTOPILOT_INVALID. (if I'm not mistaken this was agreed to before somewhere else, I just forgot where this was).

With respect to the QGC snipet given in the above this would imply two changes
* it should look for ONLY MAV_TYPE and not for MAV_TYPE AND MAV_COMP_ID_AUTOPILOT1 to set the vehicle type for ui. (this should be a totally non-breaking change!)
* the code could thus be improved/streamlined by simply checking for MAV_AUTOPILOT != MAV_AUTOPILOT_INVALID to identify if a component is a vehicle. If fear is that not all components adhere to this, one of course could keep additional checks like the one already present. But note that it would be an improvement, since currently a MAV_TYPE_CHARGING_STATION would also be handled as vehicle ...

olliw42, 2019-08-02:
> Yes, dev call. Can we get @olliw42 and @DonLakeFlyer there?

I'm sorry to have not followed up, but I was really busy. If usefull I can however attend the next dev call, without any time constraints :), pl just indicate so if a relevant topic is going to be discussed.

hamishwillee, 2019-08-05:
Thanks @olliw42 

> MAV_TYPE overrules ComponentID.

Sure. It certainly makes sense for us to make it clear which should be used to remove ambiguity (though this will break anything that currently uses component id). I added a note to the `HEARTBEAT.type` definition and will also update the other docs when this goes in. 

> I also would add in the appropriate place(s) that a non-flight controller component should have MAV_AUTOPILOT set to MAV_AUTOPILOT_INVALID. (if I'm not mistaken this was agreed to before somewhere else, I just forgot where this was).

That's a new issue. So new PR: https://github.com/mavlink/mavlink/pull/1193. I'll just merge that it - it should be self-evident.

> it should look for ONLY MAV_TYPE and not for MAV_TYPE AND MAV_COMP_ID_AUTOPILOT1 to set the vehicle type for ui. (this should be a totally non-breaking change!)

That's up to the GCS guys, and yes it would make sense eventually.

But the thing to remember is that this change will take a while to propagate through various firmwares. So PX4flow may advertise its type incorrectly for a while, but still use the correct component ID.

hamishwillee, 2019-08-05:
@olliw42

olliw42, 2019-08-05:
thx for the feedback :)

olliw42, 2019-08-08:
issue addressed so I close it. THX.

closed 2019-08-08 by olliw42