mavlink/mavlink #1156: Pl clarify if a component is required to support AUTOPILOT_VERSION

olliw42, 2019-06-04:
this is in reference to https://discuss.px4.io/t/mavlink-support-for-the-storm32-gimbal-controller-how-to-make-it-compatible/10723/8, last point.

ArduPilot has an AUTOPILOT_VERSION_REQUEST message in it’s dialect, which is used by MissionPlanner to query for AUTOPILOT_VERSION, and MissionPlanner kind of requires a component to respond to this request in order to work properly. This isn't so for QGroundControl.

This thus raises the question if a component should be required to support AUTOPILOT_VERSION for proper functioning, or not, and if so this would need to be clarified by the spec.

Comment: For PROTOCOL_VERSION the specification unambiguously states that every component should support it. However, AUTOPILOT_VERSION and PROTOCOL_VERSION kind of serve a similar purpose in that they both can be used to determine the MAVLink version. Requiring support of both thus appears to be somewhat superfluous. However, not requiring AUTOPILOT_VERSION would kind of break compatibility with current MissionPlanner.

amilcarlucas, 2019-06-11:
I guess this boils down to: which messages should be moved to minimal.xml.

By moving them would make it 100% clear what are minimal set of messages that a system needs to understand in order to be able to say that it is mavlink compatible !!

hamishwillee, 2019-06-27:
My take is that implementing AUTOPILOT_VERSION is not mandatory. Everything needs to be able to confirm it's version, but all the other information in AUTOPILOT_VERSION is only necessary for an autopilot node. I'm sure we could find arguments either way, but let's try keep minimal.xml as minimal as possible :-)

@amilcarlucas Good point re minimal.xml - we should move PROTOCOL_VERSION in there too. What we need to do before that is make minimal.xml an include file for common.xml. I had intended to do this some time ago, but was waiting for deeper nesting of includes to be supported. I worry this will break things. Will put back on my list

hamishwillee, 2019-06-27:
PS. Closing. I believe this is answered.

closed 2019-06-27 by hamishwillee

olliw42, 2019-06-27:
many thx for the clarification

could this in some way be unambiguously indicated in the docs? 

The point is that MissionPlanner/ArduPilot use this message and MissonPlanner in fact relies on every component to support it. I doubt that MissionPlanner/ArduPilot will replace it by PROTOCOL_VERSION, but it should be at least made clear (to them ;)) that this is not according to spec, and MissionPlanner might be motivated by this to actually adapt it's behavior.

:)

hamishwillee, 2019-06-28:
@olliw42 I'd happily do so, but I'm not sure of the best way to approach this. I don't think I can or should mark up this particular message as "this is not mandatory" because very little is mandatory, and in theory I'd then have to mark up everything else.

Perhaps a better approach might be to take @amilcarlucas lead and more clearly document what IS mandatory - ie better explain the minimal.xml set.

olliw42, 2019-06-28:
yeah

I guess it could be added in the version negotating page, https://mavlink.io/en/guide/mavlink_version.html
which btw is still highly confusing
e.g. if you read it, you'll find AUTOPILOT_VERSION been mentioned quite frequently, and moreover usually even before PROTOCOL_VERSION ... one easily could get the impression from that page that AUTOPILOT_VERSION is to be preferred

so, one could e.g. in the first section Determining Protocol/Message Version, move AUTOPILOT_VERSION to the latest and mark it, e.g.

A library's MAVLink support can be determined in a number of ways:
* PROTOCOL_VERSION: blablabla
* HEARTRBEAT: blablabla 
* The major version can be determined from the packet start marker byte: blablabla
* AUTOPILOT_VERSION: blabla. But note that a vehicle is not required to support AUTOPILOT_VERSION, and implementations must not rely on it, in contrast to PROTOCOL_VERSION which vehicles are required to support.

In Version Handshaking the reference to AUTOPILOT_VERSION, and even in the first line, must then go. One could add this in at the end of this subchapter, as a note, that AUTOPILOT_VERSION also could be used and that this might be sufficient for a transparent link, but that this isn't reliable since a vehicle isn't required to support it.

in Negotiating Versions AUTOPILOT_VERSION is even mentioned twice. In the first occurrence one could change the text to e.g. "Vehicles which support the AUTOPILOT_VERSION message should set the ....". In the second occurrence AUTOPILOT_VERSION must be replaced by PROTOCOL_VERSION

I guess I also would greep the whole docs for AUTOPILOT_VERSION and see if it's being used in a confusing manner in other places too