mavlink/mavlink #1159: Proposal for an opaque, tunnel message

olliw42, 2019-06-05:
The proposal is to establish a MAVlink message which allows a component to send specifc, non-standardized data to another component. 

Currently this is not possible in a MAVLink-ish way, and even adding one owns message to one owns dialect to both components typically won't work when the data transmission involves some "router", which is frequently the case. For instance: GUI to autopilot to component. The "router" or autopilot won't be able to forward since it wouldn't know about the new message and disregard it. A standardized message for transmitting non-standardized data would resolve this.

I'm not sure I should present arguments and/or examples for why such a capability is needed. I suspect that such a discussion could quickly become long and less useful. So, for briefness I assume that a majority will agree with the need, and enhance the point by mentioning that MAVLink actually already has this sort of message in other context:

An obvious one is `FILE_TRANSFER_PROTOCOL`, which is very explicitly described as opaque. In fact, the above task could be achieved with `FILE_TRANSFER_PROTOCOL`. However, I feel that this would be a bit of a hack and dirty misuse, given that even if it is opaque it serves the specific purpose of FTP. Also messages such as `GPS_INJECT_DATA` or `SERIAL_CONTROL` should be considered part of this class. 

Finally, there is `V2_EXTENSION`. In fact, this message and its accompanying specification for message_type is, except of the target_network field, essentially exactly what I would have suggested! However, it feels strange. First, the name. Second, the description "Message implementing parts of the V2 payload specs in V1 frames for transitional support". Doesn't sound like long-living and general purpose. Third, I've tried to find the extension-message-ids.xml, but failed, which makes me further wonder what the status of the `V2_EXTENSION` message is.

Hence this proposal.

I propose a message, maybe named `TUNNEL`, with target fields, a protocol field which is exactly handled like the message_type field of `V2_EXTENSION`, and an opaque data array. I would not have added a target_network field, simply because its use is beyond my imagination, but if the consensus would be that it might bring some advantages it of course should be added, in which case the new message would be essentially identical to `V2_EXTENSION`. 

The outcome of this proposal of course also could be to solidify `V2_EXTENSION` and establish this message for the said purpose, and to add its use as opaque tunnel message to the message description, or change the message description completely in this sense.

Cheers, Olli

hamishwillee, 2019-06-06:
A couple of points
- Routers are supposed to forward unknown messages. The fact that they don't is a flaw in implementations (including the C library). There are issues in place to get this fixed. Some routers handle this properly - e.g. the ESP8266
- `FILE_TRANSFER_PROTOCOL` is a badly designed message. It should not be opaque.
- Re TUNNEL What if the message to be packaged is bigger than the available payload?
- Adding support to unpackage messages from a tunnel would probably take more memory overhead and coding effort than just supporting forwarding in our main libraries, right?

Let's see what others say.

olliw42, 2019-06-06:
>Routers are supposed to forward unknown messages. The fact that they don't is a flaw in implementations (including the C library).

One only can strongly disagree. This is NOT a flaw in the implementations but a fundamental flaw of MAVLink, and a stupid one. I find it unacceptable to blame the implementations and not the real reasons. Doing so has nothing to do with rational thinking.

Routers II: That they don't is experimental fact and isn't going to change anytime soon.
 
> Re TUNNEL What if the message to be packaged is bigger than the available payload?

really? Hint: isn't it magic we just now can communicate with fixed/limited-sized packeted protocols ... :)

> FILE_TRANSFER_PROTOCOL is a badly designed message. It should not be opaque.

there is absolutely nothing wrong with opaque messages. All the world works with opaque messages. And in fact, all important successful protocols (inet, USB, wifi, ..., ..., ...) work with opaque content. I argue it's exactly why they are successful.

>Adding support to unpackage messages from a tunnel would probably take more memory overhead and coding effort than just supporting forwarding in our main libraries, right?

really? How could it be more harmful than any of the other "big" messages (of which you have plenty)? And even if you were right, which you are not, one easily could set it to 144. 
How could it be simpler than just to add an entry to common.xml?
And even if the forwarding-of-unknown issue would not exist, a tunnel message as part of common.xml would still make lots of sense. Example: see the short discussion about tighter gimbal-autopilot integration in the DO_MOUNT sign PR. See USB (which some of you seem to like to quote a lot ;)).

:)

And finally, as I've said, if the need and sense of such a message is indeed seriously questioned, then it's really not useful to dispute it here. I would think that a simple yes/no vote of you guys in the next dev call  would then be the most sensible and fastest approach to settle. And if it is indeed questioned I will happily go away and close this issue :)

hamishwillee, 2019-06-06:
Freely admit I am not an expert. But since I don't get the impression that I'm allowed to disagree, I'll take your direction and leave for the dev call.

maowen, 2019-08-01:
Any updates on this proposal?
Was this discussed at the MAVLink meeting?

I too am interested in this an opaque tunnel message. I am aware of other people that are currently doing this with the `FILE_TRANSFER_PROTOCOL` message.

I agree with @olliw42 that the `V2_EXTENSION` message is designed exactly as I would want a tunnel message. 

My belief is that there are many users who end up creating custom messages to transmit opaque data when a tunnel message would suit them fine.

@olliw42 - On a related note, you may be interested in knowing there is bug in PX4 mavlink routing/forwarding implementation that prevented the PX4 from forwarding messages that included the target system or target component routing fields. I submitted a pull request to correct this here: https://github.com/PX4/Firmware/pull/12559

hamishwillee, 2019-08-01:
Yes this got discussed. I believe he result was that we'd rather fix up the routing so that the library doesn't automatically swallow messages that it can't process locally. @LorenzMeier proposed a fix, but work has stalled https://github.com/mavlink/mavlink/issues/1166

olliw42, 2019-08-02:
providing a canonical method to get around the routing issue in some cases is just one possible reason for wanting an opaque tunnel message, but by far not the only one !!!

Therefore, whatever the conclusion and steps as regards the routing thing are, the question of whether an opaque tunnel message could or should be added to the standard message set isn't really related to it, i.e., the proposal would remain even if the routing things would be settled satisfactorily. 

IMHO.

My question would be, what are really strong reasons for not adding it? I can't see that it would do any harm but only can see that it would do some good. :)

LorenzMeier, 2019-08-02:
Please put up a PR for a complete implementation (tunnel setup and the tunnel framing message). I think it is needed and worthwhile. MAVSDK would be the perfect environment to test it.

olliw42, 2019-08-02:
> Please put up a PR for a complete implementation (tunnel setup and the tunnel framing message)

sorry if that is a silly question, but I'm honestly unsure: Is this directed at me, hamishwillee, or whoever else who wants to come forward ?

:)

hamishwillee, 2019-08-05:
> sorry if that is a silly question, but I'm honestly unsure: Is this directed at me, hamishwillee, or whoever else who wants to come forward ?

@olliw42 It is directed at anyone who wants to come forward. 

Generally:
- I'd _expect_ anyone who wants to work on this to come forward and say so here (to avoid duplication of effort). 
- I'd _hope_ that anyone who has shown interest (e.g. you and @maowen) might state what they can do to help even if they can't do this - e.g. you might say "I can't create a PR because I won't have bandwidth in a reasonable timeframe but I am very happy to review and run testing".

If no one steps up then eventually this might be closed due to lack of interest.

olliw42, 2019-08-05:
understood. MANY thx for the very detailed and clear explanation, very helpful :+1:

olliw42, 2019-08-07:
merged so I close

closed 2019-08-07 by olliw42