mavlink/mavlink #1160: MAVLINK_HELPER does not name a type

ArcticSubmarine, 2019-06-06:
Hi, 
I'm currently working on a project using mavlink and arduino, and a compiling error occures because of the mavlink_conversion.h file. 

My code :+1: 

`#include <mavlink_types.h>
#include <mavlink_helpers.h>
#include <checksum.h>
#include <protocol.h>
#include <mavlink_sha256.h>
#include <mavlink_get_info.h>
#include <mavlink_conversions.h>`

The error : 

`Arduino : 1.8.9 (Linux), Carte : "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/arduino-builder -dump-prefs -logger=machine -hardware /home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware -hardware /home/arctic_submarine/.arduino15/packages -tools /home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/tools-builder -tools /home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr -tools /home/arctic_submarine/.arduino15/packages -built-in-libraries /home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/libraries -libraries /home/arctic_submarine/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=0X2A03_0X0042 -ide-version=10809 -build-path /tmp/arduino_build_891944 -warnings=none -build-cache /tmp/arduino_cache_202917 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr -verbose /home/arctic_submarine/Arduino/Com_MAVLink_V2/Com_MAVLink_V2.ino
/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/arduino-builder -compile -logger=machine -hardware /home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware -hardware /home/arctic_submarine/.arduino15/packages -tools /home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/tools-builder -tools /home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr -tools /home/arctic_submarine/.arduino15/packages -built-in-libraries /home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/libraries -libraries /home/arctic_submarine/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=0X2A03_0X0042 -ide-version=10809 -build-path /tmp/arduino_build_891944 -warnings=none -build-cache /tmp/arduino_cache_202917 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr -verbose /home/arctic_submarine/Arduino/Com_MAVLink_V2/Com_MAVLink_V2.ino
Using board 'mega' from platform in folder: /home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/arduino/avr
Using core 'arduino' from platform in folder: /home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/arduino/avr
Detecting libraries used...
/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/arduino/avr/cores/arduino -I/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/arduino/avr/variants/mega /tmp/arduino_build_891944/sketch/Com_MAVLink_V2.ino.cpp -o /dev/null
/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/arduino/avr/cores/arduino -I/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/arduino/avr/variants/mega -I/home/arctic_submarine/Arduino/libraries/mavlink /tmp/arduino_build_891944/sketch/Com_MAVLink_V2.ino.cpp -o /dev/null
Generating function prototypes...
/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/arduino/avr/cores/arduino -I/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/arduino/avr/variants/mega -I/home/arctic_submarine/Arduino/libraries/mavlink /tmp/arduino_build_891944/sketch/Com_MAVLink_V2.ino.cpp -o /tmp/arduino_build_891944/preproc/ctags_target_for_gcc_minus_e.cpp
/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_891944/preproc/ctags_target_for_gcc_minus_e.cpp
Compilation du croquis...
/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/arduino/avr/cores/arduino -I/home/arctic_submarine/Logiciels/arduino-1.8.9-linux64/arduino-1.8.9/hardware/arduino/avr/variants/mega -I/home/arctic_submarine/Arduino/libraries/mavlink /tmp/arduino_build_891944/sketch/Com_MAVLink_V2.ino.cpp -o /tmp/arduino_build_891944/sketch/Com_MAVLink_V2.ino.cpp.o
In file included from /home/arctic_submarine/Arduino/libraries/mavlink/mavlink_helpers.h:6:0,
                 from /home/arctic_submarine/Arduino/Com_MAVLink_V2/Com_MAVLink_V2.ino:2:
/home/arctic_submarine/Arduino/libraries/mavlink/mavlink_conversions.h:37:1: error: 'MAVLINK_HELPER' does not name a type
 MAVLINK_HELPER void mavlink_quaternion_to_dcm(const float quaternion[4], float dcm[3][3])
 ^
/home/arctic_submarine/Arduino/libraries/mavlink/mavlink_conversions.h:67:1: error: 'MAVLINK_HELPER' does not name a type
 MAVLINK_HELPER void mavlink_dcm_to_euler(const float dcm[3][3], float* roll, float* pitch, float* yaw)
 ^
/home/arctic_submarine/Arduino/libraries/mavlink/mavlink_conversions.h:101:1: error: 'MAVLINK_HELPER' does not name a type
 MAVLINK_HELPER void mavlink_quaternion_to_euler(const float quaternion[4], float* roll, float* pitch, float* yaw)
 ^
/home/arctic_submarine/Arduino/libraries/mavlink/mavlink_conversions.h:117:1: error: 'MAVLINK_HELPER' does not name a type
 MAVLINK_HELPER void mavlink_euler_to_quaternion(float roll, float pitch, float yaw, float quaternion[4])
 ^
/home/arctic_submarine/Arduino/libraries/mavlink/mavlink_conversions.h:145:1: error: 'MAVLINK_HELPER' does not name a type
[... cut ...]
/home/arctic_submarine/Arduino/libraries/mavlink/mavlink_helpers.h:946:15: error: 'MAVLINK_STX' was not declared in this scope
      if (c == MAVLINK_STX)
               ^
Utilisation de la bibliothèque mavlink prise dans le dossier : /home/arctic_submarine/Arduino/libraries/mavlink (legacy)
exit status 1
Erreur de compilation pour la carte Arduino/Genuino Mega or Mega 2560`

I don't know how to solve this, as the MAVLINK_HEDER type is defined in mavlink_helpers.h, in which mavlink_conversions.h is included (so I can't include it back).

Thank you for your help

julianoes, 2019-06-07:
We can't really help you without seeing the code. So you need to at least add a link to where your code is. Also you might want to add what instructions you're following. Without more info there is not much we can do and we have to close the issue.

ArcticSubmarine, 2019-06-07:
There's no code, it's just an inclusion problem. But I managed to solve it somehow. Thanks

closed 2019-06-07 by ArcticSubmarine

julianoes, 2019-06-07:
If you describe your "somehow" then someone finding this later might be thankful. :smile:

ArcticSubmarine, 2019-06-07:
Of course : I looked at which declarations were missing in my file, using `#ifndefine ...`. I was looking for a "cleaner solution" but this surely works.