mavlink/mavlink #1164: mavlink_parse_char() is giving me an error all the time

RohanKurane, 2019-06-20:
Hey Guys,

mavlink_parse_char() was working fine till a while ago but now it is always failing on me and does not parse any message. What can be going wrong ? How can I debug this ? There is not real error message that the function returns so I cant tell what is going on. 

@LorenzMeier 
@lucasdemarchi 

Thanks
Rohan

RohanKurane, 2019-06-20:
It seems that all it does it call mavlink_frame_char and it seems to return a 0. This is for all the bytes received on the socket. Any idea what might be going on all of a sudden ?

RohanKurane, 2019-06-20:
Any clue what I can look at ? This code worked fine earlier but now it seems mavlink_parse_char() fails every byte and no messages are processed. 
@LorenzMeier 
@lucasdemarchi

RohanKurane, 2019-06-20:
One interesting thing is, I cloned the c_library_v1 to get the mavlink headers. 
But when I started QGC on my laptop I saw this message
```
Switching outbound to mavlink 2.0 due to incoming mavlink 2.0 packet
```
Do I need to clone c_library_v2 instead now ?

RohanKurane, 2019-06-20:
That did not help by cloning the v2 library. 
Any help is appreciated.

RohanKurane, 2019-06-20:
I am defintetly using version 2 as shown by calling 
```
mavlink_get_channel_status
``` 
and checking the version. 

The channel COMM_0 fluctuates its state between IDLE, GOT_STX and got GOT_LENGTH and back to IDLE. 
I do not know what is suppose to happen here. And why all of a sudden I am getting this problem

RohanKurane, 2019-06-21:
I disabled the modem manager and restarted the mavlink-router. None of that made a difference. 
Im hoping one of you guys can help me out here.

RohanKurane, 2019-06-21:
Is this odd ?
```
sudo systemctl status mavlink-router.service 
● mavlink-router.service - MAVLink Router
   Loaded: loaded (/lib/systemd/system/mavlink-router.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-06-20 22:13:40 PDT; 3min 26s ago
  Process: 7317 ExecStartPre=/bin/bash -c echo 4 > /sys/class/tty/ttyS1/rx_trig_bytes (code=exited, status=0/SUCCESS)
 Main PID: 7321 (mavlink-routerd)
   CGroup: /system.slice/mavlink-router.service
           └─7321 /usr/bin/mavlink-routerd

Jun 20 22:16:55 intel-aero mavlink-routerd[7321]: Error sending tcp packet (Broken pipe)
Jun 20 22:16:55 intel-aero mavlink-routerd[7321]: Error sending tcp packet (Broken pipe)
Jun 20 22:16:55 intel-aero mavlink-routerd[7321]: Error sending tcp packet (Broken pipe)
Jun 20 22:16:55 intel-aero mavlink-routerd[7321]: TCP Connection [8] closed
Jun 20 22:16:58 intel-aero mavlink-routerd[7321]: TCP connection [8] accepted
Jun 20 22:16:58 intel-aero mavlink-routerd[7321]: TCP Connection [8] closed
Jun 20 22:17:02 intel-aero mavlink-routerd[7321]: TCP connection [8] accepted
Jun 20 22:17:02 intel-aero mavlink-routerd[7321]: TCP Connection [8] closed
Jun 20 22:17:05 intel-aero mavlink-routerd[7321]: TCP connection [8] accepted
Jun 20 22:17:05 intel-aero mavlink-routerd[7321]: TCP Connection [8] closed
```

If I restart it, it keeps closing the connection. I do see the mavlink attributes in QGC in the mavlink widget. But QGC uses UDP to connect. The above says the TCP connection is closed which is what is used to read the stream and do a `mavlink_parse_char()` on it.

RohanKurane, 2019-06-21:
Wow. It was a wrong header usage. Im sorry @lucasdemarchi @LorenzMeier @acfloria 
Im closing this issue

closed 2019-06-21 by RohanKurane

tatjam, 2024-03-12:
Just in case anybody has this issue in the future (as it was my case :wink:), if you want to use "default" MAVLink you should use the "common/" and not the "standard/" directory!

hamishwillee, 2024-03-13:
> Just in case anybody has this issue in the future (as it was my case 😉), if you want to use "default" MAVLink you should use the "common/" and not the "standard/" directory!

This has nothing to do with this issue.

metehand003, 2024-09-04:
Sorry to bring this up but I m having this issue as well any idea how to fix this ? I am running MAVLink C v2 library on STM32F407 using CUBE IDE. When ever I start transmitting an encaplated data message. It just stops parsing.